rockchip-linux/mpp #224: 最新版本

jiale-gdyd, 2021-08-17:
解码H264视频没问题，但是解码jpg图片会抛出段错误异常
"mpp_dec_parser" received signal SIGSEGV, Segmentation fault.

FumasterLin, 2021-08-17:
本地测试没问题
1、git log看一下
2、如何复现的？

jiale-gdyd, 2021-08-18:
commit ac4c6718fc8661fe80b5309dd2f4526cf0b7953c (HEAD -> develop, origin/develop, origin/HEAD)
Author: Johnson Ding <johnson.ding@rock-chips.com>
Date:   Tue Jul 20 15:25:58 2021 +0800

    [vepu_common]: Support RGBA little endian format
    
    Change-Id: Ic7a390a1db11d3416f1783d72c7be3a052bb6e89
    Signed-off-by: Johnson Ding <johnson.ding@rock-chips.com>

commit 9274775ae9c66605fa2ae5a1375f98f11700191e
Author: Johnson Ding <johnson.ding@rock-chips.com>
Date:   Mon Jul 19 18:06:05 2021 +0800

    [jpegd]: Support RGB 16-bit format output
    
    Change-Id: I54763a031b538c24860573e3c639c79a978e4a58
    Signed-off-by: Johnson Ding <johnson.ding@rock-chips.com>

commit be76be4634c591d3a536e61402405c4978930e0b
Author: Herman Chen <herman.chen@rock-chips.com>
Date:   Wed Aug 11 09:39:26 2021 +0800

    [hal_jpegd_vdpu]: Add pp switch for MppDev
    
    Change-Id: I9e38ac2097b4fff6e33fd8c4fd9f16fbe92adf68
    Signed-off-by: Herman Chen <herman.chen@rock-chips.com>
    Signed-off-by: Yandong Lin <yandong.lin@rock-chips.com>

commit 957b5e0365ecd6db7fd0557daefc5a589ea09419
Author: sayon.chen <sayon.chen@rock-chips.com>
Date:   Fri Aug 6 10:37:08 2021 +0800

    [h265e_ps]: Svc case disable tmvp_en
    
    Change-Id: I93068be4cc528c228258c757d6bd05bb18245212
    Signed-off-by: sayon.chen <sayon.chen@rock-chips.com>

commit 49f29006a3c728bb3570279b7b5407ec29e09f6d

就是使用rkmedia中的rkmedia_vdec_test.c测试发现的，视频解码时没有问题的，但是JPG解码就不行。mpp version: 2ceca7b5 author: sayon.chen    2021-07-01 [hal_jpege_vpu]: Fix qtable memory leak这个版本是OK的

qvoid, 2021-08-18:
有没有完整的崩溃的调用栈信息？使用mpi_dec_test是否有问题？

jiale-gdyd, 2021-08-18:
mpp[3683]: mpp_info: mpp version: c7ed08c9 author: Yandong Lin   2021-08-16 [h264d_parser]: optimize the size of some struct
mpp[3683]: mpp_rt: NOT found ion allocator
mpp[3683]: mpp_rt: found drm allocator
[New LWP 3690]
[New LWP 3691]
mpp[3683]: mpp_info: mpp version: c7ed08c9 author: Yandong Lin   2021-08-16 [h264d_parser]: optimize the size of some struct
[New LWP 3692]
[New LWP 3693]
[New LWP 3694]
[New LWP 3695]
[New LWP 3696]
[New LWP 3697]
[New LWP 3698]
[New LWP 3699]
[New LWP 3700]
[New LWP 3701]
[LWP 3695 exited]
[New LWP 3702]

Thread 10 "mpp_dec_parser" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 3692]
0x9c60f140 in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x9c60f140 in ?? () from /lib/libc.so.6
#1  0xa150be08 in ?? () from /oem/usr/lib/librockchip_mpp.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

jiale-gdyd, 2021-08-18:
只能看到这个。根据mpp_dec_parser这个，应该是mpp/codec/mpp_dec.cpp --> void *mpp_dec_parser_thread(void *data)这个线程出现问题了

qvoid, 2021-08-18:
可以编译个debug库试试看。还有用
```
mpi_dec_test -i <input.jpg> -t 8 -w <width> -h <height>
```
看下解码有没有问题。
>  mpi_dec_test 在MPP的编译后的build/linux目录下可以找到

jiale-gdyd, 2021-08-18:
Starting program: /root/xindun/mpi_dec_test -i /root/xindun/signal.jpg -t 8 -w 1280 -h 720
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.
mpp[3807]: mpi_dec_utils: cmd parse result:
mpp[3807]: mpi_dec_utils: input  file name: /root/xindun/signal.jpg
mpp[3807]: mpi_dec_utils: output file name: 
mpp[3807]: mpi_dec_utils: config file name: 
mpp[3807]: mpi_dec_utils: width      : 1280
mpp[3807]: mpi_dec_utils: height     :  720
mpp[3807]: mpi_dec_utils: type       : 8
mpp[3807]: mpi_dec_utils: debug flag : 0
mpp[3807]: mpi_dec_utils: max frames : 0
mpp[3807]: mpi_dec_test: mpi_dec_test start
mpp[3807]: mpi_dec_test: input file size 45585
mpp[3807]: mpp_rt: NOT found ion allocator
mpp[3807]: mpp_rt: found drm allocator
mpp[3807]: mpp_info: mpp version: c7ed08c9 author: Yandong Lin   2021-08-16 [h264d_parser]: optimize the size of some struct
mpp[3807]: mpi_dec_test: 0x2fea0 mpi_dec_test decoder test start w 1280 h 720 type 8
[New LWP 3808]
mpp[3807]: mpp_device: mpp_dev_init found unsupported client type 3 in platform 50202
mpp[3807]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt dev type 1 -> 3 switch failed ret -6
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs setup output format 0 failed
mpp[3807]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[3807]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[3807]: mpi_dec_test: 0x2fea0 decoded frame 0
mpp[3807]: mpp_device: mpp_dev_init found unsupported client type 3 in platform 50202
mpp[3807]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt dev type 1 -> 3 switch failed ret -6
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs setup output format 0 failed
mpp[3807]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 12
mpp[3807]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[3807]: mpi_dec_test: 0x2fea0 decoded frame 1
mpp[3807]: mpp_device: mpp_dev_init found unsupported client type 3 in platform 50202
mpp[3807]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt dev type 1 -> 3 switch failed ret -6
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs setup output format 0 failed
mpp[3807]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 12
mpp[3807]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[3807]: mpi_dec_test: 0x2fea0 decoded frame 2
mpp[3807]: mpp_device: mpp_dev_init found unsupported client type 3 in platform 50202
mpp[3807]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt dev type 1 -> 3 switch failed ret -6
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs setup output format 0 failed
mpp[3807]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[3807]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 12

这个倒是不会崩溃，但是也解不了码

jiale-gdyd, 2021-08-18:
是RV1126，SDK是1.7.0

jiale-gdyd, 2021-08-18:
你们那边有RV1126的EVB板吗？在板上更换最新编译的mpp库，在板子上跑jpg解码正常吗？

HermanChen, 2021-08-18:
加这个补丁试试

From 8ca5be73ba68b07da510b9649a09b517113c3eec Mon Sep 17 00:00:00 2001
From: Herman Chen <herman.chen@rock-chips.com>
Date: Wed, 18 Aug 2021 10:05:56 +0800
Subject: [PATCH] [hal_jpegd]: Add have_pp check

Change-Id: Id9b9a5004d0bdc5fd88e084966ada360f575b1a7

---
 mpp/hal/vpu/jpegd/hal_jpegd_base.h   |  2 ++
 mpp/hal/vpu/jpegd/hal_jpegd_common.c | 12 +++++++++++-
 mpp/hal/vpu/jpegd/hal_jpegd_common.h |  1 +
 mpp/hal/vpu/jpegd/hal_jpegd_vdpu1.c  |  1 +
 mpp/hal/vpu/jpegd/hal_jpegd_vdpu2.c  |  1 +
 5 files changed, 16 insertions(+), 1 deletion(-)

diff --git a/mpp/hal/vpu/jpegd/hal_jpegd_base.h b/mpp/hal/vpu/jpegd/hal_jpegd_base.h
index 63e9de80..43a1ef59 100644
--- a/mpp/hal/vpu/jpegd/hal_jpegd_base.h
+++ b/mpp/hal/vpu/jpegd/hal_jpegd_base.h
@@ -39,6 +39,7 @@ typedef struct JpegdHalCtx {
     MppBufSlots            frame_slots;
     MppDev                 dev;
     MppClientType          dev_type;
+    RK_U32                 codec_type;
     void                   *regs;
     MppBufferGroup         group;
     MppBuffer              frame_buf;
@@ -55,6 +56,7 @@ typedef struct JpegdHalCtx {
     RK_S32                 pkt_fd;    /* input stream's physical address(fd) */
     RK_S32                 frame_fd;  /* output picture's physical address(fd) */
 
+    RK_U32                 have_pp;
     PPInfo                 pp_info;
 } JpegdHalCtx;
 
diff --git a/mpp/hal/vpu/jpegd/hal_jpegd_common.c b/mpp/hal/vpu/jpegd/hal_jpegd_common.c
index a105e348..7941aa82 100644
--- a/mpp/hal/vpu/jpegd/hal_jpegd_common.c
+++ b/mpp/hal/vpu/jpegd/hal_jpegd_common.c
@@ -432,6 +432,15 @@ void jpegd_write_qp_ac_dc_table(JpegdHalCtx *ctx,
     return;
 }
 
+void jpegd_check_have_pp(JpegdHalCtx *ctx)
+{
+    ctx->codec_type = mpp_get_vcodec_type();
+    ctx->have_pp = ((ctx->dev_type == VPU_CLIENT_VDPU1) &&
+                    (ctx->codec_type & (1 << VPU_CLIENT_VDPU1_PP))) ||
+                   ((ctx->dev_type == VPU_CLIENT_VDPU2) &&
+                    (ctx->codec_type & (1 << VPU_CLIENT_VDPU2_PP)));
+}
+
 MPP_RET jpegd_setup_output_fmt(JpegdHalCtx *ctx, JpegdSyntax *s, RK_S32 output)
 {
     jpegd_dbg_func("enter\n");
@@ -442,7 +451,8 @@ MPP_RET jpegd_setup_output_fmt(JpegdHalCtx *ctx, JpegdSyntax *s, RK_S32 output)
     MppFrame frm = NULL;
     MPP_RET ret = MPP_OK;
 
-    if (ctx->set_output_fmt_flag && (ctx->output_fmt != s->output_fmt)) {
+    if (ctx->have_pp && ctx->set_output_fmt_flag &&
+        ctx->output_fmt != s->output_fmt) {
         MppFrameFormat fmt = MPP_FMT_BUTT;
 
         /* Using pp to convert all format to yuv420sp */
diff --git a/mpp/hal/vpu/jpegd/hal_jpegd_common.h b/mpp/hal/vpu/jpegd/hal_jpegd_common.h
index d1ce7f14..6770895f 100644
--- a/mpp/hal/vpu/jpegd/hal_jpegd_common.h
+++ b/mpp/hal/vpu/jpegd/hal_jpegd_common.h
@@ -75,6 +75,7 @@ RK_U32 jpegd_vdpu_tail_0xFF_patch(MppBuffer stream, RK_U32 length);
 void jpegd_write_qp_ac_dc_table(JpegdHalCtx *ctx,
                                 JpegdSyntax*syntax);
 
+void jpegd_check_have_pp(JpegdHalCtx *ctx);
 MPP_RET jpegd_setup_output_fmt(JpegdHalCtx *ctx, JpegdSyntax *syntax,
                                RK_S32 output);
 
diff --git a/mpp/hal/vpu/jpegd/hal_jpegd_vdpu1.c b/mpp/hal/vpu/jpegd/hal_jpegd_vdpu1.c
index c63954ba..5df8dae4 100644
--- a/mpp/hal/vpu/jpegd/hal_jpegd_vdpu1.c
+++ b/mpp/hal/vpu/jpegd/hal_jpegd_vdpu1.c
@@ -785,6 +785,7 @@ MPP_RET hal_jpegd_vdpu1_init(void *hal, MppHalCfg *cfg)
     pp_info->pp_enable = 0;
     pp_info->pp_in_fmt = PP_IN_FORMAT_YUV420SEMI;
     pp_info->pp_out_fmt = PP_OUT_FORMAT_YUV420INTERLAVE;
+    jpegd_check_have_pp(JpegHalCtx);
 
     JpegHalCtx->output_fmt = MPP_FMT_YUV420SP;
     JpegHalCtx->set_output_fmt_flag = 0;
diff --git a/mpp/hal/vpu/jpegd/hal_jpegd_vdpu2.c b/mpp/hal/vpu/jpegd/hal_jpegd_vdpu2.c
index 1c11111f..0c15b3a7 100644
--- a/mpp/hal/vpu/jpegd/hal_jpegd_vdpu2.c
+++ b/mpp/hal/vpu/jpegd/hal_jpegd_vdpu2.c
@@ -771,6 +771,7 @@ MPP_RET hal_jpegd_vdpu2_init(void *hal, MppHalCfg *cfg)
     pp_info->pp_enable = 0;
     pp_info->pp_in_fmt = PP_IN_FORMAT_YUV420SEMI;
     pp_info->pp_out_fmt = PP_OUT_FORMAT_YUV420INTERLAVE;
+    jpegd_check_have_pp(JpegHalCtx);
 
     JpegHalCtx->output_fmt = MPP_FMT_YUV420SP;
     JpegHalCtx->set_output_fmt_flag = 0;
-- 
2.27.0

HermanChen, 2021-08-18:
加这个补丁看看
[8ca5be7.diff.zip](https://github.com/rockchip-linux/mpp/files/7004040/8ca5be7.diff.zip)

jiale-gdyd, 2021-08-18:
额，打了补丁后还是一样的问题

HermanChen, 2021-08-18:
有出错 log 么？

jiale-gdyd, 2021-08-19:
Thread 10 "mpp_dec_parser" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 695]
0x9c670140 in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x9c670140 in ?? () from /lib/libc.so.6
#1  0xa156ce08 in ?? () from /oem/usr/lib/librockchip_mpp.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

HermanChen, 2021-08-19:
到这里取一下最新版本看看：
https://github.com/hermanchen/mpp/

jiale-gdyd, 2021-08-19:
Thread 10 "mpp_dec_parser" received signal SIGSEGV, Segmentation fault.
[Switching to LWP 1155]
0x9c5da140 in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x9c5da140 in ?? () from /lib/libc.so.6
#1  0xa14d702c in ?? () from /oem/usr/lib/librockchip_mpp.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

还是有问题的

jiale-gdyd, 2021-08-19:
width:[1280], height:[720], path:[./signal.jpg], mode:[hard], type:[jpeg], calib:[false], flip:[n]
opencv version:[4.5.3]
mpp[1176]: mpp_info: mpp version: 2ceca7b5 author: sayon.chen    2021-07-01 [hal_jpege_vpu]: Fix qtable memory leak
mpp[1176]: mpp_rt: NOT found ion allocator
mpp[1176]: mpp_rt: found drm allocator
mpp[1176]: mpp_info: mpp version: 2ceca7b5 author: sayon.chen    2021-07-01 [hal_jpege_vpu]: Fix qtable memory leak

这个版本是正常的

HermanChen, 2021-08-19:
了解，可以二分法试下看看哪个版本出现的问题……
我们这边测试 https://github.com/hermanchen/mpp/ 这里的最新版本应该是好的

jiale-gdyd, 2021-08-19:
[drm]: Add DRM_CLOEXEC|DRM_RDWR flag on fd import …
1. Add drm fd open flag.
2. Remove unused handle process on mmap.
Change-Id: Id0b29b8914c1e8e8ae6af40877df21f73b46346e

@HermanChen
HermanChen committed on 15 Jul

刚试了下，从这个开始(包括这个)d到最新的不行了

HermanChen, 2021-08-19:
了解，这个补丁后面还有几个 drm 的补丁，主要是修改 drm fd 打开时候的访问权限问题

HermanChen, 2021-08-19:
可能外部分配的 buffer 也要修改下。

jiale-gdyd, 2021-08-19:
现在可以用的这个CPU占用比较高

jiale-gdyd, 2021-08-19:
请问下，这个问题啥时候能修好呢?

Jeffwhen, 2021-08-24:
我碰到了一样的问题，我去看了一下代码，意思是如果解码 jpeg 输出要是默认以外的格式都要使用 `PP` 结尾的解码器？

`rv1126` 的支持列表里面没有 `PP` 结尾的解码类型，意思是 `rv1126` 的 VPU 硬件上不支持 post process 吗？

HermanChen, 2021-08-24:
测试试一下 https://github.com/hermanchen/mpp/ 这里的最新版本，看看还会不会有问题

HermanChen, 2021-08-24:
如果没有问题，我就把开发分支的版本同步过来

Jeffwhen, 2021-08-24:
@HermanChen 还是有问题

```
mpp[899]: mpp_rt: NOT found ion allocator
mpp[899]: mpp_rt: found drm allocator
mpp[899]: mpp_info: mpp version: 0faeb94 author: Johnson Ding   2021-08-23 [hal_jpegd]: Check output format  before decoding
mpp[899]: jpegd_parser: jpegd_init exit
mpp[899]: mpp_dec: detail: 0x8c4600 control cmd 0031000a param 0xaef01434 start
mpp[899]: mpp_dec: ctrl proc 0 cmd 0031000a
mpp[899]: jpegd_parser: jpegd_control enter
mpp[899]: jpegd_parser: jpegd_control exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_control enter
mpp[899]: HAL_JPEG_VDPU2: output_format:2
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_control exit ret 0
mpp[899]: mpp_dec: ctrl proc 1 done send 1
mpp[899]: mpp_dec: mpp_pkt_in_rdy 0
mpp[899]: mpp_dec: detail: 0x8c4600 control cmd 0031000a param 0xaef01434 finish
unsupported descriptor subtype VS_STILL_IMAGE_FRAME
unsupported descriptor subtype VS_COLORFORMAT
unsupported descriptor subtype VS_STILL_IMAGE_FRAME
unsupported descriptor subtype VS_COLORFORMAT
bmHint: 0001
bFormatIndex: 2
bFrameIndex: 10
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 0
wCompWindowSize: 0
wDelay: 32
dwMaxVideoFrameSize: 1843200
dwMaxPayloadTransferSize: 3072
bInterfaceNumber: 1
attempt to claim already-claimed interface 1
mpp[899]: mpp_dec: mpp_pkt_in_rdy 0
mpp[899]: mpp_dec: poll ready
mpp[899]: mpp_dec: task in ready
mpp[899]: jpegd_parser: jpegd_prepare enter
mpp[899]: jpegd_parser: jpegd_split_frame enter
mpp[899]: jpegd_parser: jpegd_split_frame exit
mpp[899]: jpegd_parser: jpegd_prepare exit
mpp[899]: jpegd_parser: jpegd_parse enter
mpp[899]: jpegd_parser: jpegd_decode_frame enter
mpp[899]: jpegd_parser: jpegd_decode_frame exit
mpp[899]: jpegd_parser: jpegd_allocate_frame enter
mpp[899]: jpegd_parser: jpegd_allocate_frame exit
mpp[899]: jpegd_parser: jpegd_update_frame enter
mpp[899]: jpegd_parser: jpegd_update_frame exit
mpp[899]: jpegd_parser: jpegd_parse exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs enter
mpp[899]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt enter
mpp[899]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_gen_regs enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_regs_init enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_regs_init exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_write_code_word_number enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_write_code_word_number exit
mpp[899]: HAL_JPEGD_COMMON: jpegd_write_qp_ac_dc_table enter
mpp[899]: HAL_JPEGD_COMMON: jpegd_write_qp_ac_dc_table exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_chroma_table_id enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_chroma_table_id exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_stream_offset enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_stream_offset exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_setup_pp enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[899]: HAL_JPEG_VDPU2: output_frame_fd:2, reg131:2
mpp[899]: HAL_JPEG_VDPU2: jpegd_setup_pp exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_gen_regs exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start enter
mpp[899]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 14 Bad address
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 14
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait enter
mpp[899]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait exit
mpp[899]: mpp_dec: mpp_pkt_in_rdy 0
mpp[899]: mpp_dec: mpp_pkt_in_rdy 0
mpp[899]: mpp_dec: poll ready
mpp[899]: mpp_dec: task in ready
mpp[899]: jpegd_parser: jpegd_prepare enter
mpp[899]: jpegd_parser: jpegd_split_frame enter
mpp[899]: jpegd_parser: jpegd_split_frame exit
mpp[899]: jpegd_parser: jpegd_prepare exit
mpp[899]: jpegd_parser: jpegd_parse enter
mpp[899]: jpegd_parser: jpegd_decode_frame enter
mpp[899]: jpegd_parser: jpegd_decode_frame exit
mpp[899]: jpegd_parser: jpegd_allocate_frame enter
mpp[899]: jpegd_parser: jpegd_allocate_frame exit
mpp[899]: jpegd_parser: jpegd_update_frame enter
mpp[899]: jpegd_parser: jpegd_update_frame exit
mpp[899]: jpegd_parser: jpegd_parse exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs enter
mpp[899]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt enter
mpp[899]: HAL_JPEGD_COMMON: jpegd_setup_output_fmt exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_gen_regs enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_regs_init enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_regs_init exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_write_code_word_number enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_write_code_word_number exit
mpp[899]: HAL_JPEGD_COMMON: jpegd_write_qp_ac_dc_table enter
mpp[899]: HAL_JPEGD_COMMON: jpegd_write_qp_ac_dc_table exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_chroma_table_id enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_chroma_table_id exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_stream_offset enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_set_stream_offset exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_setup_pp enter
mpp[899]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[899]: HAL_JPEG_VDPU2: output_frame_fd:2, reg131:2
mpp[899]: HAL_JPEG_VDPU2: jpegd_setup_pp exit
mpp[899]: HAL_JPEG_VDPU2: jpegd_gen_regs exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_gen_regs exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start enter
mpp[899]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 12
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start exit
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait enter
^Cmpp[899]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[899]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait exit
```

Jeffwhen, 2021-08-24:
@HermanChen mpi_dec_test 的日志

```
linaro@rv1126:~$ sudo mpi_dec_test -i ./IMG_20210709_202627R_20210709_202718093.jpg -o jpg.bin -width 960 -height 1280 -t 8 -d
mpp[460]: mpi_dec_utils: invalid debug flag
mpp[460]: mpi_dec_utils: cmd parse result:
mpp[460]: mpi_dec_utils: input  file name: ./IMG_20210709_202627R_20210709_202718093.jpg
mpp[460]: mpi_dec_utils: output file name: jpg.bin
mpp[460]: mpi_dec_utils: config file name:
mpp[460]: mpi_dec_utils: width      :  960
mpp[460]: mpi_dec_utils: height     : 1280
mpp[460]: mpi_dec_utils: type       : 8
mpp[460]: mpi_dec_utils: debug flag : 0
mpp[460]: mpi_dec_utils: max frames : 0
mpp[460]: mpi_dec_test: mpi_dec_test start
mpp[460]: mpi_dec_test: input file size 85561
mpp[460]: mpp_rt: NOT found ion allocator
mpp[460]: mpp_rt: found drm allocator
mpp[460]: mpp_info: mpp version: 0faeb94 author: Johnson Ding   2021-08-23 [hal_jpegd]: Check output format  before decoding
mpp[460]: mpi_dec_test: 0x43d1a0 mpi_dec_test decoder test start w 960 h 1280 type 8
mpp[460]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[460]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 14 Bad address
mpp[460]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 14
mpp[460]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[460]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[460]: mpi_dec_test: 0x43d1a0 decoded frame 0
mpp[460]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[460]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[460]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 12
mpp[460]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[460]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[460]: mpi_dec_test: 0x43d1a0 decoded frame 1
mpp[460]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[460]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[460]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 12
```

Jeffwhen, 2021-08-24:
@HermanChen 我应该升级内核么？我看是系统调用报的错。

HermanChen, 2021-08-24:
看下编译的时候有没有加上 HAVE_DRM？

jiale-gdyd, 2021-08-24:
我拉取了[vepu_common]: Support RGBA little endian format
Change-Id: If896999f442b58078543b46404799ee1f3cb3dd1

1. Add drm fd open flag.
2. Remove unused handle process on mmap.

Change-Id: Id0b29b8914c1e8e8ae6af40877df21f73b46346e

FumasterLin, 2021-08-24:
合入如下的diff后再试试

```
diff --git a/build/linux/arm/make-Makefiles.bash b/build/linux/arm/make-Makefiles.bash
index 9dcbc7e8..1e891339 100755
--- a/build/linux/arm/make-Makefiles.bash
+++ b/build/linux/arm/make-Makefiles.bash
@@ -24,5 +24,6 @@ fi
 cmake -DCMAKE_BUILD_TYPE=Release \
       -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
       -DRKPLATFORM=ON \
+      -DHAVE_DRM=ON \
       -G "Unix Makefiles" \
       ${MPP_TOP}
```

Jeffwhen, 2021-08-25:
@FumasterLin 加上这个也一样，最新版本

FumasterLin, 2021-08-25:
编译mpp的输出log发来看看

Jeffwhen, 2021-08-25:
@FumasterLin 截了一小段 `make VERBOSE=1` 的日志，可以看到 `HAVE_DRM` 标志已经生效，编译时有这个 flag

```
[  0%] Building CXX object osal/CMakeFiles/osal.dir/mpp_soc.cpp.o
cd /home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/build/linux/arm/osal && /usr/bin/arm-linux-gnueabihf-g++  -DARMLINUX -DHAVE_DRM -D__gnu_linux__ -I/home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/inc -I/home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/utils -I/home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/osal/inc -I/home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/osal -I/home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/osal/allocator -I/home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/osal/driver/inc  -fno-exceptions -fno-rtti -fPIC -O3 -DNDEBUG   -fPIC -Wall -Wextra -Wshadow -ffast-math -march=armv7-a -mfloat-abi=hard -mfpu=neon -Wno-multichar -fPIC -o CMakeFiles/osal.dir/mpp_soc.cpp.o -c /home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/osal/mpp_soc.cpp
```

FumasterLin, 2021-08-25:
确认下编译的时候有没有这个log出来

`-- compile with drm support`

Jeffwhen, 2021-08-25:
@FumasterLin 有。

```
-- cmake version 3.16.3
-- The C compiler identification is GNU 10.3.0
-- The CXX compiler identification is GNU 10.3.0
-- Check for working C compiler: /usr/bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /usr/bin/arm-linux-gnueabihf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detected ARMv7 system processor
-- Performing Test GCC_HAS_NO_NARROWING
-- Performing Test GCC_HAS_NO_NARROWING - Success
-- Performing Test GCC_HAS_STACK_REALIGN
-- Performing Test GCC_HAS_STACK_REALIGN - Failed
-- Found Git: /usr/bin/git (found version "2.27.0")
-- current version:
-- 0faeb94 author: Johnson Ding   2021-08-23 [hal_jpegd]: Check output format  before decoding
-- git version history:
-- 0faeb94 author: Johnson Ding   2021-08-23 [hal_jpegd]: Check output format  before decoding  (grafted, HEAD -> develop, origin/develop, origin/HEAD)
-- total 1 git version recorded
-- Install git hooks done
-- Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE)
-- Checking for one of the modules 'pthread'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- compile with drm support
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    RKPLATFORM

-- Build files have been written to: /home/deepdot/rv1126/rv1126rp_src/extra/hermanmpp/build/linux/arm
```

FumasterLin, 2021-08-25:
那就奇怪了。

H264/H265能解吗？

Jeffwhen, 2021-08-25:
@HermanChen 是我弄错了😂，install 位置不对，一直没有装上重新编译的版本。重新测试还是有一个报错，不过看起来是正常的？

```diff
mpp[1490]: mpi_dec_utils: cmd parse result:
mpp[1490]: mpi_dec_utils: input  file name: /home/rpdzkj/IMG_20210709_202627R_20210709_202718093.jpg
mpp[1490]: mpi_dec_utils: output file name: jpg.bin
mpp[1490]: mpi_dec_utils: config file name:
mpp[1490]: mpi_dec_utils: width      :  960
mpp[1490]: mpi_dec_utils: height     : 1280
mpp[1490]: mpi_dec_utils: type       : 8
mpp[1490]: mpi_dec_utils: debug flag : 0
mpp[1490]: mpi_dec_utils: max frames : 0
mpp[1490]: mpi_dec_test: mpi_dec_test start
mpp[1490]: mpi_dec_test: input file size 85561
mpp[1490]: mpp_rt: NOT found ion allocator
mpp[1490]: mpp_rt: found drm allocator
mpp[1490]: mpp_info: mpp version: 0faeb94 author: Johnson Ding   2021-08-23 [hal_jpegd]: Check output format  before decoding
mpp[1490]: mpi_dec_test: 0x4f11a0 mpi_dec_test decoder test start w 960 h 1280 type 8
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
- mpp[1490]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 14 Bad address
- mpp[1490]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 14
- mpp[1490]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
- mpp[1490]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 0
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 1
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 2
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 3
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 4
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 5
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 6
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 7
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 8
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 9
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 10
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[1490]: mpi_dec_test: 0x4f11a0 decoded frame 11
mpp[1490]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
```

Jeffwhen, 2021-08-25:
@HermanChen 解出来的输出是全 0，还是不行

Jeffwhen, 2021-08-25:
@FumasterLin 解 `h264` 也不行

```
rpdzkj@rv1126rp:~/Documents/app/build$ sudo mpi_dec_test -i ~/solarpower.h264 -o h264.bin -width 1920 -height 1080 -t 7 -n 1
mpp[2245]: mpi_dec_utils: cmd parse result:
mpp[2245]: mpi_dec_utils: input  file name: /home/rpdzkj/solarpower.h264
mpp[2245]: mpi_dec_utils: output file name: h264.bin
mpp[2245]: mpi_dec_utils: config file name:
mpp[2245]: mpi_dec_utils: width      : 1920
mpp[2245]: mpi_dec_utils: height     : 1080
mpp[2245]: mpi_dec_utils: type       : 7
mpp[2245]: mpi_dec_utils: debug flag : 0
mpp[2245]: mpi_dec_utils: max frames : 1
mpp[2245]: mpi_dec_test: mpi_dec_test start
mpp[2245]: mpi_dec_test: input file size 78231609
mpp[2245]: mpp_info: mpp version: 0faeb94 author: Johnson Ding   2021-08-23 [hal_jpegd]: Check output format  before decoding
mpp[2245]: mpi_dec_test: 0x4a88c0 mpi_dec_test decoder test start w 1920 h 1080 type 7
mpp[2245]: mpp_rt: NOT found ion allocator
mpp[2245]: mpp_rt: found drm allocator
mpp[2245]: mpi_dec_test: 0x4a88c0 decode_get_frame get info changed found
mpp[2245]: mpi_dec_test: 0x4a88c0 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
mpp[2245]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 14 Bad address
mpp[2245]: hal_h264d_rkv_reg: rkv_h264d_start send cmd failed 14
mpp[2245]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[2245]: hal_h264d_rkv_reg: rkv_h264d_wait poll cmd failed 5
mpp[2245]: mpi_dec_test: 0x4a88c0 decode get frame 0 err 1 discard 0
mpp[2245]: mpi_dec_test: 0x4a88c0 reach max frame number 1
mpp[2245]: mpi_dec_test: 0x4a88c0 input 22 pkt output 2 frm decode 1 frames
mpp[2245]: mpi_dec_test: test success max memory 3.98 MB
rpdzkj@rv1126rp:~/Documents/app/build$ ls -lh h264.bin
-rw-r--r-- 1 root root 0 Aug 25 17:08 h264.bin
```

FumasterLin, 2021-08-25:
还是怀疑allocator不对导致。直接按照下面的修改下，再试试。log也加上，确保有编译替换成功。

```
diff --git a/osal/linux/os_allocator.c b/osal/linux/os_allocator.c
index b7dbfeea..03a7fb21 100644
--- a/osal/linux/os_allocator.c
+++ b/osal/linux/os_allocator.c
@@ -36,23 +36,15 @@ MPP_RET os_allocator_get(os_allocator *api, MppBufferType type)
         *api = allocator_std;
     } break;
     case MPP_BUFFER_TYPE_ION : {
-        *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
-#if HAVE_DRM
-               (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
-#endif
-               allocator_std;
+        mpp_log("get drm alloc\n");
+        *api = allocator_drm;
     } break;
     case MPP_BUFFER_TYPE_EXT_DMA: {
         *api = allocator_ext_dma;
     } break;
     case MPP_BUFFER_TYPE_DRM : {
-#if HAVE_DRM
-        *api = (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_DRM)) ? allocator_drm :
-#else
-        * api =
-#endif
-               (mpp_rt_allcator_is_valid(MPP_BUFFER_TYPE_ION)) ? allocator_ion :
-               allocator_std;
+        mpp_log("get drm alloc\n");
+        *api = allocator_drm;
     } break;
     default : {
         ret = MPP_NOK;
```

Jeffwhen, 2021-08-25:
@FumasterLin 输出如下：

```
rpdzkj@rv1126rp:~$ sudo mpi_dec_test -i ~/IMG_20210709_202627R_20210709_202718093.jpg -o jpg.bin -width 960 -height 1280 -t 8 -f 0 -n 1
mpp[2730]: mpi_dec_utils: cmd parse result:
mpp[2730]: mpi_dec_utils: input  file name: /home/rpdzkj/IMG_20210709_202627R_20210709_202718093.jpg
mpp[2730]: mpi_dec_utils: output file name: jpg.bin
mpp[2730]: mpi_dec_utils: config file name:
mpp[2730]: mpi_dec_utils: width      :  960
mpp[2730]: mpi_dec_utils: height     : 1280
mpp[2730]: mpi_dec_utils: type       : 8
mpp[2730]: mpi_dec_utils: debug flag : 0
mpp[2730]: mpi_dec_utils: max frames : 1
mpp[2730]: mpi_dec_test: mpi_dec_test start
mpp[2730]: mpi_dec_test: input file size 85561
mpp[2730]: mpp_log: get drm alloc
mpp[2730]: mpp_log: get drm alloc
mpp[2730]: mpp_info: mpp version: 0faeb94 author: Johnson Ding   2021-08-23 [hal_jpegd]: Check output format  before decoding
mpp[2730]: mpi_dec_test: 0x4851a0 mpi_dec_test decoder test start w 960 h 1280 type 8
mpp[2730]: HAL_JPEG_VDPU2: jpegd_setup_pp pp_enable 0
mpp[2730]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 14 Bad address
mpp[2730]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_start send cmd failed 14
mpp[2730]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[2730]: HAL_JPEG_VDPU2: hal_jpegd_vdpu2_wait poll cmd failed 5
mpp[2730]: mpi_dec_test: 0x4851a0 decoded frame 0
mpp[2730]: mpi_dec_test: 0x4851a0 input 0 pkt output 0 frm decode 0 frames
mpp[2730]: mpi_dec_test: test success max memory 0.00 MB
```

另外，测试发现旧版本的 `mpp` 是可以用的，只是不能指定输出格式，rv1126 不带 PP 解码，所以需要使用 rga 来转码吗？

HermanChen, 2021-08-30:
问题已解决，可以更新下代码试试了~

Jeffwhen, 2021-09-01:
我测试没问题了。

HermanChen, 2021-09-01:
好的，问题解决麻烦关闭下问题~

jiale-gdyd, 2021-09-06:
我拉取最新的代码还是有问题:

解码H264视频没问题，但是解码jpg图片会抛出段错误异常
"mpp_dec_parser" received signal SIGSEGV, Segmentation fault.

如果我在最新的基础上，将以下提交的static MPP_RET os_allocator_drm_mmap(void *ctx, MppBufferInfo *data);函数该回去原来的话，解码h264/h265和jpg图片都是没有问题的

[drm]: Add DRM_CLOEXEC|DRM_RDWR flag on fd import …
1. Add drm fd open flag.
2. Remove unused handle process on mmap.
Change-Id: Id0b29b8914c1e8e8ae6af40877df21f73b46346e

@HermanChen
HermanChen committed on 15 Jul

HermanChen, 2021-09-06:
fd 标志问题需要在分配 drm 内存时也加上这个 flag，这样就可以解决这个问题