rockchip-linux/mpp #225: MPP_DEC_SET_IMMEDIATE_OUT - not working for h264 decoder

bakwc, 2021-08-17:
Hi! I'm trying to decode video with lowest possible latency.
I tried to apply `MPP_DEC_SET_IMMEDIATE_OUT`, but it seems like it has no effect. Latency is the same as without applying `MPP_DEC_SET_IMMEDIATE_OUT`.

I made a test sample (code + data) that reproduce it (modified mpi_enc_test, https://github.com/bakwc/rk_latency `main.cpp`)

### Info
Version: Latest (https://github.com/rockchip-linux/mpp develop)
SOC: rk3399
Frame size: 1920x1080, avg 100 kb

### Metrics
```
Avg latency: 40ms (time between frame sent to coder and decoded frame received)
Avg frame decode time: 8 ms (total work time / total frames count)
```

### Steps to reproduce

```
git clone https://github.com/bakwc/rk_latency
cd rk_latency
mkdir build
cd build
cmake ..
make
cd ../frames/
../build/rk_latency
```

### What I want to achive

Stable latency <= 16 ms.

Could you give some advice, or maybe there is already low-latency decoding example exists?

bakwc, 2021-08-18:
If I put frame, and then wait when it will be decoded, timing jumps from 2ms to 2000 ms: https://pastebin.com/0espvmbE
Sometimes frames decoded in 2ms. And sometimes performance drops to 200 / 400 / 1000 ms or higher.
Is there any option that can prevent it? I'd like to have some stable latency.

bakwc, 2021-08-18:
Seems like `MPP_DEC_SET_IMMEDIATE_OUT` used to set `immediate_out`, that was used in `h264d_api.c ``p_Dec->immediate_out`. But it was removed in https://github.com/rockchip-linux/mpp/commit/d9c59b7c0ad2ff2c2a49e6d7613a4160cdf7f551 - so now `MPP_DEC_SET_IMMEDIATE_OUT` is just ignored?
I will try to use `fast_out` instead.

bakwc, 2021-08-18:
I tried to enable `fast_out`, disable `batch_mode`, enable `sort_pts`, nothing works. Latency is just random between few ms and few seconds :(
```
    ret = mpp_dec_cfg_set_u32(cfg, "base:fast_out", fast_out);
    if (ret) {
        printf("%p failed to set fast_out ret %d\n", ctx, ret);
        goto MPP_TEST_OUT;
    }
```

bakwc, 2021-08-18:
I fixed that random performance drops by inserting usleep(2000) into infinite loop. After that everything works fine (6-7 ms frame decode latency 1920x1080p).
But still it's strange that options `MPP_DEC_SET_IMMEDIATE_OUT` and `fast_out` have no effect at all.
If anyone interested, here is working solution https://github.com/bakwc/rk_latency

closed 2021-08-18 by bakwc

HermanChen, 2021-08-19:
MPP_DEC_SET_IMMEDIATE_OUT can only remove the frame output process delay at dpb buffer management stage. It is a fix on process logic. It can not improve the decoding runtime schedule behavior.