rockchip-linux/mpp #226: jpegd_parser: jpeg_image_check_size unsupported resolution

XingP, 2021-08-18:
尝试用MPP的demo对JPEG的解码，

命令格式类似这样：mpi_dec_test -i xxx.jpg -t 8 -w xxx -h xxx -o output.yuv 

然后有时候会出现标题里的报错，跟了一下源码发现应该是hal_jpegd_common.c里jpeg_image_check_size对输入图像有分辨率上的要求，最小为48*48Bytes。
请问具体是这里的限制吗？这样的限制又是出于什么原因，能否解除掉呢？

此外，mpi_dec_test解码JPEG时会有-w/-h的参数要求输入，请问这两个参数具体有什么意义？在图像分辨率很大而输入参数较小的情况下，有可能会出现 mpp_dec: Assertion slot_size <= buffer_size failed at mpp_dec_advanced_thread 的报错，应该怎么避免？

HermanChen, 2021-08-19:
太小的图像使用硬件解码是很不划算的，需要拷贝码流，生成寄存器，配置硬件，等中断返回，一般情况下，D1 以上的jpeg图像用硬件解码才比较划算，硬件也不支持 48x48 以下的 jpeg。

mpi_dec_test 的 -w 和 -h 参数主要是给 jpeg 解码使用的，用来在解码之前开辟输出buffer，其他的视频格式不需要。
如果图像分辨率大于配置的，硬件解码输出图像时，会写出到非法地址去，出现报错和异常，这个问题需要上层应用去保证。

XingP, 2021-08-20:
好的，明白了。谢谢！

closed 2021-08-20 by XingP