rockchip-linux/mpp #230: 如何用普通用户解码

liyuming1978, 2021-08-24:
目前root用户解码是可以的。但是那个解码rkvdec驱动是root only的，导致普通用户没法解码

HermanChen, 2021-08-25:
在启动脚本里把设备文件的权限变成 777

liyuming1978, 2021-08-25:
不行，  mpp[9057]: vcodec_service: vcodec_service_cmd_send ioctl VPU_IOC_SET_REG failed ret -1 errno 14 Bad address。   需要RAW_IO权限

liyuming1978, 2021-08-25:
感觉这个可能和 业界做的差不多的，  都是需要 RAW_IO   SYSTEM_ADMIN  2个权限，  有可能还需要kmen权限 @HermanChen   。 这带来一个问题，  不同的应用怎么隔离

HermanChen, 2021-08-25:
Bad address 这个应该是有内部的逻辑上的错误
加下面这个补丁看看

```
diff --git a/build/linux/arm/make-Makefiles.bash b/build/linux/arm/make-Makefiles.bash
index 9dcbc7e8..1e891339 100755
--- a/build/linux/arm/make-Makefiles.bash
+++ b/build/linux/arm/make-Makefiles.bash
@@ -24,5 +24,6 @@ fi
 cmake -DCMAKE_BUILD_TYPE=Release \
       -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
       -DRKPLATFORM=ON \
+      -DHAVE_DRM=ON \
       -G "Unix Makefiles" \
       ${MPP_TOP}

```

liyuming1978, 2021-08-25:
我直接用 这个develop主线分支编译的， 这个补丁已经有了
# generate Makefile
cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
      -DRKPLATFORM=ON \
      -DHAVE_DRM=ON \
      -G "Unix Makefiles" \
      ${MPP_TOP}

liyuming1978, 2021-08-25:
确实 log 里面  有  mpp[9273]: mpp_rt: NOT found drm allocator ，   这个普通用户 需要设置哪个文件的权限啊 @HermanChen

liyuming1978, 2021-08-25:
搞明白了， 只需要设置3个驱动的权限就行， 不需要设置RAW_IO SYSTEM_ADMIN 2个权限

closed 2021-08-25 by liyuming1978