rockchip-linux/mpp #237: mpp decode H265

Allenhe123, 2021-09-07:
mpp解码h264，不同的相机都是OK的。但是解码h265的时候，只有部分相机可以成功，部分相机解码失败，同一套代码。
错误如下：
 mpp[11983]: H265D_PARSER: IS_IRAP frame found error

请问是mpp对h265的解码支持不够完善还是我们的代码写的有问题呢？
@HermanChen

Allenhe123, 2021-09-07:
想问一下h265编码后的视频数据没有随机切入点就无法解码吗？@HermanChen

HermanChen, 2021-09-07:
估计是 265 解码的时候 I 帧没送来？

Allenhe123, 2021-09-07:
同样的h265流数据，我们用其他播放器是可以解码并播放的（用的potplayer），也就是说软解码的方式没问题。
但是用mpp的硬件加速器解码就失败了。。

HermanChen, 2021-09-07:
麻烦提供下码流

Allenhe123, 2021-09-07:
https://github.com/Allenhe123/h265-stream
已经将两个h265码流上传到该仓库，一个是正常mpp可以解码的，一个是mpp解码出错的。

zinsayon, 2021-09-07:
解码报错的源，第一个I帧数据本身有异常，所以硬件解码会报错正常，但是到第二个gop后就正常解码，属于正常现象，如果觉得有问题，请参考mpi_dec_test 做测试

Allenhe123, 2021-09-07:
如果这个是正常现象的话，为什么mpp无法解码呢？
有没有办法可以跳过第一个gop，从第二个正常的gop开始解码吗？ @HermanChen @zinsayon

Allenhe123, 2021-09-07:
另外，我们就是用的mpi_dec_test做的测试，解码h265报错。

zinsayon, 2021-09-08:
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: cmd parse result:
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: input  file name: /sdcard/mpp-cannot-decode--file.265
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: output file name: 
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: config file name: 
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: width      : 1920
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: height     : 1080
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: type       : 16777220
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: debug flag : 0
08-13 13:58:05.566  1719  1719 I mpi_dec_utils: max frames : 0
08-13 13:58:05.567  1719  1719 I mpi_dec_test: mpi_dec_test start
08-13 13:58:05.567  1719  1719 I mpi_dec_test: input file size 5271359
08-13 13:58:05.567  1719  1719 I mpp_info: mpp version: c138e87 author: sayon.chen     2021-08-26 [h265e]: Add sao disable cfg
08-13 13:58:05.567  1719  1719 I mpi_dec_test: 0xec300550 mpi_dec_test decoder test start w 1920 h 1080 type 16777220
08-13 13:58:05.568  1719  1719 I mpp_rt  : found ion allocator
08-13 13:58:05.568  1719  1719 I mpp_rt  : found drm allocator
08-13 13:58:05.568  1719  1719 I mpp_rt  : use drm allocator for mpp_service
08-13 13:58:05.589  1719  1721 I mpp_log : device /dev/mpp_service select in vproc
08-13 13:58:05.592  1719  1719 I mpi_dec_test: 0xec300550 decode_get_frame get info changed found
08-13 13:58:05.592  1719  1719 I mpi_dec_test: 0xec300550 decoder require buffer w:h [1920:1080] stride [2304:1080] buf_size 4478976
08-13 13:58:05.599  1719  1721 I H265D_PARSER: IS_IRAP frame found error
08-13 13:58:05.602     0     0 I mpp_rkvdec2 fdf80200.rkvdec: resetting...
08-13 13:58:05.603     0     0 I mpp_rkvdec2 fdf80200.rkvdec: reset done
08-13 13:58:05.728  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 0
08-13 13:58:05.728  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 1
08-13 13:58:05.752  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 2
08-13 13:58:05.752  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 3
08-13 13:58:05.765  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 4
08-13 13:58:05.765  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 5
08-13 13:58:05.781  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 6
08-13 13:58:05.781  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 7
08-13 13:58:05.786  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 8
08-13 13:58:05.786  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 9
08-13 13:58:05.793  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 10
08-13 13:58:05.793  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 11
08-13 13:58:05.793  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 12
08-13 13:58:05.797  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 13
08-13 13:58:05.804  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 14
08-13 13:58:05.809  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 15
08-13 13:58:05.811  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 16
08-13 13:58:05.817  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 17
08-13 13:58:05.821  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 18
08-13 13:58:05.828  1719  1719 I mpi_dec_test: 0xec300550 decode get frame 19
这个是这边跑你提供码流出来结果，除第0帧硬件报错，后面每一帧都正常解码，这会造成第一个gop的yuv数据异常，但第二个gop开始数来yuv数据就是正常了，如果你那边跑出来现象不一致，建议更新一下mpp代码

zinsayon, 2021-09-08:
更新到develop最新

Allenhe123, 2021-09-08:
收到，谢谢

Allenhe123, 2021-09-08:
还有2个问题：
1. mpp程序更新了，需要更新配套的kernel，rga代码这些吗？
2. 我们在解码4k（3840x2160）的h264视频的时候，发现解码出的视频有拖拽现象，我录制了一个mp4视频放在https://github.com/Allenhe123/h265-stream/test.mp4，1080p和2k的不会有这个现象，请问这个是什么原因造成的呢？
@zinsayon

Allenhe123, 2021-09-09:
更新mpp的代码后，h265可以正常解码，谢谢。@zinsayon
可以帮分析一下https://github.com/Allenhe123/h265-stream下面的test.mp4视频现象的原因吗？ 
是解码4K图像时，哪个参数需要调整吗？

Allenhe123, 2021-09-10:
显示拖拽现象已经解决。
有一些疑问：
1。decode_simple函数内，put_packget成功后，get_decode_frame有时候会失败。
解码出的frame_count有时候是0，有时候是2，有时候是1，有时候4，5，3，6。。。不均匀。
这个是什么原因呢？ BP帧造成的吗？
2。我多个线程同时从多个camera读取流数据，每个线程都调用decode_simple(pkt)去解码，mpp的底层实现是如何的？
每个解码线程一个队列，共用解码器吗？
3。如果我在decode_simple里面判断frame_count > 0就 break整个解码循环，继续读取packet然后调用decode_simple进行解码。这样处理的话，frame_count就很加均匀，每次都是1，但是画面会出现拖拽现象。
  PS： frame_count不均匀的时候，我显示的时候只会用最后一帧显示； 但是frame_count改成均匀后就可以每帧都显示（有拖拽现象）。

spacet, 2021-09-10:
> 显示拖拽现象已经解决。
> 有一些疑问：
> 1。decode_simple函数内，put_packget成功后，get_decode_frame有时候会失败。
> 解码出的frame_count有时候是0，有时候是2，有时候是1，有时候4，5，3，6。。。不均匀。
> 这个是什么原因呢？ BP帧造成的吗？ [...]

请问显示”显示拖拽现象“是怎么解决的呢？

Allenhe123, 2021-09-10:
只要按照官方的示例来编码，就没有拖拽现象了。我们的程序有拖拽现象是因为我们想均匀解码。

spacet, 2021-09-10:
> 只要按照官方的示例来编码，就没有拖拽现象了。我们的程序有拖拽现象是因为我们想均匀解码。

感谢

Allenhe123, 2021-09-10:
[image]

@zinsayon
h265解码4k（3840x2160）是没有问题的；但是解码1080p与2k（2688x1520）分辨率的h265的时候，解码可以成功，但是解码后的NV12图像显示是花屏。。。。请问这是咋回事？

mo123, 2021-09-20:
I can't play any H265 videos on Android 11 RK356x if the height is not divisible by 16.
Lots of videos width is 1920 but height is only divisible by 2 or 8(height is between 800 to 1280).
Can this be fixed?
@HermanChen