rockchip-linux/mpp #242: onshot模式编码drm buf为JPEG疑似内存泄漏

xukefang, 2021-09-10:
你好，我想将nv12 buf编码成jpg图片，使用drm将buf包装后，通过MppBufferInfo 拿到Mpp buffer，再通过mpp_buffer_import拿到Mpp Frame，然后使用encode拿到编码后的packet，循环运行时发现进程本身内存没有增长，但是系统内存使用率不断增高。是不是哪一步操作不对，导致内存没有正确释放？

示例代码：
    // buf ==》drm Buf
    MppBufferInfo bufInfo;
    bufInfo.size = frameSize;
    bufInfo.ptr = drmBuf;
    bufInfo.fd = bufFd;
    bufInfo.type = MPP_BUFFER_TYPE_DRM;

    MppBuffer frmBuf{nullptr};
    mpp_buffer_import(&frmBuf, &bufInfo);
    mpp_frame_set_buffer(_inFrame, frmBuf);
   _mpi->encode(_ctx, _inFrame, &_outPacket);
    // 释放
    mpp_packet_deinit(&_outPacket);
    mpp_frame_deinit(&_inFrame);
    if (frmBuf) {
        mpp_buffer_put(frmBuf);
        frmBuf = nullptr;
    }

    //destory frm
    drm_buf_destroy(&drmCtx, drmFd, bufFd, handle, drmBuf, actualSize);
    drm_deinit(&drmCtx, drmFd);

程序结束示例：
mpp[30662]: mpp_buffer: ~MppBufferService cleaning misc group

HermanChen, 2021-09-11:
可以在支持中 ls -l 下 /proc/进程号/fd/，看下这个进程使用的 fd 有没有在不断增加。

xukefang, 2021-09-13:
fd并没有持续增加，经过多次排查，内存会持续增长到200M左右停止增加，处理5 fps的nv12格式的1080p图片的情况下每秒增加大约15M的内存，直到大约200M左右的时候停止增加

HermanChen, 2021-09-28:
这个估计是内核有做 dmabuf 使用的 cache 导致的，可以通过 echo 1 > /proc/mpp_service/vdpu/session_buffers 来修改 cache 数量。