rockchip-linux/mpp #245: MPP解码表现异常

XingP, 2021-09-16:
在SDK（1.1.3）中得到了MPP与gstreamer-rockchip的编译产物，并在在RK3568开发板上进行了配置，可以通过gstreamer命令查看到插件信息
```
    linaro@linaro-alip:~$ gst-inspect-1.0 rockchipmpp
Plugin Details:
  Name                     rockchipmpp
  Description              Rockchip Mpp Video Plugin
  Filename                 /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstrockchipmpp.so
  Version                  1.10.0
  License                  LGPL
  Source module            gstreamer-rockchip
  Binary package           GStreamer Plug-ins source release
  Origin URL               Unknown package origin
 
  mppvideodec: Rockchip's MPP video decoder
  mpph264enc: Rockchip Mpp H264 Encoder
  mppvp8enc: Rockchip Mpp VP8 Encoder
  mppjpegenc: Rockchip Mpp JPEG Encoder
  mppjpegdec: Rockchip's MPP JPEG image decoder
 
  5 features:
  +-- 5 elements
 
```
此时调用gstreamer（APT方式安装，1.14.4）进行视频解码，有MPP相关打印出现：
VP8解码时，
Rga built version:274b345 
mpp[23285]: mpp_rt: NOT found ion allocator 
mpp[23285]: mpp_rt: found drm allocator 
mpp[23285]: mpp_info: mpp version: 49f29006 author: Jeffy Chen 2021-08-04 [drm]: Add mmap flag detection 
mpp[23285]: mpp_buf_slot: set frame info: w 0 h 0 hor 0 ver 0 
mpp[23285]: mpp_dec: setting default w 0 h 0 h_str 0 v_str 0
 
此时视频可播放，但**CPU一核100%占用，其余三核基本空闲**。
通过 _echo 0x0100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug_ 打开调试打印，可以看到如下打印，但 _cat /sys/class/devfreq/fde60000.gpu/load_  查看到的值始终为0@200000000Hz
```
[27936.502408] fdea0400.vdpu: pid: 23422, session: 000000009b78f8ee, time: 12555 us 
[27936.520128] fdea0400.vdpu: pid: 23422, session: 000000009b78f8ee, time: 12506 us 
[27937.174173] fdea0400.vdpu: pid: 23422, session: 000000009b78f8ee, time: 16783 us 
[27937.197021] fdea0400.vdpu: pid: 23422, session: 000000009b78f8ee, time: 14977 us 
[27937.219011] fdea0400.vdpu: pid: 23422, session: 000000009b78f8ee, time: 16094 us 
```
……
 
H264解码时，
```
mpp[23285]: mpp_info: mpp version: 49f29006 author: Jeffy Chen 2021-08-04 [drm]: Add mmap flag detection 
mpp[23285]: h264d_api: is_avcC=1 mpp[23285]: hal_h264d_vdpu34x: control info: fmt 7, w 1920, h 1008 
mpp[23285]: mpp_buf_slot: set frame info: w 1920 h 1008 hor 1920 ver 1008 
mpp[23285]: mpp_dec: setting default w 1920 h 1008 h_str 1920 v_str 1008 
mpp[23285]: h264d_sps: ERROR: Not support chroma_format_idc=3.
```
 
查到对应MPP源码_h264d_sps.c_中的错误输出，应该是视频格式不支持，此时未解码，四核CUP都空闲。
 
移除gstreamer-rockchip编译产物后进行对比测试，
linaro@linaro-alip:~$ gst-inspect-1.0 rockchipmpp 
No such element or plugin 'rockchipmpp'
 
没有gstreamer-rockchip插件时，应该为gstreamer调用CPU软解，VP8与H264均可成功解码播放，此时有1-2核基本100%占用，其余核也处于高占用状态。
 
上面用cat /sys/class/devfreq/fde60000.gpu/load 或 MPP调试打印的方法来判断是否调用到了硬解，结果不一致，那种判断方式更可靠呢？而综合来看，在MPP与gstreamer-rockchip配合的情况下，gstreamer视频解码应该是有通过该插件进入了MPP的处理流程，但其表现却并不符合预期，请问是哪里出了问题？
此外，之前用mpi_dec_test/mpi_enc_test的Demo已经是可以进行图像的编解码的，但是后来偶然发现mpi_test会打印错误信息，请问这是正常的吗？
```
linaro@linaro-alip:~$ mpi_test
mpp[3632]: mpi_test: mpi_test start
mpp[3632]: mpi_test: mpi_test decoder test start
mpp[3632]: mpi: mpp_create enter ctx 0x7ffae5fad0 mpi 0x7ffae5fad8
mpp[3632]: mpp_info: mpp version: 49f29006 author: Jeffy Chen    2021-08-04 [drm]: Add mmap flag detection
mpp[3632]: mpi: mpp_create leave ret 0 ctx 0xccc9900 mpi 0x7fb728ddf8
mpp[3632]: mpi: mpp_init enter ctx 0xccc9900 type 0 coding 0
mpp[3632]: mpp: unable to create dec unused for soc rk3568 unsupported
mpp[3632]: mpi: mpp_init leave ctx 0xccc9900 ret -1
mpp[3632]: mpi_test: mpp_init failed
mpp[3632]: mpi: mpp_destroy enter ctx 0xccc9900
mpp[3632]: mpi: mpp_destroy leave ctx 0xccc9900 ret 0
mpp[3632]: mpi_test: mpi_test failed
```

HermanChen, 2021-09-16:
这个问题会在 redmine 上支持，麻烦提供下 redmine 号并关闭问题

XingP, 2021-09-16:
[FAE 项目-Defect #315777] 
【RK3568】 gstreamer与MPP硬件解码的相关问题

请顺便redmine 上也回复一下如何判断硬件解码的问题，谢谢！

closed 2021-09-16 by XingP