rockchip-linux/mpp #248: mpp H264 decode success, but H265 decode failed

wjkxgfw, 2021-09-22:
mpp[1633]: mpp_info: mpp version: 3fedc6cd author: zouxf         2020-10-23 Merge remote-tracking branch 'firefly-linux/firefly'
mpp[1633]: mpp_rt: NOT found ion allocator
mpp[1633]: mpp_rt: found drm allocator
mpp[1633]: mpp_device: mpp_device_init failed to find device for coding 16777220 type 0
mpp[1633]: mpp: deprecated block control, use timeout control instead
mpp[1633]: H265D_PARSER: extradata is encoded as hvcC format
mpp[1633]: mpp: deprecated block control, use timeout control instead
mpp[1633]: mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 9 Bad file descriptor
mpp[1633]: H265HAL: RK_HEVC_DEC: ERROR: mpp_device_send_reg Failed!!!
mpp[1633]: mpp_device: mpp_device_send_reg ioctl VPU_IOC_SET_REG failed ret -1 errno 9 Bad file descriptor
mpp[1633]: H265HAL: RK_HEVC_DEC: ERROR: mpp_device_send_reg Failed!!!

HermanChen, 2021-09-23:
Please check the kernel driver version.
Run mpp_platform_test and show the log.

wjkxgfw, 2021-09-23:
mpp[8830]: mpp_plat_test: kernel version: 4.4
mpp[8830]: mpp_plat_test: ioctl  version: vcodec_service
mpp[8830]: mpp_plat_test: 
mpp[8830]: mpp_plat_test: chip name: firefly,roc-rk3399-pc rockchip,rk3399
mpp[8830]: mpp_plat_test: 
mpp[8830]: mpp_plat_test: chip vcodec type 0004020a
mpp[8830]: mpp_plat_test: found vpu2 codec
mpp[8830]: mpp_plat_test: found rkvdec decoder
mpp[8830]: mpp_plat_test: found vpu2 encoder
mpp[8830]: mpp_plat_test: 
mpp[8830]: mpp_plat_test: start probing decoder device name:
mpp[8830]: mpp_plat_test: H.264 decoder: (null)
mpp[8830]: mpp_plat_test: H.265 decoder: (null)
mpp[8830]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp[8830]: mpp_plat_test: VP9   decoder: (null)
mpp[8830]: mpp_plat_test: avs   decoder: /dev/vpu_service
mpp[8830]: mpp_plat_test: 
mpp[8830]: mpp_plat_test: start probing encoder device name:
mpp[8830]: mpp_plat_test: H.264 encoder: /dev/vpu_service
mpp[8830]: mpp_plat_test: H.265 encoder: /dev/vpu_service
mpp[8830]: mpp_plat_test: MJPEG encoder: /dev/vpu_service
mpp[8830]: mpp_plat_test: mpp platform test done

wjkxgfw, 2021-09-23:
mpi_dec_test -t 7 -i Tennis1080p.h264 -n 10             

decode h264 failed

mpp[8295]: mpi_dec_utils: cmd parse result:
mpp[8295]: mpi_dec_utils: input  file name: Tennis1080p.h264
mpp[8295]: mpi_dec_utils: output file name: 
mpp[8295]: mpi_dec_utils: config file name: 
mpp[8295]: mpi_dec_utils: width      :    0
mpp[8295]: mpi_dec_utils: height     :    0
mpp[8295]: mpi_dec_utils: type       : 7
mpp[8295]: mpi_dec_utils: debug flag : 0
mpp[8295]: mpi_dec_utils: max frames : 10
mpp[8295]: mpi_dec_test: mpi_dec_test start
mpp[8295]: mpi_dec_test: input file size 10786528
mpp[8295]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[8295]: mpi_dec_test: 0x556557e7a0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[8295]: mpp_rt: NOT found ion allocator
mpp[8295]: mpp_rt: found drm allocator
mpp[8295]: vcodec_service: open vcodec_service (null) failed
mpp[8295]: hal_h264d_api: mpp_dev_init failed ret: -1
mpp[8295]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[8295]: mpp_hal: mpp_hal_init could not found coding type 7
mpp[8295]: mpp_dec: mpp_dec_init could not init hal
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[8295]: mpp: error found on mpp initialization
mpp[8295]: mpi_dec_test: 0x556557e7a0 mpp_init failed
mpp[8295]: mpi_dec_test: test failed ret -1
mpi_dec_test[8295]: mpp_mem_pool: put_pool found 1 used buffer size 80
mpi_dec_test[8295]: mpp_mem_pool: put_pool found 4 used buffer size 192

gst-launch-1.0 filesrc location=/usr/local/test.mp4 ! qtdemux ! h264parse ! mppvideodec ! rkximagesink
decode h264 success

HermanChen, 2021-09-23:
Please check the dts in kernel. There are two codecs on RK3399. /dev/vpu_service for 1080p encoder + decoder and /dev/rkvdec for 4K decoder. But the mpp_platform_test log shows there is only /dev/vpu_service available.

wjkxgfw, 2021-09-23:
The problem has been solved.
Thank you very much for your reply!

closed 2021-09-23 by wjkxgfw

sukesh-g, 2022-05-28:
hai @wjkxgfw , @HermanChen am facing the same kind of issue, can you please guide to me to solve the issue.

am encountering below issue while using rtsp gstremer.

mpp[7274]: mpp_info: mpp version: 41c9c809 author: sayon.chen    2020-12-24 [mpp2str]: Add rc_gop_mode 2 str
mpp[7274]: mpp_rt: NOT found ion allocator
mpp[7274]: mpp_rt: found drm allocator
mpp[7274]: mpp: deprecated block control, use timeout control instead
mpp[7274]: h264d_api: is_avcC=1
mpp[7274]: mpp: deprecated block control, use timeout control instead
mpp[7274]: h264d_api: is_avcC=1

Thanks,
sukesh