rockchip-linux/mpp #249: rockpie 3328上运行mpi_enc_test，死机

tongshi203, 2021-09-28:
你好，
   rockpie上运行解码程序：./mpi_dec_test -i out.h264 -w 1920 -h 1080 -t 7 -o 1080.yuv 是正确的
  但是运行编码程序，就会导致rockpie死机，运行如下：
root@rockpie:/home/rock/mpp_rockchip/build/linux/aarch64/test# ./mpi_enc_test -i 1080.yuv -w 1920 -h 1080 -t 7 -o 1080.h264
mpp[913]: mpi_enc_utils: cmd parse result:
mpp[913]: mpi_enc_utils: input  file name: 1080.yuv
mpp[913]: mpi_enc_utils: output file name: 1080.h264
mpp[913]: mpi_enc_utils: width      : 1920
mpp[913]: mpi_enc_utils: height     : 1080
mpp[913]: mpi_enc_utils: format     : 0
mpp[913]: mpi_enc_utils: type       : 7
mpp[913]: mpi_enc_test: mpi_enc_test start
mpp[913]: mpp_rt: NOT found ion allocator
mpp[913]: mpp_rt: found drm allocator
mpp[913]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[913]: mpi_enc_test: 0x55a5f5f8c0 mpi_enc_test encoder test start w 1920 h 1080 type 7
mpp[913]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7776000 [486000 : 8262000] fps [30:30] gop 60
mpp[913]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1080]
mpp[913]: mpp_enc: send header for set cfg change input/format 
mpp[913]: mpp_enc: mode vbr bps [486000:7776000:8262000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]

HermanChen, 2021-09-28:
没看到崩溃log啊

tongshi203, 2021-09-28:
通过ssh登陆到rockpie上运行编码的，一运行到这个地方就死机了，不知道怎么看log

附rockpie上mpp编译过程：
git clone https://github.com/rockchip-linux/mpp mpp_rockchip;
cd mpp_rockchip/build/linux/aarch64;
bash make-Makefiles.bash;
make;
cd mpp_rockchip/build/linux/aarch64/test;
运行编解码

tongshi203, 2021-09-28:
请问我有操作不正确的地方吗？

HermanChen, 2021-09-28:
运行下 mpp_platform_test 和 mpp_runtime_test 看下

tongshi203, 2021-09-28:
root@rockpie:/home/rock/mpp_rockchip/build/linux/aarch64/osal/test# ./mpp_platform_test 
mpp[839]: mpp_plat_test: kernel version: 4.4
mpp[839]: mpp_plat_test: ioctl  version: vcodec_service
mpp[839]: mpp_plat_test: 
mpp[839]: mpp_plat_test: chip name: radxa,rockpie rockchip,rk3328
mpp[839]: mpp_plat_test: 
mpp[839]: mpp_plat_test: chip vcodec type 0104020a
mpp[839]: mpp_plat_test: found vpu2 codec
mpp[839]: mpp_plat_test: found rkvdec decoder
mpp[839]: mpp_plat_test: found vpu2 encoder
mpp[839]: mpp_plat_test: found h265 stand-alone encoder
mpp[839]: mpp_plat_test: 
mpp[839]: mpp_plat_test: start probing decoder device name:
mpp[839]: mpp_plat_test: H.264 decoder: /dev/rkvdec
mpp[839]: mpp_plat_test: H.265 decoder: /dev/rkvdec
mpp[839]: mpp_plat_test: MJPEG decoder: /dev/vpu_service
mpp[839]: mpp_plat_test: VP9   decoder: /dev/rkvdec
mpp[839]: mpp_plat_test: avs   decoder: /dev/vpu_service
mpp[839]: mpp_plat_test: 
mpp[839]: mpp_plat_test: start probing encoder device name:
mpp[839]: mpp_plat_test: H.264 encoder: /dev/vepu
mpp[839]: mpp_plat_test: H.265 encoder: /dev/h265e
mpp[839]: mpp_plat_test: MJPEG encoder: /dev/vepu
mpp[839]: mpp_plat_test: mpp platform test done
root@rockpie:/home/rock/mpp_rockchip/build/linux/aarch64/osal/test# ./mpp_runtime_test
mpp[840]: mpp_rt: NOT found ion allocator
mpp[840]: mpp_rt: found drm allocator
mpp[840]: mpp_rt_test: mpp found ion buffer is invalid
mpp[840]: mpp_rt_test: mpp found drm buffer is valid

tongshi203, 2021-09-28:
mpp[839]: mpp_plat_test: H.264 encoder: /dev/vepu 
为什么rk3328 编码不用/dev/vpu_service，这两个有什么区别？

HermanChen, 2021-09-28:
rk3328 有几个器件，内核驱动写成不同的设备：
/dev/vpu_service 是 1080p 规格的解码器，包含 jpeg, h263, h264, mpeg2/4 这些
/dev/vepu 是 1080p 规格的编码器，包含 h264, vp8, jpeg
/rkvdec 是 4K 规格的解码器，包含 h264, h265, vp9

tongshi203, 2021-09-28:
我刚在rk3399上测试了相同的编码，是ok的。
运行./mpp_platform_test ，发现它的 编码用的是mpp[3025]: mpp_plat_test: H.264 encoder: /dev/vpu_service。
是不是rk3328 cpu性能影响编码呢？