rockchip-linux/mpp #253: mpp encoding YUYV to H264 / H265 (RK3568)

avafinger, 2021-10-09:
Encoding a camera output /dev/video0 (YUYV) to H264 or H265 get lots of horizontal trashed lines as in the frame sample.
[image]

How can i get rid of the noise?

testing with:
`mpi_enc_test -i /dev/video0 -f 8 -w 1280 -h 720 -o out_1280x720.h264 -t 7 -n 150 -d 1`

avafinger, 2021-10-09:
I found out if i encode only one frame I don't get broken horizontal lines.

HermanChen, 2021-10-12:
Which chip are you using?

avafinger, 2021-10-12:
Chip: RK3568
Kernel: 4.19

avafinger, 2021-10-13:
@HermanChen 

I would like to add this info to not confuse you (there are two different issues):

1. Grabbing YUV (200 frames) and encode it to H264 / H265 works only for the first 4 frames, the remaining frames are corrupt with that horizontal lines, this is on RK3568, camera is USB, YUV,H264 and MJPEG
`sudo mpi_enc_test -i /dev/video0 -f 8 -w 1280 -h 720 -o out_200F_1280x720.h264 -t 7 -n 200`
2. Grabbing YUV (200 frames) and encode it to H264 / H265 works fine for the 200 frames on RK3399 (same kernel as 3568), camera is USB, YUV,H264 and MJPEG. I have here two ov4689 and one USB attached. /dev/video0 is ov4689 and /dev/video10 is USB.
`sudo mpi_enc_test -i /dev/video10 -f 8 -w 1280 -h 720 -o out_200F_1280x720.h264 -t 7 -n 200`
3. Grabbing NV12 (200 frames) and encode it to H264 / H265 breaks the kernel on RK3399, this is on RK3399 with the same kernel used in RK3568, camera is OVUSB, YUV,H264 and MJPEG
```
sudo mpi_enc_test -i /dev/video0 -f 0 -w 1920 -h 1080 -o out_1frames_1920x1080.h264 -t 7 -n 2
[sudo] password for ubuntu: 
mpp[642]: mpi_enc_utils: cmd parse result:
mpp[642]: mpi_enc_utils: input  file name: /dev/video0
mpp[642]: mpi_enc_utils: output file name: out_1frames_1920x1080.h264
mpp[642]: mpi_enc_utils: width      : 1920
mpp[642]: mpi_enc_utils: height     : 1080
mpp[642]: mpi_enc_utils: format     : 0
mpp[642]: mpi_enc_utils: type       : 7
mpp[642]: mpi_enc_test: mpi_enc_test start
mpp[642]: mpi_enc_test: open camera device
mpp[642]: camera_source: get width 1920 height 1080
mpp[642]: camera_source: get dma buf(0)-fd: 5
mpp[642]: mpp_rt: NOT found ion allocator
mpp[642]: mpp_rt: found drm allocator
mpp[642]: camera_source: get dma buf(1)-fd: 8
mpp[642]: camera_source: get dma buf(2)-fd: 9
mpp[642]: camera_source: get dma buf(3)-fd: 10
mpp[642]: mpi_enc_test: new framecap ok
mpp[642]: mpp_info: mpp version: 6cc2ef5f author: Herman Chen   2021-09-17 [mpp_list]: Add list_mode and list_move_tail
mpp[642]: mpi_enc_test: 0x558eaef060 mpi_enc_test encoder test start w 1920 h 1080 type 7
mpp[642]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7776000 [486000 : 8262000] fps [30:30] gop 60
mpp[642]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1920:1080] stride [1920:1080]
mpp[642]: mpp_enc: send header for set cfg change input/format 
mpp[642]: mpp_enc: mode vbr bps [486000:7776000:8262000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
mpp[642]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 110 Connection timed out
mpp[642]: hal_h264e_vepu2_v2: hal_h264e_vepu2_wait_v2 poll cmd failed 110
mpp[642]: mpi_enc_test: 0x558eaef060 encoded frame 0    size 200    
mpp[642]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 110 Connection timed out
mpp[642]: hal_h264e_vepu2_v2: hal_h264e_vepu2_wait_v2 poll cmd failed 110
mpp[642]: mpi_enc_test: 0x558eaef060 encoded frame 1    size 0      
mpp[642]: mpi_enc_test: 0x558eaef060 encode max 2 frames
mpp[642]: mpi_enc_test: 0x558eaef060 mpi_enc_test success total frame 2 bps 24000
mpp[642]: mpp_buffer: ~MppBufferService cleaning misc group

```

Kernel halt with:
[image]

Obs: kernel used is rock solid, i can compile kernels on the board and heavy stuff, not a single problem so far.

HermanChen, 2021-10-14:
We can not help much on the camera or sensor part...
The camera_source in mpp is just a demo and may not work for all the platform.
I think it is generally a sdk / environment issue.

HermanChen, 2021-10-18:
the buffer size is for decoder or encoder?

philfleck, 2023-10-02:
Hi @avafinger, Hi @HermanChen 
could you resolve the issue (artifacts when encoding video from eg uvccamera with YUYV)? I am facing a similar problem but on the RK3588 (nanopc t6) and the RK3568B2.

Any help appreciated
best p

avafinger, 2023-10-02:
Can you share your command? I will check if i can reproduce here.

avafinger, 2023-10-02:
I can only speculate, i think the problem is with RGA conversion with dmabuf.
See this comment here: https://github.com/rockchip-linux/mpp/issues/356#issuecomment-1427641674
The reverse may be true.

If not using dmabuf you face this error:
```
 RgaBlit(1465) RGA_BLIT fail: Invalid argument
 RgaBlit(1466) RGA_BLIT fail: Invalid argument
fd-vir-phy-hnd-format[0, 0x7f8b3ce000, (nil), 0, 0]
rect[0, 0, 640, 480, 1280, 480, 65536, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[29, (nil), (nil), 0, 0]
rect[0, 0, 640, 480, 1280, 480, 65536, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail

```

with dmabuf (render):
[image]

without dmabuf (render):
[image]

Unfortunately, i can't do anything about it.

Maybe Rockchip ( @HermanChen ) can..

avafinger, 2023-10-02:
You should try with different window sizes, i know there is/are problems with alignment / stride with RGA.
The camera i tested here on rk3588 is limited to 640x480.

philfleck, 2023-10-03:
hi thanks for the input,

i used something among this line, without setting strides. i tested with an Microsoft lifecam 720p@30fps

`mpi_enc_test -i /dev/video0 -f 8 -w 1280 -h 720 -o out_1280x720.h264 -t 7`

best p

philfleck, 2023-10-03:
btw, the artefacts appear across all resolutions ...

avafinger, 2023-10-03:
I'll leave it open.
Maybe someone has found a workaround and can post their findings here...

philfleck, 2023-10-05:
> You should use `ioctl(DMA_BUF_IOCTL_SYNC)` to synchronize the cacheable DMA-BUF before accessing it, and call it again after accessing it to invalidate the cache.
> 
> https://github.com/airockchip/librga/blob/fb3357d09008222bc5e27bdaadf74a0c5ea4c86e/samples/allocator_demo/src/rga_allocator_dma_cache_demo.cpp
> 
> https://github.com/airockchip/librga/blob/fb3357d09008222bc5e27bdaadf74a0c5ea4c86e/samples/utils/allocator/dma_alloc.cpp#L81
> 
> https://www.kernel.org/doc/html/v5.9/driver-api/dma-buf.html#cpu-access-to-dma-buffer-objects

is this shown within the mpi_enc_test examples?

nyanmisaka, 2023-10-05:
> is this shown within the mpi_enc_test examples?

Nope, but it's a part of the librga sample. IMHO mpi_enc_test should not be responsible for this. It assumes that the user's input is a normal file rather than a device file.

HermanChen, 2023-10-07:
This error seems a cache issue. Any cpu copy on the image?

avafinger, 2023-10-07:
Most likely there is. The issue with YUYV happens on mpi_enc_test,
gstreamer ***@***.***), ffplay (rendering), SDL (rendering with texture).

gstreamer has DMA_BUF_IOCTL_SYNC, maybe not used when YUYV.
Any possible explanation for not happening with NV12?

BTW, i quit github, so not sure this reply can reach the destination....
But i can still follow the conversation.
Please, feel free to close this issue here if completed.

HermanChen, 2023-10-07:
Check the stream from mpi_enc_test first, if there is the lines on the stream then it is an error on copying frame before mpi_enc_test. If there is no lines in the steam then it is the decoding and render issue.

HermanChen, 2023-10-07:
How to reach you by email? @avafinger

avafinger, 2023-10-07:
alex dot mobigo dot gmail dot com
Thanks!

avafinger, 2023-10-07:
You can reproduce it:

Rendering only is fine :
GST_GL_X11_NO_DECORATIONS=1 DISPLAY=:0.0 gst-launch-1.0 v4l2src
device=/dev/video22 io-mode=2 !
'video/x-raw,format=YUY2,width=640,height=480' ! glimagesink
render-rectangle='<0,120,640,480>'

Rendering only has artifacts (with dma-buf):
GST_GL_X11_NO_DECORATIONS=1 DISPLAY=:0.0 gst-launch-1.0 v4l2src
device=/dev/video22 ! 'video/x-raw,format=YUY2,width=640,height=480' !
glimagesink render-rectangle='<0,120,640,480>'

Rendering only is fine (no dma-buf):
DISPLAY=:0.0 ffplay -f v4l2 -framerate 30 -pixel_format yuyv422 -video_size
640x480 -i /dev/video22 -top 0 -left 1920 -noborder

Recording with mpp and playing back on X86 has artifacts:
(mpp demo) mpi_enc_test -i /dev/video22 -f 8 -w 640 -h 480 -o
out_640x480.h264 -t 7 -n 850 -d 1
(gstreamer)  GST_GL_X11_NO_DECORATIONS=1 DISPLAY=:0.0 gst-launch-1.0
v4l2src device=/dev/video22 !
'video/x-raw,format=YUY2,width=640,height=480' ! mpph264enc ! filesink
location=test_640x480_30fps_h264.mp4

'
v4l2-ctl --all -d /dev/video22
Driver Info:
Driver name      : uvcvideo
Card type        : Microsoft LifeCam VX-800: Micro
Bus info         : usb-fc880000.usb-1
Driver version   : 5.10.110
Capabilities     : 0x84a00001
Video Capture
Metadata Capture
Streaming
Extended Pix Format
Device Capabilities
Device Caps      : 0x04200001
Video Capture
Streaming
Extended Pix Format
Media Driver Info:
Driver name      : uvcvideo
Model            : Microsoft LifeCam VX-800: Micro
Serial           :
Bus info         : usb-fc880000.usb-1
Media version    : 5.10.110
Hardware revision: 0x00000101 (257)
Driver version   : 5.10.110
Interface Info:
ID               : 0x03000002
Type             : V4L Video
Entity Info:
ID               : 0x00000001 (1)
Name             : Microsoft LifeCam VX-800: Micro
Function         : V4L2 I/O
Flags            : default
Pad 0x01000007   : 0: Sink
 Link 0x02000010: from remote pad 0x100000a of entity 'Processing 2' (Video
Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
Width/Height      : 640/480
Pixel Format      : 'YUYV' (YUYV 4:2:2)
Field             : None
Bytes per Line    : 1280
Size Image        : 614400
Colorspace        : Default
Transfer Function : Default (maps to Rec. 709)
YCbCr/HSV Encoding: Default (maps to ITU-R 601)
Quantization      : Default (maps to Limited Range)
Flags             :
Crop Capability Video Capture:
Bounds      : Left 0, Top 0, Width 640, Height 480
Default     : Left 0, Top 0, Width 640, Height 480
Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height
480, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480,
Flags:
Streaming Parameters Video Capture:
Capabilities     : timeperframe
Frames per second: 30.000 (30/1)
Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-10 max=10 step=1
default=2 value=2
                       contrast 0x00980901 (int)    : min=0 max=20 step=1
default=10 value=10
                     saturation 0x00980902 (int)    : min=0 max=10 step=1
default=4 value=4
                            hue 0x00980903 (int)    : min=-5 max=5 step=1
default=0 value=0
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                          gamma 0x00980910 (int)    : min=100 max=200
step=1 default=130 value=130
                           gain 0x00980913 (int)    : min=32 max=48 step=1
default=34 value=34
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2
value=2 (60 Hz)
0: Disabled
1: 50 Hz
2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500
step=1 default=6500 value=6500 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=10 step=1
default=4 value=4
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1
default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3
value=3 (Aperture Priority Mode)
1: Manual Mode
3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=8 max=16384
step=1 default=512 value=512 flags=inactive

'

avafinger, 2023-10-07:
Forgot to mention mpv, just rendering the camera output is fine with my
setup (most likely no dma-buf):

DISPLAY=:0.0 mpv --hwdec=rkmpp --vo=gpu --gpu-api=opengl --opengl-es=yes
--gpu-context=x11egl --drm-draw-plane=overlay
--drm-drmprime-video-plane=primary --drm-device=/dev/dri/card0
--demuxer-lavf-format=video4linux2
--demuxer-lavf-o-set=input_format=yuyv422 av://v4l2:/dev/video22
--loop-file=inf --no-border --audio-device=alsa:device=rockchip-hdmi0 -v

hbiyik, 2023-11-15:
now i have the same issue, i have not investigated fully but here are my observations with rk3588

1) it happens when using decoder with TIMEOUT_NON_BLOCKING with external dma32 buffers, i have a different architecture where there is TIMOUT_BLOCKING with internal buffers is used when decoding, it does not happen there. (i do not know which is relevant)
2) in my case it also happen when i do not use rga as well, so pure decoder output
3) it happens with mpeg1/2 or AV1 decoders does not happen with h264 or hevc decoder. (at least in my code)

I also have a feeling it is some form of synronization issue somewhere i will dig into it.

avafinger, 2023-11-15:
In all situations, i described reproducing the issue *rga* was involved (
*fd* - dma32-buf), it is interesting you one without rga.
Do you have a command line example to reproduce your item 3.?

hbiyik, 2023-11-16:
always happens when i play this file: https://github.com/hbiyik/FFmpeg/files/13342554/video_green.mkv.zip

typically with ffplay. i really think it is a dmabuf sync issue.

avafinger, 2023-11-16:
I don't think it is exactly the same problem for this specific issue but it
is related to sync.
I can play video_green.mkv with Jeffy's ffplay (with rga ioctl) and Nyam's
modified jellyfin ffplay, but not with your branch FFmpeg-exp_refactor_all
and FFmpeg-encoder (the old one).
The ffplay modification is just to render with  SDL2 (x11 and bsp)  and the
rga call with ioctl and not librga.

What i have noticed is your old branch FFmpeg-encoder displayed a still
image after a while (only one frame i think the last), suggesting the sync
issue.
I used -loop 0 command line parameter for an infinite loop.

[image: ffmpeg.png]

avafinger, 2023-11-16:
Looks like the ones that worked were soft-decoded.

hbiyik, 2023-11-16:
Yeah, this was a bad example dye to another [bug](https://github.com/hbiyik/FFmpeg/issues/35), ffplay gets stuck after 4th frame in this file.

I remembered now running `mpv --hwdec=rkmpp video_green.mkv` should reproduce it, like expkained [here](https://github.com/hbiyik/FFmpeg/issues/37)

But you are right, even mpv uses drmprime to render in this case ffmpeg decoder is using rga to convert 16byte aligned strides to 64, to [satisfy panforks egl](https://github.com/hbiyik/FFmpeg/issues/37).

So you are right in my case it is also rga.

avafinger, 2023-11-17:
That's right, mpv worked here.

avafinger, 2023-11-17:
Testing  HD 720P camera and mpv.
Would anyone know how to force an hw conversion (mpv yuyv422 -> yuv422p)
and prove my point?
Below is the command line that works but is soft conversion.

***@***.***:~$ DISPLAY=:0.0 mpv --hwdec=rkmpp --vo=gpu --gpu-api=opengl
--opengl-es=yes --gpu-context=x11egl --drm-draw-plane=overlay
--drm-drmprime-video-plane=primary --drm-device=/dev/dri/card0
--demuxer-lavf-format=video4linux2
--demuxer-lavf-o-set=input_format=yuyv422 av://v4l2:/dev/video22
--loop-file=inf --no-border --audio-device=alsa:device=rockchip-hdmi0 -v
[cplayer] Command line options: '--hwdec=rkmpp' '--vo=gpu'
'--gpu-api=opengl' '--opengl-es=yes' '--gpu-context=x11egl'
'--drm-draw-plane=overlay' '--drm-drmprime-video-plane=primary'
'--drm-device=/dev/dri/card0' '--demuxer-lavf-format=video4linux2'
'--demuxer-lavf-o-set=input_format=yuyv422' 'av://v4l2:/dev/video22'
'--loop-file=inf' '--no-border' '--audio-device=alsa:device=rockchip-hdmi0'
'-v'
[cplayer] mpv 0.34.1 Copyright © 2000-2021 mpv/MPlayer/mplayer2 projects
[cplayer]  built on UNKNOWN
[cplayer] FFmpeg library versions:
[cplayer]    libavutil       56.70.100
[cplayer]    libavcodec      58.134.100
[cplayer]    libavformat     58.76.100
[cplayer]    libswscale      5.9.100
[cplayer]    libavfilter     7.110.100
[cplayer]    libswresample   3.9.100
[cplayer] FFmpeg version: 4.4.2-0ubuntu0.22.04.1
[cplayer]
[cplayer] Configuration: ./waf configure --prefix=/usr
--libdir=/usr/lib/aarch64-linux-gnu --confdir=/etc/mpv
--zshdir=/usr/share/zsh/vendor-completions --enable-cdda --enable-dvdnav
--enable-libmpv-shared --enable-sdl2 --disable-build-date --enable-dvbin
[cplayer] List of enabled features: alsa asm caca cdda cplayer cplugins
cuda-hwaccel cuda-interop debug-build drm dvbin dvdnav egl egl-drm
egl-helpers egl-x11 ffmpeg ffmpeg-aviocontext-bytes-read ffnvcodec gbm
gbm.h gl gl-wayland glibc-thread-name glob glob-posix gpl iconv jack
javascript jpeg lcms2 libarchive libass libavdevice libbluray libdl libm
libmpv-shared libplacebo librt linux-fstatfs linux-input-event-codes lua
lua52 memfd_create optimize plain-gl posix posix-or-mingw pthreads pulse
rubberband sdl2 sdl2-audio sdl2-gamepad sdl2-video sixel spirv-cross
spirv-cross-shared stdatomic uchardet vaapi vaapi-drm vaapi-egl
vaapi-vulkan vaapi-wayland vaapi-x-egl vaapi-x11 vdpau vector vt.h vulkan
wayland wayland-protocols x11 xv zimg zlib
[cplayer] Reading config file /etc/mpv/encoding-profiles.conf
[cplayer] Applying profile 'default'...
[cplayer] Reading config file /etc/mpv/mpv.conf
[cplayer] Applying profile 'default'...
[cplayer] Setting option 'hwdec' = 'rkmpp' (flags = 4)
[cplayer] Setting option 'vo' = 'gpu' (flags = 4)
[cplayer] Setting option 'gpu-context' = 'drm' (flags = 4)
[cplayer] Setting option 'drm-draw-plane' = 'overlay' (flags = 4)
[cplayer] Setting option 'drm-drmprime-video-plane' = 'primary' (flags = 4)
[cplayer] Setting option 'hwdec' = 'rkmpp' (flags = 8)
[cplayer] Setting option 'vo' = 'gpu' (flags = 8)
[cplayer] Setting option 'gpu-api' = 'opengl' (flags = 8)
[cplayer] Setting option 'opengl-es' = 'yes' (flags = 8)
[cplayer] Setting option 'gpu-context' = 'x11egl' (flags = 8)
[cplayer] Setting option 'drm-draw-plane' = 'overlay' (flags = 8)
[cplayer] Setting option 'drm-drmprime-video-plane' = 'primary' (flags = 8)
[cplayer] Setting option 'drm-device' = '/dev/dri/card0' (flags = 8)
[cplayer] Setting option 'demuxer-lavf-format' = 'video4linux2' (flags = 8)
[cplayer] Setting option 'demuxer-lavf-o-set' = 'input_format=yuyv422'
(flags = 8)
[cplayer] Setting option 'loop-file' = 'inf' (flags = 8)
[cplayer] Setting option 'border' = 'no' (flags = 8)
[cplayer] Setting option 'audio-device' = 'alsa:device=rockchip-hdmi0'
(flags = 8)
[cplayer] Setting option 'v' = '' (flags = 8)
[cplayer] Waiting for scripts...
[... 90 lines cut ...]
SP=1.000000 CL=mpeg2/4/h264
[cplayer] Set property: shared-script-properties -> 1
[cplayer] VO: [gpu] 1280x720 yuv422p
[cplayer] VO: Description: Shader-based GPU Renderer
[vo/gpu] reconfig to 1280x720 yuv422p bt.709/bt.709/bt.1886/limited/display
SP=1.000000 CL=mpeg2/4/h264
[vo/gpu] Resize: 1280x720
[vo/gpu] Window size: 1280x720 (Borders: l=0 t=0 r=0 b=0)
[vo/gpu] Video source: 1280x720 (1:1)
[vo/gpu] Video display: (0, 0) 1280x720 -> (0, 0) 1280x720
[vo/gpu] Video scale: 1.000000/1.000000
[vo/gpu] OSD borders: l=0 t=0 r=0 b=0
[vo/gpu] Video borders: l=0 t=0 r=0 b=0
[vo/gpu] Reported display depth: 8
[vo/gpu] Texture for plane 0: 1280x720
[vo/gpu] Texture for plane 1: 640x720
[vo/gpu] Texture for plane 2: 640x720
[vo/gpu] Testing FBO format rgba16f
[vo/gpu] Using FBO format rgba16f.
[vo/gpu] No advanced processing required. Enabling dumb mode.
[cplayer] first video frame after restart shown
[cplayer] playback restart complete @ 0.000000, audio=eof, video=playing
[vo/gpu/x11] Disabling screensaver.
[cplayer] Set property: shared-script-properties -> 1
[cplayer] Set property: shared-script-properties -> 1

hbiyik, 2023-11-17:
in drm prime it is not possible to output in planar form,

how ever you can do

`FFMPEG_RKMPP_PIXFMT=yuv422p mpv --msg-level=ffmpeg=debug somefile.mp4`,

this will force the output to yuv422p, however i am not sure how the decoder will output yuyv422, i assume it should output NV16.

You can see the decoder flow in log if you do `--msg-level=ffmpeg=debug`

hbiyik, 2023-11-20:
@avafinger 
i fixed my problem in https://github.com/hbiyik/FFmpeg/commit/65d4cf4bd74904dcbcd935685d2160da83f8f72d, fyi, this happens when the input size of picture is small, and rga is faster than dma, you have to sync the dma with DMA_BUF_IOCTL_SYNC

hbiyik, 2023-12-01:
now the similar inferface is implemented in mpp https://github.com/rockchip-linux/mpp/commit/e15972e9af2a593dbf844615eb28365a8b86d0b7

avafinger, 2023-12-02:
> now the similar inferface is implemented in mpp e15972e

i tested again with the latest mpp using the enc demo and YUYV on RK3568
platform (kernel 5.10.160): the same results, mpp, and gstreamer. NV12
works.
I have yet to test your ffmpeg.

v4l2-ctl --all -d /dev/video9
Driver Info:
Driver name      : uvcvideo
Card type        : USB 2.0 Camera: HD 720P Webcam
Bus info         : usb-xhci-hcd.1.auto-1
Driver version   : 5.10.160
Capabilities     : 0x84a00001
Video Capture
Metadata Capture
Streaming
Extended Pix Format
Device Capabilities
Device Caps      : 0x04200001
Video Capture
Streaming
Extended Pix Format
Media Driver Info:
Driver name      : uvcvideo
Model            : USB 2.0 Camera: HD 720P Webcam
Serial           :
Bus info         : usb-xhci-hcd.1.auto-1
Media version    : 5.10.160
Hardware revision: 0x00000000 (0)
Driver version   : 5.10.160
Interface Info:
ID               : 0x03000002
Type             : V4L Video
Entity Info:
ID               : 0x00000001 (1)
Name             : USB 2.0 Camera: HD 720P Webcam
Function         : V4L2 I/O
Flags            : default
Pad 0x01000007   : 0: Sink
 Link 0x02000010: from remote pad 0x100000a of entity 'Extension 4' (Video
Pixel Formatter): Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
Width/Height      : 1280/720
Pixel Format      : 'YUYV' (YUYV 4:2:2)
Field             : None
Bytes per Line    : 2560
Size Image        : 1843200
Colorspace        : sRGB
Transfer Function : Rec. 709
YCbCr/HSV Encoding: ITU-R 601
Quantization      : Default (maps to Limited Range)
Flags             :
Crop Capability Video Capture:
Bounds      : Left 0, Top 0, Width 1280, Height 720
Default     : Left 0, Top 0, Width 1280, Height 720
Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 1280, Height
720, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 1280, Height
720, Flags:
Streaming Parameters Video Capture:
Capabilities     : timeperframe
Frames per second: 10.000 (10/1)
Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=64 step=1
default=0 value=0
                       contrast 0x00980901 (int)    : min=0 max=64 step=1
default=32 value=32
                     saturation 0x00980902 (int)    : min=1 max=128 step=1
default=64 value=64
                            hue 0x00980903 (int)    : min=-40 max=40 step=1
default=0 value=0
                          gamma 0x00980910 (int)    : min=72 max=500 step=1
default=100 value=100
                           gain 0x00980913 (int)    : min=0 max=100 step=1
default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1
value=1 (50 Hz)
0: Disabled
1: 50 Hz
2: 60 Hz
                      sharpness 0x0098091b (int)    : min=1 max=10 step=1
default=3 value=3
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1
default=1 value=1

avafinger, 2023-12-25:
FYI, i tested the 1.0.3 (2023-12-08) which added dmabuf sync support on my
kernel 5.10.110 (RK3588), unfortunately, the problem persists.
Can someone confirm the problem is not on my side, if you can run on
5.10.160 would also be interesting to check if my kernel is synced with the
latest rga driver.

p3ngu19z, 2024-04-19:
Any progress to resolve?

I have similar problem with on radxa zero 3w and radxa-zero3_debian_bullseye_xfce_b6.img build

Camera YUYV only

qqzlqqzlqqzl, 2024-08-07:
好东西，问题解决了

fancong5201314, 2025-02-07:
Mark it, I also encountered a similar issue, and adding **_io-mode=dmabuf-import_** resolved it. Thank you!

my-run, 2025-04-03:
Does ffmpeg could use "io-mode=dmabuf-import"