rockchip-linux/mpp #262: mpi_enc_multi_test crashes on r1126

Consti10, 2021-11-23:
Hello,

Is multi-threaded encoding supported on rv1126 ? When I run mpi_enc_multi_test it crashes without any info:

[root@RV1126_RV1109:/oem/usr/bin]# ./mpi_enc_multi_test  -o /tmp/mpi_test.h264 -
w 1280 -h 720 -f NV12 
mpp[1220]: mpi_enc_multi_test: cmd parse result:
mpp[1220]: mpi_enc_multi_test: input  file name: 
mpp[1220]: mpi_enc_multi_test: output file name: /tmp/mpi_test.h264
mpp[1220]: mpi_enc_multi_test: width      : 1280
mpp[1220]: mpi_enc_multi_test: height     : 720
mpp[1220]: mpi_enc_multi_test: type       : 0
mpp[1220]: mpi_enc_multi_test: debug flag : 0
Floating point exception (core dumped)

mpi_enc_test works fine.

HermanChen, 2021-11-24:
add -p 1 option

Consti10, 2022-02-04:
Example command that works:
./mpi_enc_multi_test  -o /tmp/mpi_test.h264 -w 1280 -h 720 -f NV12 -t 7 -n 100 -p 1

closed 2022-02-08 by HermanChen