rockchip-linux/mpp #263: rk3399+ubuntu, 运行mpi_dec_test解码h264时报 mpp_drm: os_allocator_drm_alloc drm_alloc failed ret -1

whwalker, 2021-11-29:
我的平台是rk3399+ubuntu 20.04,运行系统自带的mpi_dec_test可以解码h264, 从github下载源码编译的mpp_dec_test解码h264一直报错内存分配出错，请问是什么原因？
具体log见下面

自己编译的mpi_dec_test运行log:
./mpi_dec_test -i RemixOS.h264 -n 10
mpp[30529]: mpi_dec_utils: input file RemixOS.h264 size 36038629
mpp[30529]: mpi_dec_utils: cmd parse result:
mpp[30529]: mpi_dec_utils: input  file name: RemixOS.h264
mpp[30529]: mpi_dec_utils: output file name: 
mpp[30529]: mpi_dec_utils: width      :    0
mpp[30529]: mpi_dec_utils: height     :    0
mpp[30529]: mpi_dec_utils: type       :    7
mpp[30529]: mpi_dec_utils: max frames :   10
mpp[30529]: mpi_dec_test: mpi_dec_test start
mpp[30529]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[30529]: mpi_dec_test: 0xaaaaebaf2ee0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[30529]: mpp_rt: NOT found ion allocator
mpp[30529]: mpp_rt: found drm allocator
mpp[30529]: mpp_drm: os_allocator_drm_alloc drm_alloc failed ret -1
mpp[30529]: mpp_buffer: mpp_buffer_create failed to create buffer with size 3840
mpp[30529]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[30529]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[30529]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[30529]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[30529]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[30529]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[30529]: mpp_hal: mpp_hal_init could not found coding type 7
mpp[30529]: mpp_dec: mpp_dec_init could not init hal
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[30529]: mpp: error found on mpp initialization
mpp[30529]: mpi_dec_test: 0xaaaaebaf2ee0 mpp_init failed
mpp[30529]: mpi_dec_test: test failed ret -1
mpp[30529]: mpp_buffer: ~MppBufferService cleaning leaked group
mpi_dec_test[30529]: mpp_mem_pool: put_pool found 1 used buffer size 80
mpi_dec_test[30529]: mpp_mem_pool: put_pool found 4 used buffer size 192

系统自带的mpi_dec_test运行正常,log如下：
mpi_dec_test -i RemixOS.h264 -n 10
mpp[30532]: mpi_dec_test: cmd parse result:
mpp[30532]: mpi_dec_test: input  file name: RemixOS.h264
mpp[30532]: mpi_dec_test: output file name: 
mpp[30532]: mpi_dec_test: config file name: 
mpp[30532]: mpi_dec_test: width      :    0
mpp[30532]: mpi_dec_test: height     :    0
mpp[30532]: mpi_dec_test: type       : 7
mpp[30532]: mpi_dec_test: debug flag : 0
mpp[30532]: mpi_dec_test: max frames : 10
mpp[30532]: mpi_dec_test: mpi_dec_test start
mpp[30532]: mpi_dec_test: input file size 36038629
mpp[30532]: mpp_info: mpp version: 41c9c809 author: sayon.chen    2020-12-24 [mpp2str]: Add rc_gop_mode 2 str
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[30532]: mpp_rt: NOT found ion allocator
mpp[30532]: mpp_rt: found drm allocator
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode_get_frame get info changed found
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 0
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 1
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 2
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 3
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 4
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 5
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 6
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 7
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 8
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 decode get frame 9
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 reach max frame number 10
mpp[30532]: mpi_dec_test: 0xaaaaf1829580 input 157 pkt output 12 frm decode 11 frames
mpp[30532]: mpi_dec_test: test success max memory 7.97 MB

系统信息如下：
sky@localhost:~$ cat /proc/version 
Linux version 5.7.17 (eason@skysi-admin) (gcc version 6.3.1 20170404 (Linaro GCC 6.3-2017.05), GNU ld (Linaro_Binutils-2017.05) 2.27.0.20161019) #4 SMP PREEMPT Mon Jul 12 11:32:31 CST 2021
sky@localhost:~$ uname -a
Linux localhost.localdomain 5.7.17 #4 SMP PREEMPT Mon Jul 12 11:32:31 CST 2021 aarch64 aarch64 aarch64 GNU/Linux    
sky@localhost:~$ cat /etc/issue
Ubuntu 20.04.3 LTS \n \l

编译：
cd mpp/build/linux/aarch64; 修改 arm.linux.cross.cmake，然后 ./make-Makefiles.bash; make; 编译出的。
arm.linux.cross.cmake 内容如下：
cmake_minimum_required( VERSION 2.6.3 )

SET(CMAKE_SYSTEM_NAME Linux)
#SET(CMAKE_C_COMPILER "aarch64-linux-gnu-gcc")
#SET(CMAKE_CXX_COMPILER "aarch64-linux-gnu-g++")
SET(CMAKE_C_COMPILER "gcc")
SET(CMAKE_CXX_COMPILER "g++")
SET(CMAKE_SYSTEM_PROCESSOR "armv8-a")

add_definitions(-fPIC)
add_definitions(-DARMLINUX)
add_definitions(-D__gnu_linux__)

HermanChen, 2021-11-29:
估计是你那这的 drm 分配器打开和 mmap 的 flag 不对，对比下看看

whwalker, 2021-12-03:
我看 os_allocator_drm_open 里面是 读写 打开的/dev/dri/card0,   
```
fd = open(dev_drm, O_RDWR | O_CLOEXEC);
```
还没有走到 os_allocator_drm_mmap 里面去，是在 os_allocator_drm_alloc 里面调用 drm_alloc时失败了
```
static MPP_RET os_allocator_drm_alloc(void *ctx, MppBufferInfo *info)
{
  ...
    drm_dbg_func("dev %d alloc alignment %d size %d\n", p->drm_device,
                 p->alignment, info->size);

    ret = drm_alloc(p->drm_device, info->size, p->alignment,
                    (RK_U32 *)&info->hnd, p->flags);   // 在这里失败了
  ...
}
```
drm_alloc  调用 drm_ioctl ，drm_ioctl调用 ioctl 失败的
```
ret = drm_ioctl(fd, DRM_IOCTL_MODE_CREATE_DUMB, &dmcb);
```
log如下：
```
mpp[269074]: mpi_dec_utils: width      :    0
mpp[269074]: mpi_dec_utils: height     :    0
mpp[269074]: mpi_dec_utils: type       :    7
mpp[269074]: mpi_dec_utils: max frames :   10
mpp[269074]: mpi_dec_test: mpi_dec_test start
mpp[269074]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[269074]: mpi_dec_test: 0xaffe3a9bfee0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[269074]: mpp_rt: NOT found ion allocator
mpp[269074]: mpp_rt: found drm allocator
mpp[269074]: mpp_drm: os_allocator_drm_open enter
mpp[269074]: mpp_drm: os_allocator_drm_open leave dev 5
mpp[269074]: mpp_drm: os_allocator_drm_open enter
mpp[269074]: mpp_drm: os_allocator_drm_open leave dev 6
mpp[269074]: mpp_drm: os_allocator_drm_alloc dev 5 alloc alignment 4096 size 3840
mpp[269074]: mpp_drm: os_allocator_drm_alloc, size=3840, alignment=4096
mpp[269074]: mpp_drm: drm_alloc drm_ioctl ret =-1
mpp[269074]: mpp_drm: os_allocator_drm_alloc drm_alloc failed ret -1
mpp[269074]: mpp_buffer: mpp_buffer_create failed to create buffer with size 3840
mpp[269074]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[269074]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[269074]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[269074]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
mpp[269074]: mpp_buffer: mpp_buffer_put invalid input: buffer NULL from rkv_h264d_deinit
```

不太明白“drm 分配器打开和 mmap 的 flag 不对” ，从log看还没有走到  os_allocator_drm_mmap 里面去调用 mmap，麻烦再帮忙看看，谢谢。

whwalker, 2021-12-03:
或者能否给个RK3399上可编译运行的demo源码?

JeffyCN, 2022-01-27:
mpp默认是打开/dev/dri/card0，如果机器中存在多个drm设备时候可能打开的不是vop drm驱动设备。