rockchip-linux/mpp #265: 使用mpi_dec_test解码mpi_enc_test编码的MJEPG图片，会打印日志 vcodec_service: vcodec_service_reg_wr reg size mismatch wr 636 rd 736

imxys, 2021-12-14:
环境：RK3399
MPP：github版本见日志
这个打印日志应该是解码一帧MJPEG就打印一条这个log的，所以解码视频时刷屏很厉害；麻烦看一下怎么回事，谢谢！

日志记录如下
```
[root@root:~]# /app_data/mpi_enc_test -t 8 -w 1920 -h 1080 -o /tmp/demo.mjpeg
[root@root:~]# /app_data/mpi_enc_test -t 8 -w 1920 -h 1080
mpp[5169]: mpi_enc_utils: cmd parse result:
mpp[5169]: mpi_enc_utils: input  file name: (null)
mpp[5169]: mpi_enc_utils: output file name: (null)
mpp[5169]: mpi_enc_utils: width      : 1920
mpp[5169]: mpi_enc_utils: height     : 1080
mpp[5169]: mpi_enc_utils: format     : 0
mpp[5169]: mpi_enc_utils: type       : 8
mpp[5169]: mpi_enc_test: mpi_enc_test start
mpp[5169]: mpi_enc_test: jpege default encode only one frame. Use -n [num] for rc case
mpp[5169]: mpp_rt: NOT found ion allocator
mpp[5169]: mpp_rt: found drm allocator
mpp[5169]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[5169]: mpi_enc_test: 0x2d98cdc0 mpi_enc_test encoder test start w 1920 h 1080 type 8
mpp[5169]: mpp_enc: MPP_ENC_SET_RC_CFG bps 7776000 [486000 : 8262000] fps [30:30] gop 60
mpp[5169]: mpi_enc_test: 0x2d98cdc0 encoded frame 0    size 149254 
mpp[5169]: mpi_enc_test: 0x2d98cdc0 encode max 1 frames
mpp[5169]: mpi_enc_test: 0x2d98cdc0 mpi_enc_test success total frame 1 bps 35820960
[root@root:~]# 
[root@root:~]# 
[root@root:~]#
[root@root:~]# /app_data/mpi_dec_test -t 8 -w 1920 -h 1080 -i /tmp/demo.mjpeg 
mpp[6175]: mpp_rt: NOT found ion allocator
mpp[6175]: mpp_rt: found drm allocator
mpp[6175]: mpi_dec_utils: input file /tmp/demo.mjpeg size 149254
mpp[6175]: mpi_dec_utils: cmd parse result:
mpp[6175]: mpi_dec_utils: input  file name: /tmp/demo.mjpeg
mpp[6175]: mpi_dec_utils: output file name: 
mpp[6175]: mpi_dec_utils: width      : 1920
mpp[6175]: mpi_dec_utils: height     : 1080
mpp[6175]: mpi_dec_utils: type       :    8
mpp[6175]: mpi_dec_utils: max frames :    0
mpp[6175]: mpi_dec_test: mpi_dec_test start
mpp[6175]: mpp_info: mpp version: 786b79f9 author: Yandong Lin   2021-11-03 [h265d_parser]: Fix h265d parser crash issue
mpp[6175]: mpi_dec_test: 0x25c43120 mpi_dec_test decoder test start w 1920 h 1080 type 8
mpp[6175]: vcodec_service: vcodec_service_reg_wr reg size mismatch wr 636 rd 736                <=  this!!!!!!!
mpp[6175]: mpi_dec_test: 0x25c43120 decoded frame 0
mpp[6175]: mpi_dec_test: decode 1 frames time 20 ms delay  20 ms fps 47.73
mpp[6175]: mpi_dec_test: test success max memory 0.00 MB
[root@root:~]# 

```

HermanChen, 2021-12-15:
试试看这里的代码会不会有一样的问题：https://github.com/hermanchen/mpp/

HermanChen, 2021-12-15:
看了下，这个 log 是个 warning，在兼容旧 vcodec_servcie 内核驱动时的 log，运行时没有问题的话不影响

imxys, 2021-12-15:
> 看了下，这个 log 是个 warning，在兼容旧 vcodec_servcie 内核驱动时的 log，运行时没有问题的话不影响

好的；那请问内核的驱动是否需要更新，以及在哪里进行更新呢？是这个项目吗 https://github.com/rockchip-linux/kernel

HermanChen, 2021-12-15:
https://github.com/rockchip-linux/kernel/tree/develop-4.19/drivers/video/rockchip/mpp
建议升到 4.19 内核，使用新的 mpp 驱动

JeffyCN, 2022-01-27:
印象中这个log是因为jpeg解码器pp转换功能开启关闭时候寄存器表大小不一样导致，没开pp会检测失败打印警告

kadirbeytorun, 2022-04-19:
@JeffyCN Hey Jeffy, I receive this warning when I access USB cam using RKFacial library. There is no crash or error, but the warning is quite annoying. How could I fix the issue? Or should I just turn off that warning?

JeffyCN, 2022-04-19:
@kadirbeytorun , i think it's harmless to just remove that warning. that warning means you are not using the internal hardware format conversion module(pp), that would confuse this size check logic.