rockchip-linux/mpp #267: rk3399 codec performance

Allenhe123, 2021-12-20:
Reference to the newest(version 2.1) datasheet's video decoder:
H.264/AVC, Base/Main/High/High10 profile@level5.1, up to 4Kx2K@30fps
H.265/HEVC, Main/Main10/ profile@level5.1 High-tier ;up to 4Kx2K@60fps

But we got a different performace data in out test.
We decoded h264 and h265 videos of 4K(3840x2160):
h264@100fps.
h265@100fps.
The test file is a mp4 video, a expressway with many cars runnning. The Code stream max to 4096 kb/s.

Questions:
1. why our test results are different with that in datasheet?
2. We also done a high temperature test with decoder 4K video running. The OS of rk3399 will be rebooted by watchdog about 7-10 times, and the hanged. Is this issue related to the decoder and rga with a higher fps job than fps in datasheet?

Thanks in advance for you reply.

HermanChen, 2021-12-21:
1. 3399 的 datasheet 是针对高码率和10bit 码流的，H.265 60fps / H.264 30fps 指的是在 100% 的情况下都能达到的性能，你们测试的 100fps 的片源只有 4M 码率，分辨率也没有达到 4096x2304，也不是 10bit，所以帧率要高是很正常的。
2. 应该是过热了，需要加散热片，或者在内核里打开功耗控制的开关

Allenhe123, 2021-12-21:
谢谢回复，请问一下怎么在内核里面打开功耗控制开关呢？

HermanChen, 2021-12-21:
这个要看发布的 sdk 包里的 dts 配置了

Allenhe123, 2021-12-21:
我们用的是这个仓库：
https://github.com/friendlyarm/kernel-rockchip  -b  nanopi4-linux-v4.4.y kernel-rockchip
dts路经是：arch/arm64/boot/dts/rockchip/rk3399.dtsi 吗？
要修改里面的哪个字段呢？

Allenhe123, 2021-12-21:
https://github.com/rockchip-linux/kernel
https://github.com/rockchip-toybrick/kernel
这2个仓库有什么区别呢？都是官方的吗？

HermanChen, 2021-12-22:
这些就不太清楚了，需要到那些仓库上问问

Allenhe123, 2021-12-22:
功耗开关怎么修改呢？

Allenhe123, 2021-12-23:
Symbol: ROCKCHIP_THERMAL [=n]                                                                 
  │ Type  : tristate                                                                            
  │ Prompt: Rockchip thermal driver                                               
  │   Location:                                                                                    
  │     -> Device Drivers                                                                    
  │ (1)   -> Generic Thermal sysfs driver (THERMAL [=y])  

请问是修改ROCKCHIP_THERMAL这个吗

Allenhe123, 2022-01-13:
问题已解决，close

closed 2022-01-13 by Allenhe123