rockchip-linux/mpp #271: rk3399编码质量问题

wanhaiping, 2022-01-26:
环境android7  rk3399
测试中发现rk3399在编码宽1280高720时正常，宽720高1280时质量会差很多。
所以在编码宽720高1280时，先旋转90度再编码正常，但是旋转也要耗时，是否有这个问题，还是哪里设置不对？ 谢谢

wanhaiping, 2022-01-26:
编码h264

HermanChen, 2022-01-26:
https://github.com/rockchip-linux/mpp/blob/786b79f9767d24bed618be60046546b508f9190e/mpp/hal/vpu/h264e/hal_h264e_vepu_v2.c#L638

修改成 mbrc->cp_distance_mbs = 0; 这样关掉一帧之内的宏块级码控看下

wanhaiping, 2022-01-26:
修改成 mbrc->cp_distance_mbs = 0后。1280x720，720x1280两种方式下质量都有所改善，但两者编码后的质量依然差很多

HermanChen, 2022-01-26:
这个正常的，因为编码器硬件实现决定了水平方向搜索范围大，如果图像是竖向的，压缩率和图像质量都会受影响

wanhaiping, 2022-01-26:
我是用游戏画面测试的
图像设置为横向，画面横竖向运动时，画质在正常范围内。
图像设置为竖向，画面完全横向运动时，画面正常，只要有很小的竖向运动，画质就差不少。
感觉像是运动搜索的问题

HermanChen, 2022-01-27:
的确是这样的，水平方向运行对编码器比较友好，只要有垂直运动，对压缩率影响就比较大

wanhaiping, 2022-01-27:
好的，我明白了。在编码前硬件能做旋转处理吗，VEPU_REG_IN_IMG_ROTATE_MODE这个设置和旋转有关？

HermanChen, 2022-01-27:
编码器支持在编码器做 90/270 这样的旋转，可以的。
如果问题己解决，麻烦关闭问题：）

wanhaiping, 2022-01-27:
旋转还有问题，VEPU_REG_IN_IMG_ROTATE_MODE默认设置是0，我直接设置成1，2或者3，都是花屏，
编码前旋转应该怎么设置

HermanChen, 2022-01-27:
通过 prep:rotation 设置

wanhaiping, 2022-01-27:
旋转前
[image]
旋转后
[image]

HermanChen, 2022-01-27:
这应该是宽高stride配置有问题

wanhaiping, 2022-01-27:
看效果应该是没有旋转的，我是通过MediaCodec调用mpp库，在函数vpu_api_set_enc_cfg添加mpp_enc_cfg_set_s32(enc_cfg, "prep:rotation", MPP_ENC_ROT_90);

JeffyCN, 2022-01-27:
3399好像不支持旋转

wanhaiping, 2022-01-27:
好的，了解了。查看代码这个设置在rk系列和旋转没太大关系，不确定是硬件本来就不支持，还是别的原因没实现，感谢回复！

JeffyCN, 2022-01-27:
https://github.com/rockchip-linux/mpp/issues/183

closed 2022-01-27 by JeffyCN