rockchip-linux/mpp #272: h264_rkmpp decoder in ffmpeg: "Failed to send packet to decoder (code = -11)"

gagara, 2022-01-29:
ffmpeg unable to decode h264 stream with `h264_rkmpp` decoder.
Command I use:
```
ffmpeg -y -benchmark -t 30 -analyzeduration 1000000 -probesize 1000000 -fflags +igndts -hwaccel drm -hwaccel_device /dev/dri/card0 -c:v h264_rkmpp -rtsp_transport tcp -i <rtsp_stream_url> -vf fps=2,hwdownload,format=nv12 -s 640x360 -an -c:v pam -pix_fmt gray -f image2pipe /dev/null
```
ffmpeg debug output snippet:
```
  Stream #0:0, 22, 1/90000: Video: h264 (Main), 1 reference frame, yuvj420p(pc, bt709, progressive, left), 640x360 (640x368), 0/1, 20 fps, 20 tbr, 90k tbn, 40 tbc
Successfully opened the file.
Parsing a group of options: output url /dev/null.
Applying option vf (set video filters) with argument fps=2,hwdownload,format=nv12.
Applying option s (set frame size (WxH or abbreviation)) with argument 640x480.
Applying option an (disable audio) with argument 1.
Applying option c:v (codec name) with argument pam.
Applying option pix_fmt (set pixel format) with argument gray.
Applying option f (force format) with argument image2pipe.
Successfully parsed a group of options.
Opening an output file: /dev/null.
[file @ 0x55b1a95950] Setting default whitelist 'file,crypto,data'
Successfully opened the file.
[AVHWDeviceContext @ 0x55b1ac9c40] Opened DRM device /dev/dri/card0: driver rockchip version 1.0.0.
[h264_mp4toannexb @ 0x55b1ac9c80] The input looks like it is Annex B already
[h264_rkmpp @ 0x55b1b58e90] Initializing RKMPP decoder.
mpp[16497]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[16497]: mpp_rt: NOT found ion allocator
mpp[16497]: mpp_rt: found drm allocator
mpp[16497]: mpp: deprecated block control, use timeout control instead
mpp[16497]: mpp: deprecated block control, use timeout control instead
[h264_rkmpp @ 0x55b1b58e90] RKMPP decoder initialized successfully.
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (h264_rkmpp) -> pam (native))
Press [q] to stop, [?] for help
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[h264_rkmpp @ 0x55b1b58e90] Wrote 37 bytes to decoder
[h264_rkmpp @ 0x55b1b58e90] Wrote 1147 bytes to decoder
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[h264_rkmpp @ 0x55b1b58e90] Wrote 17658 bytes to decoder
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[h264_rkmpp @ 0x55b1b58e90] Wrote 292 bytes to decoder
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[h264_rkmpp @ 0x55b1b58e90] Wrote 307 bytes to decoder
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[h264_rkmpp @ 0x55b1b58e90] Buffer full writing 336 bytes to decoder
[h264_rkmpp @ 0x55b1b58e90] Failed to send packet to decoder (code = -11)
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[h264_rkmpp @ 0x55b1b58e90] Buffer full writing 374 bytes to decoder
[h264_rkmpp @ 0x55b1b58e90] Failed to send packet to decoder (code = -11)
cur_dts is invalid st:0 (0) [init:0 i_done:0 finish:0] (this is harmless if it occurs once at the start per stream)
[h264_rkmpp @ 0x55b1b58e90] Buffer full writing 746 bytes to decoder
....................
```
However it works fine with commit `9ec098a9819` and prior.

markluofd, 2022-06-21:
I met the same problem

[h264_rkmpp @ 0x55b8824a40] Failed to send packet to decoder (code = -11)
    Last message repeated 92 times
[h264_rkmpp @ 0x55b8824a40] Decoder noticed an info change (1920x1080), format=0
Output #0, null, to 'pipe:':
  Metadata:
    major_brand     : mp42
    minor_version   : 1
    compatible_brands: isommp41mp42
    encoder         : Lavf59.25.100
  Stream #0:0(und): Video: wrapped_avframe, drm_prime(tv, bt709, progressive), 1920x1080 [SAR 1:1 DAR 16:9], q=2-31, 200 kb/s, 29.97 fps, 29.97 tbn (default)
    Metadata:
      creation_time   : 2021-05-10T12:46:47.000000Z
      handler_name    : Core Media Video
      vendor_id       : [0][0][0][0]
      encoder         : Lavc59.33.100 wrapped_avframe
frame=    4 fps=0.0 q=-0.0 Lsize=N/A time=00:00:00.13 bitrate=N/A speed=2.14x
video:2kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: unknown

chuqingq, 2022-08-23:
I met the same problem too.

JeffyCN, 2022-08-23:
has anyone tried gstreamer mpp?

chuqingq, 2022-08-24:
> has anyone tried gstreamer mpp?

Is there any official gstreamer mpp repo, or unofficial gstreamer mpp repo under maintenance please?

JeffyCN, 2022-08-24:
> > has anyone tried gstreamer mpp?
> 
> Is there any official gstreamer mpp repo, or unofficial gstreamer mpp repo under maintenance please?

try
https://github.com/JeffyCN/rockchip_mirrors/tree/gstreamer-rockchip

734140764, 2023-01-30:
met the same problem on 3588s too. anyone give a suggestion

JeffyCN, 2023-01-30:
-11 means busy, the mpp internal input queue is full. if it didn't recover, could be mpp stop decoding.

also maybe try the newest ffmpeg in my github

pipiwoaini, 2023-08-18:
我也遇到了同样的问题。 有大佬解决吗

tu-nie, 2024-07-07:
解决了吗，大佬们

tu-nie, 2024-07-08:
> 我遇到了同样的问题
> 
> [h264_rkmpp @ 0x55b8824a40]无法向解码器发送数据包（代码 = -11） 最后一条消息重复了 92 次 [h264_rkmpp @ 0x55b8824a40] 解码器注意到信息更改 （1920x1080），format=0 输出 #0，null，到 'pipe：'： 元数据： major_brand：mp42 minor_version：1 compatible_brands：isommp41mp42 编码器：Lavf59.25.100流 #0：0（und）： 视频： wrapped_avframe， drm_prime（tv， bt709， progressive）， 1920x1080 [SAR 1：1 DAR 16：9]， q=2-31， 200 kb/s， 29.97 fps， 29.97 tbn （默认） 元数据： creation_time：2021-05-10T12：46：47.000000Z handler_name：核心媒体视频vendor_id：[0][0][0][0] 编码器：Lavc59.33.100 wrapped_avframe帧= 4 fps=0.0 q=-0.0 Lsize=N/A 时间=00：00：00.13 比特率=N/A 速度=2.14x视频：2kB 音频：0kB 字幕：0kB 其他流：0kB 全局标头：0kB 多路复用开销：未知

您好，解决了吗

wn2000, 2024-07-09:
I met the same issue on RK3328, 4.4.159 kernel.

Did some bisecting and found the issue started from this commit: 6b5ee5a.

wn2000, 2024-07-09:
I checked out the repo at 6b5ee5a, and enabled debug env (`export mpi_debug=0xff; export mpp_task_debug=0xff`).
The following is the log leading to the error.
@HermanChen Would you be able to recognize something wrong from this log?
The only change I made before generating this log, was to insert a print here:
https://github.com/rockchip-linux/mpp/blob/6b5ee5a122d42d72209340cfbe8fe4ced22b6d2d/mpp/mpp.cpp#L318-L319
to make it:
```c++
if (!eos && ret <= mDecResTaskCnt) {
    mpp_err_f("ret %d <= mDecResTaskCnt %d\n", ret, mDecResTaskCnt);
    goto RET;
}
```
so in this case `ret` and `mDecResTaskCnt` were both 1.

```log
mpp[6450]: mpi: mpp_create enter ctx 0xd44b10 mpi 0xd44b14
mpp[6450]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[6450]: mpi: mpp_create leave ret 0 ctx 0xd00820 mpi 0xf54811f0
mpp[6450]: mpi: mpp_init enter ctx 0xd00820 type 0 coding 7
mpp[6450]: mpp_task_impl: mpp_task_queue_init enter
mpp[6450]: mpp_task_impl: mpp_port_init enter queue 0xd5fd40 type 0
mpp[6450]: mpp_task_impl: mpp_port_init leave queue 0xd5fd40 port 0xd5fe20
mpp[6450]: mpp_task_impl: mpp_port_init enter queue 0xd5fd40 type 1
mpp[6450]: mpp_task_impl: mpp_port_init leave queue 0xd5fd40 port 0xd00860
mpp[6450]: mpp_task_impl: mpp_task_queue_init leave ret 0 queue 0xd5fd40
mpp[6450]: mpp_task_impl: mpp_task_queue_init enter
mpp[6450]: mpp_task_impl: mpp_port_init enter queue 0xd360c0 type 0
mpp[6450]: mpp_task_impl: mpp_port_init leave queue 0xd360c0 port 0xd361a0
mpp[6450]: mpp_task_impl: mpp_port_init enter queue 0xd360c0 type 1
mpp[6450]: mpp_task_impl: mpp_port_init leave queue 0xd360c0 port 0xd08280
mpp[6450]: mpp_task_impl: mpp_task_queue_init leave ret 0 queue 0xd360c0
mpp[6450]: mpp_rt: NOT found ion allocator
mpp[6450]: mpp_rt: found drm allocator
mpp[6450]: mpi: mpp_init leave ret 0
mpp[6450]: mpp_task_impl: _mpp_port_poll enter port 0xd00860
mpp[6450]: mpi: mpi_control enter ctx 0xd00820 cmd 200004 parm 0xffcea514
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from try_get_input_packet poll output port timeout -1 ret 13073896
mpp[6450]: mpp: deprecated block control, use timeout control instead
mpp[6450]: mpp_task_impl: _mpp_port_poll leave
mpp[6450]: mpi: mpi_control leave ret 0
mpp[6450]: mpi: mpi_control enter ctx 0xd00820 cmd 200005 parm 0xffcea518
mpp[6450]: mpp: deprecated block control, use timeout control instead
mpp[6450]: mpi: mpi_control leave ret 0
mpp[6450]: mpi: mpi_control enter ctx 0xd00820 cmd 310002 parm 0xd20a60
mpp[6450]: mpi: mpi_control leave ret 0
mpp[6450]: mpi: mpi_control enter ctx 0xd00820 cmd 310007 parm 0xffcea620
mpp[6450]: mpi: mpi_control leave ret 0
mpp[6450]: mpi: mpi_control enter ctx 0xd00820 cmd 310007 parm 0xffcea6a0
mpp[6450]: mpi: mpi_control leave ret 0
mpp[6450]: mpi: mpi_decode_put_packet enter ctx 0xd00820 packet 0xf3aa80
mpp[6450]: mpp_task_impl: _mpp_port_poll enter port 0xd5fe20
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from poll poll input port timeout 0 count 4
mpp[6450]: mpp_task_impl: _mpp_port_poll leave
mpp[6450]: mpp_task_impl: _mpp_port_dequeue caller dequeue enter port 0xd5fe20
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from dequeue dequeue input port task 0xd2aaa0 input_port -> input_hold done
mpp[6450]: mpp_task_impl: _mpp_port_dequeue caller dequeue leave port 0xd5fe20 task 0xd2aaa0 ret 0
mpp[6450]: mpp_task_impl: _mpp_port_enqueue caller enqueue enter port 0xd5fe20 task 0xd2aaa0
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from enqueue enqueue input port task 0xd2aaa0 input_hold -> output_port done
mpp[6450]: mpp_task_impl: _mpp_port_enqueue signal port 0xd5fda8
mpp[6450]: mpp_task_impl: _mpp_port_enqueue caller enqueue leave port 0xd5fe20 task 0xd2aaa0 ret 0
mpp[6450]: mpp_task_impl: _mpp_port_poll enter port 0xd5fe20
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from poll poll input port timeout 0 count 3
mpp[6450]: mpp_task_impl: _mpp_port_poll leave
mpp[6450]: mpp_task_impl: _mpp_port_dequeue caller dequeue enter port 0xd5fe20
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from dequeue dequeue input port task 0xd2aabc input_port -> input_hold done
mpp[6450]: mpp_task_impl: _mpp_port_dequeue caller dequeue leave port 0xd5fe20 task 0xd2aabc ret 0
mpp[6450]: mpi: mpi_decode_put_packet leave ret 0
mpp[6450]: mpp_task_impl: _mpp_port_poll enter port 0xd00860
mpp[6450]: mpi: mpi_decode_put_packet enter ctx 0xd00820 packet 0xf3afc0
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from try_get_input_packet poll output port timeout 0 count 1
[... 88 lines cut ...]
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from poll poll input port timeout 0 count 1
mpp[6450]: mpp_task_impl: _mpp_port_poll leave
mpp[6450]: mpp: put_packet ret 1 <= mDecResTaskCnt 1
mpp[6450]: mpp_task_impl: _mpp_port_poll enter port 0xd5fe20
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from poll poll input port timeout 0 count 1
mpp[6450]: mpp_task_impl: _mpp_port_poll leave
mpp[6450]: mpi: mpi_decode_put_packet leave ret 1
[h264_rkmpp @ 0xcfdc80] Failed to write data to decoder (code = -1313558101)
[h264_rkmpp @ 0xcfdc80] Failed to send packet to decoder (code = -1313558101)
mpp[6450]: mpi: mpi_control enter ctx 0xd00820 cmd 310007 parm 0xffcea620
mpp[6450]: mpi: mpi_control leave ret 0
mpp[6450]: mpi: mpi_control enter ctx 0xd00820 cmd 310007 parm 0xffcea6a0
mpp[6450]: mpi: mpi_control leave ret 0
mpp[6450]: mpi: mpi_decode_put_packet enter ctx 0xd00820 packet 0xd55020
mpp[6450]: mpp_task_impl: _mpp_port_poll enter port 0xd5fe20
[... cut ...]
mpp[6450]: mpp_task_impl: _mpp_port_poll leave
mpp[6450]: mpp: put_packet ret 1 <= mDecResTaskCnt 1
mpp[6450]: mpp_task_impl: _mpp_port_poll enter port 0xd5fe20
mpp[6450]: mpp_task_impl: mpp 0xd1b6b8 input from poll poll input port timeout 0 count 1
mpp[6450]: mpp_task_impl: _mpp_port_poll leave
mpp[6450]: mpi: mpi_decode_put_packet leave ret 1
[h264_rkmpp @ 0xcfdc80] Failed to write data to decoder (code = -1313558101)
[h264_rkmpp @ 0xcfdc80] Failed to send packet to decoder (code = -1313558101)
```

wn2000, 2024-07-09:
BTW how to enable debug log with the latest code? I tried to set `mpi_debug` in the environment but it had no effect.

HermanChen, 2024-07-09:
set_env is not valid on linux environment.
ffmpeg work flow is not fully compatible with mpp.

HermanChen, 2024-07-09:
mpp does not support ffmpeg officially. But we will fix the issue later.

wn2000, 2024-07-09:
> mpp does not support ffmpeg officially. But we will fix the issue later.

Thanks for explaining that. That makes sense now.

I tried @hbiyik's ffmpeg with the latest mpp and that worked. Fantastic work! Thanks both of you.