rockchip-linux/mpp #273: ./mpi_enc_test: symbol lookup error: ./mpi_enc_test: undefine symbol: mpp_spinlock_unlock

xiaoxiaojiea, 2022-02-16:
mpp_develop中的aarch64使用的是交叉编译器编译的，cmake 与 make 都成功，mpi_enc_test 编译出来了，在注意运行报错 /lib/ld-linux-aarch64.so.1: No such file or directory ，然后移动到板子上 报错 ./mpi_enc_test: symbol lookup error: ./mpi_enc_test: undefine symbol: mpp_spinlock_unlock ，请问是我的交叉编译器设置的问题吗？还是 arm.linux.cross.cmake 文件需要改动那里？

HermanChen, 2022-02-16:
交叉编译出来的东西在主机是不能运行的，报 ld 的问题正常
在板上运行出错，是因为只更新了 test 程序，没去更新 mpp 库本身，需要更新 librockchip_mpp.so.0

xiaoxiaojiea, 2022-02-16:
@HermanChen 您所说的更新mpp库，得到 librockchip_mpp.so.0 是要自己编译 mpp 库是不

HermanChen, 2022-02-17:
是的，需要 mpp 库和 mpi_enc_test 是同一个代码编译出来的，这样才能匹配上

xiaoxiaojiea, 2022-02-17:
> 是的，需要 mpp 库和 mpi_enc_test 是同一个代码编译出来的，这样才能匹配上

那为什么直接在板子上编译的 test 程序可以直接运行，我也没有编译 mpp 库哦

lmw0320, 2022-10-20:
> > 是的，需要 mpp 库和 mpi_enc_test 是同一个代码编译出来的，这样才能匹配上
> 
> 那为什么直接在板子上编译的 test 程序可以直接运行，我也没有编译 mpp 库哦

请教下，问题解决了么？？我也遇到类似的情况：mpi_dec_test: symbol lookup error: mpi_dec_test: undefined symbol: _mpp_log_l

我的操作是在rk3399pro板卡上进行的(debian10系统），先git clone官方的mpp库下来，然后根据教程：
`cd build/linux/aarch64
./make-Makefiles.bash
sudo make && sudo make install
`
然后执行命令： mpi_dec_test 测试下，就报了如上的错误。。求指点。。

HermanChen, 2022-10-20:
ldd 看下这个可执行程序链接到哪个 mpp 库了

HermanChen, 2022-10-20:
估计是 mpi_dec_test 更新了，64 位库更新了，但是实际跑到了 32 位的库，这个库没更新

lmw0320, 2022-10-20:
[image]

lmw0320, 2022-10-20:
ldd /usr/local/bin/mpi_dec_test
	linux-vdso.so.1 (0x0000007f92f08000)
	librockchip_mpp.so.1 => /lib/aarch64-linux-gnu/librockchip_mpp.so.1 (0x0000007f92d04000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f92cd5000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000007f92b4a000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f92a8d000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000007f92a69000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f928f7000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f92eda000)

lmw0320, 2022-10-21:
> 估计是 mpi_dec_test 更新了，64 位库更新了，但是实际跑到了 32 位的库，这个库没更新

大佬，有没解决方案？ 期待答复

HermanChen, 2022-10-21:
/lib/aarch64-linux-gnu/librockchip_mpp.so.1 看下这个库的版本号，有没有 _mpp_log_l 符号
ls -l /lib/aarch64-linux-gnu/librockchip_mpp*

lmw0320, 2022-10-21:
> /lib/aarch64-linux-gnu/librockchip_mpp.so.1 看下这个库的版本号，有没有 _mpp_log_l 符号 ls -l /lib/aarch64-linux-gnu/librockchip_mpp*

查看库的版本是这样么？
strings /lib/aarch64-linux-gnu/librockchip_mpp.so.1 |grep GLIB
结果是：
GLIBCXX_3.4
GLIBC_2.17

另外，ls -l /lib/aarch64-linux-gnu/librockchip_mpp*的执行结果：
lrwxrwxrwx 1 root root      20 Feb 22  2021 /lib/aarch64-linux-gnu/librockchip_mpp.so -> librockchip_mpp.so.1
-rw-r--r-- 1 root root 1535216 Feb 22  2021 /lib/aarch64-linux-gnu/librockchip_mpp.so.0
lrwxrwxrwx 1 root root      20 Feb 22  2021 /lib/aarch64-linux-gnu/librockchip_mpp.so.1 -> librockchip_mpp.so.0
-rw-r--r-- 1 root root 3239362 Feb 22  2021 /lib/aarch64-linux-gnu/librockchip_mpp_static.a

HermanChen, 2022-10-21:
librockchip_mpp.so.0 才是真正的库文件
strings /lib/aarch64-linux-gnu/librockchip_mpp.so.0 | grep author

lmw0320, 2022-10-21:
> strings /lib/aarch64-linux-gnu/librockchip_mpp.so.0 | grep author

执行了这个命令，返回是空的

lmw0320, 2022-10-25:
没有办法解决了么？？ 我试了好几次重编译，还是老问题啊。。

lmw0320, 2022-10-25:
上传编译过程截图：---原来第一次执行 ./make-Makefiles.bash命令时，还有提示：Looking for pthread_create in pthreads - not found。 我后面自己百度，根据博客的建议，执行了：sudo apt install doxygen，就没有那个提示了。但是仍旧不行。。

[image]
[image]
[image]

HermanChen, 2022-10-25:
> > strings /lib/aarch64-linux-gnu/librockchip_mpp.so.0 | grep author
> 
> 执行了这个命令，返回是空的

不应该啊，先看下文件有没有实体大小，正常带 git 编译的时候，会把版本信息写到库里

HermanChen, 2022-10-25:
另外，看这个 log 是把库安装到 /usr/local/lib/ 目录下，是不是要安装到 /usr/local/lib64/ 下才是对的？

lmw0320, 2022-10-25:
[image]

lmw0320, 2022-10-25:
> 另外，看这个 log 是把库安装到 /usr/local/lib/ 目录下，是不是要安装到 /usr/local/lib64/ 下才是对的？

我没去特地指定安装目录。。这个在哪里设定的？

HermanChen, 2022-10-25:
测试的话，就拷贝过去试试看，把 local 路径的库更新到 aarch64 下面去

lmw0320, 2022-10-25:
我后面肯定要直接用mpp来编解码视频文件的啊。。:（....
目前是因为对模型测试都好了，就是要加速视频的编解码，不然延时太厉害了。。

lmw0320, 2022-10-25:
> 测试的话，就拷贝过去试试看，把 local 路径的库更新到 aarch64 下面去

不是很明白你的意思，你是说把/usr/local/lib下面的那些so库，放到哪里的aarch64？？

HermanChen, 2022-10-25:
是啊。。。。。

lmw0320, 2022-10-25:
> 是啊。。。。。

我是拷贝了 /usr/local/lib下的所有so文件到了/lib/aarch64-linux-gnu/下  ： sudo cp *.so* /lib/aarch64-linux-gnu/
然后执行了mpi_enc_test，是不报错了。。。但是也没有任何信息返回。。

HermanChen, 2022-10-25:
export mpp_syslog_perror=1

HermanChen, 2022-10-25:
log 是写到 syslog 里的

lmw0320, 2022-10-26:
早上来重启了板卡继续测试mpp，命令和昨天的一样，结果好像能跑了，但是输出视频是0字节。。截图出来是这样的：
[image]

lmw0320, 2022-10-26:
我刚也查了/var/log/syslog，情况截图如下：
[image]

lmw0320, 2022-10-26:
> export mpp_syslog_perror=1

想请教下，我这种情况说明mpp成功安装了么？？
另外就是，为什么执行mpi_dec_test的时候，为什么不能正确执行呢？？

lmw0320, 2022-10-26:
> log 是写到 syslog 里的

大佬，江湖救急啊。。

HermanChen, 2022-10-26:
这个 log 说明已经正确了，但是你的输入是 mp4 文件，需要把里面的 264 或者 265 祼码流数据解出来给 mpp 解码，不能输入这种带封装的数据，同时 mpp 输出的是 yuv 数据，不会转压成 mp4 文件
还有 unknow vpu service ioctl 的那个是个 warning，是 mpp 检查内核 ioctl 支持列表时候的打印，新版本内核已经去掉了

lmw0320, 2022-10-27:
> 

哦哦，好的，很感谢多次的耐心答复！！

closed 2022-10-27 by HermanChen