rockchip-linux/mpp #274: 关于使用例程报错： mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory

lozenyin, 2022-02-18:
本人使用源码进行编译安装，过程很顺利
但是编译完成后使用例程测试是出现错误
`
rpdzkj@ubuntu:~/locke/test/mpp/build/linux/arm/test$ sudo ./mpi_enc_test -w 1280 -h 720 -t 7 -i /home/rpdzkj/locke/test_yuv.yuv -o ts.h264 -n 10
mpp[1776]: mpi_enc_utils: cmd parse result:
mpp[1776]: mpi_enc_utils: input  file name: /home/rpdzkj/locke/test_yuv.yuv
mpp[1776]: mpi_enc_utils: output file name: ts.h264
mpp[1776]: mpi_enc_utils: width      : 1280
mpp[1776]: mpi_enc_utils: height     : 720
mpp[1776]: mpi_enc_utils: format     : 0
mpp[1776]: mpi_enc_utils: type       : 7
mpp[1776]: mpi_enc_test: mpi_enc_test start
mpp[1776]: mpp_rt: NOT found ion allocator
mpp[1776]: mpp_rt: found drm allocator
mpp[1776]: mpp_info: mpp version: 3c5dc7fe author: sayon.chen    2020-11-23 [h265d_syntax]: Fix h265 syntax fill sps rps issue
mpp[1776]: mpi_enc_test: 0x511a80 mpi_enc_test encoder test start w 1280 h 720 type 7
mpp[1776]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [1280:720] stride [1280:720]
mpp[1776]: h264e_api_v2: MPP_ENC_SET_RC_CFG bps 3456000 [216000 : 3672000] fps [30:30] gop 60
mpp[1776]: mpp_enc_v2: send header for set cfg change input/format
mpp[1776]: mpp_enc_v2: mode vbr bps [216000:3456000:3672000] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_wait poll cmd failed 5
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_wait:980  failed return -1009
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return -1009
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 0    size 206
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 1    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 2    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 3    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 4    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 5    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 6    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 7    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 8    size 244
mpp[1776]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[1776]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_hal_start:977  failed return 12
mpp[1776]: mpp_enc_v2: mpp 0x511ac8 mpp_enc_normal:1372 failed return 12
mpp[1776]: mpi_enc_test: 0x511a80 encoded frame 9    size 244
mpp[1776]: mpi_enc_test: 0x511a80 encode max 10 frames
mpp[1776]: mpi_enc_test: 0x511a80 mpi_enc_test success total frame 10 bps 57648

`

HermanChen, 2022-02-18:
估计是内核驱动不匹配，mpp_platform_test 跑下看看

lozenyin, 2022-02-18:
那这个问题我该怎么解决？

lozenyin, 2022-02-18:
rpdzkj@ubuntu:~/locke/test/mpp/build/linux/arm/osal/test$ sudo ./mpp_platform_test
mpp[803]: mpp_plat_test: chip name: rockchip,rv1109-evb-ddr3-v12-facial-gate rockchip,rv1109
mpp[803]: mpp_plat_test:
mpp[803]: mpp_plat_test: chip vcodec type 1005020a
mpp[803]: mpp_plat_test: found vpu2 codec
mpp[803]: mpp_plat_test: found rkvdec decoder
mpp[803]: mpp_plat_test: found rkvenc encoder
mpp[803]: mpp_plat_test: found vpu2 encoder
mpp[803]: mpp_plat_test:
mpp[803]: mpp_plat_test: start probing decoder device name:
mpp[803]: mpp_plat_test: H.264 decoder: /dev/mpp_service
mpp[803]: mpp_plat_test: H.265 decoder: /dev/mpp_service
mpp[803]: mpp_plat_test: MJPEG decoder: /dev/mpp_service
mpp[803]: mpp_plat_test: VP9   decoder: (null)
mpp[803]: mpp_plat_test: avs   decoder: (null)
mpp[803]: mpp_plat_test:
mpp[803]: mpp_plat_test: start probing encoder device name:
mpp[803]: mpp_plat_test: H.264 encoder: /dev/mpp_service
mpp[803]: mpp_plat_test: H.265 encoder: /dev/mpp_service
mpp[803]: mpp_plat_test: MJPEG encoder: /dev/mpp_service
mpp[803]: mpp_plat_test: mpp platform test done

lozenyin, 2022-02-18:
这里打印是1109，可是我用的是RV1126来的。如果是内核驱动不匹配的话，我要采取什么措施呢？

closed 2022-02-18 by lozenyin

lozenyin, 2022-02-18:
但是在此之前他是能运行的哦，我曾经编译后成功运行起来例程，但是在我安装RGA库后想测试自己的代码就发现了这个问题了

HermanChen, 2022-02-18:
正常应该可以，1109  和 1126 只是支持分辨率上有区别，可以试下解码能不能起来

lozenyin, 2022-02-18:
rpdzkj@ubuntu:~/locke/test/mpp/build/linux/arm/test$ sudo ./mpi_dec_test -t 7 -i /home/rpdzkj/locke/test11.h264 -n 10
[sudo] password for rpdzkj:
mpp[2140]: mpi_dec_test: cmd parse result:
mpp[2140]: mpi_dec_test: input  file name: /home/rpdzkj/locke/test11.h264
mpp[2140]: mpi_dec_test: output file name:
mpp[2140]: mpi_dec_test: config file name:
mpp[2140]: mpi_dec_test: width      :    0
mpp[2140]: mpi_dec_test: height     :    0
mpp[2140]: mpi_dec_test: type       : 7
mpp[2140]: mpi_dec_test: debug flag : 0
mpp[2140]: mpi_dec_test: max frames : 10
mpp[2140]: mpi_dec_test: mpi_dec_test start
mpp[2140]: mpi_dec_test: input file size 12020629
mpp[2140]: mpp_info: mpp version: 3c5dc7fe author: sayon.chen    2020-11-23 [h265d_syntax]: Fix h265 syntax fill sps rps issue
mpp[2140]: mpi_dec_test: 0x4c42e0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[2140]: mpp_rt: NOT found ion allocator
mpp[2140]: mpp_rt: found drm allocator
mpp[2140]: mpi_dec_test: 0x4c42e0 decode_get_frame get info changed found
mpp[2140]: mpi_dec_test: 0x4c42e0 decoder require buffer w:h [1920:1080] stride [1920:1088] buf_size 4177920
mpp[2140]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[2140]: hal_h264d_rkv_reg: rkv_h264d_start send cmd failed 12
mpp[2140]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[2140]: hal_h264d_rkv_reg: rkv_h264d_wait poll cmd failed 5
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 0 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 1 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 2 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 3 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 4 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 5 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 6 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 7 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 8 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 decode get frame 9 err 1 discard 0
mpp[2140]: mpi_dec_test: 0x4c42e0 reach max frame number 10
mpp[2140]: mpi_dec_test: 0x4c42e0 input 50 pkt output 11 frm decode 10 frames
mpp[2140]: mpi_dec_test: test success max memory 19.92 MB

解码的倒是显示了成功了

lozenyin, 2022-02-18:
解码的那个还是失败的了，显示err 1

HermanChen, 2022-02-18:
估计还是内核驱动匹配问题，有旧的能跑的内核么？

HermanChen, 2022-02-18:
是的

HermanChen, 2022-02-18:
估计是推的库和之前的有不同，你是更新的 librockchip_mpp.so.0 么

HermanChen, 2022-02-19:
不需要，mpp 是直接调用内核的 drm ioctl 的

lozenyin, 2022-02-19:
它现在的报错显示无法分配内存就是说mpp与内核的drm连接处出问题了？
以下是我的编译脚本，在原来的基础上添加了一下编译参数

#!/bin/bash
# Run this from within a bash shell

set +e
MPP_PWD=`pwd`
MPP_TOP=${MPP_PWD}/../../..
# toolchain detection
check_cmd(){
    "$@" >> /dev/null 2>&1
}
check_system_arm_linux_gcc(){
    check_cmd arm-linux-gcc -v
}

check_system_arm_linux_gcc
if [ $? -eq 127 ];then
    MPP_TOOLCHAIN=${MPP_TOP}/../prebuilts/toolschain/usr/bin
    export PATH=$PATH:${MPP_TOOLCHAIN}
fi

# generate Makefile
cmake -DCMAKE_BUILD_TYPE=Release \
      -DCMAKE_TOOLCHAIN_FILE=./arm.linux.cross.cmake \
      -DRKPLATFORM=ON \
      -DHAVE-DRM=ON \
      -DCMAKE_INSTALL_PREFIX=/usr/lib \
      -G "Unix Makefiles" \
      ${MPP_TOP}

您看看有什么不对的地方？谢谢！

lozenyin, 2022-02-19:
我尝试去掉HAVE_DRM在cmake 发现在没有HAVE_DRM的情况下会报Performing Test CMAKE_HAVE_LIBC_PTHREAD 失败

rpdzkj@ubuntu:~/locke/lib/mpp/build/linux/arm$ sudo ./make-Makefiles.bash
-- cmake version 3.16.3
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/arm-linux-gnueabihf-gcc
-- Check for working C compiler: /usr/bin/arm-linux-gnueabihf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /usr/bin/arm-linux-gnueabihf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detected ARMv7 system processor
-- Performing Test GCC_HAS_NO_NARROWING
-- Performing Test GCC_HAS_NO_NARROWING - Success
-- Performing Test GCC_HAS_STACK_REALIGN
-- Performing Test GCC_HAS_STACK_REALIGN - Failed
-- Found Git: /usr/bin/git (found version "2.25.1")
-- current version:
-- 3c5dc7fe author: sayon.chen    2020-11-23 [h265d_syntax]: Fix h265 syntax fill sps rps issue
-- git version history:
-- 3c5dc7fe author: sayon.chen    2020-11-23 [h265d_syntax]: Fix h265 syntax fill sps rps issue  (HEAD -> release, origin/release)
-- 472a0b75 author: sayon.chen    2020-11-23 [rc_v2]: Fix pre_i_bit struct leak issue
-- db2492ed author: Justa         2020-11-23 [hal_jpege_vepu]: Fixed jpeg encode abnormal color
-- a6c4efe0 author: yandong.lin   2020-11-20 [hal_vp9d]: Reconstruct HalVp9dCtx Structure
-- 36e20cff author: yandong.lin   2020-11-20 [vdpu34x]: Update register naming
-- 06ef6826 author: yandong.lin   2020-11-20 [mpp_init]: Handle initialization failures correctly
-- 4ddca651 author: Herman Chen   2020-11-19 [hal_vp9d_vdpu34x]: Add vdpu34x vp9 decoder
-- 65eb5f98 author: Herman Chen   2020-11-19 [vdpu34x]: Add vdpu34x rcb buffer alloc function
-- 2a6fb9fe author: sayon.chen    2020-11-19 [vdpu34x]: Support cmv alone set for h264&hevc
-- ca353b87 author: Herman Chen   2020-11-18 [hal_bufs]: Enlarge max buffer elem count
-- total 10 git version recorded
-- Install git hooks done
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for one of the modules 'pthread'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- compile without drm support
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    RKPLATFORM

您说的log是要加在代码上还是在编译过程中？

lozenyin, 2022-02-19:
rpdzkj@ubuntu:~/locke/lib/mpp/build/linux/arm/osal/test$ sudo ./mpp_mem_test
mpp[2023]: mpp_mem_test: calloc  success ptr 0x0x4cb0c0
mpp[2023]: mpp_mem_test: realloc success ptr 0x0x4cd558
mpp[2023]: mpp_mem_test: mpp_mem_test done

这个osel例程是测试什么的呢？

HermanChen, 2022-02-21:
加log可以加#error 在编译之前就判断
mem 是测试 malloc 的，测试硬件用的 buffer 是 mpp_buffer_test