rockchip-linux/mpp #275: rk3568运行mpi_enc_test

Csf-mycount, 2022-02-25:
请问这是什么问题呢
./mpi_enc_test -w 640 -h 272 -t 7 -i /userdata/video.yuv -o /userdata/jj.h264 -n 10
mpp[4146]: mpi_enc_utils: cmd parse result:
mpp[4146]: mpi_enc_utils: input  file name: /userdata/video.yuv
mpp[4146]: mpi_enc_utils: output file name: /userdata/jj.h264
mpp[4146]: mpi_enc_utils: width      : 640
mpp[4146]: mpi_enc_utils: height     : 272
mpp[4146]: mpi_enc_utils: format     : 0
mpp[4146]: mpi_enc_utils: type       : 7
mpp[4146]: mpi_enc_test: mpi_enc_test start
mpp[4146]: mpp_rt: NOT found ion allocator
mpp[4146]: mpp_rt: found drm allocator
mpp[4146]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[4146]: mpi_enc_test: 0x31343a00 mpi_enc_test encoder test start w 640 h 272 type 7
mpp[4146]: mpp_enc: MPP_ENC_SET_RC_CFG bps 652800 [40800 : 693600] fps [30:30] gop 60
mpp[4146]: h264e_api_v2: MPP_ENC_SET_PREP_CFG w:h [640:272] stride [640:272]
mpp[4146]: mpp_enc: send header for set cfg change input/format 
mpp[4146]: mpp_enc: mode vbr bps [40800:652800:693600] fps fix [30/1] -> fix [30/1] gop i [60] v [0]
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: mpp_serivce: mpp_service_cmd_poll ioctl MPP_IOC_CFG_V1 failed ret -1 errno 5 Input/output error
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_wait poll cmd failed 5
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_wait:1260 failed return -1009
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return -1009
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 0    size 151    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 1    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 2    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 3    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 4    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 5    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 6    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 7    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 8    size 189    
mpp[4146]: mpp_serivce: mpp_service_cmd_send ioctl MPP_IOC_CFG_V1 failed ret -1 errno 12 Cannot allocate memory
mpp[4146]: hal_h264e_vepu541: hal_h264e_vepu541_start send cmd failed 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_hal_start:1257 failed return 12
mpp[4146]: mpp_enc: mpp 0x31343a70 mpp_enc_normal:1820 failed return 12
mpp[4146]: mpi_enc_test: 0x31343a00 encoded frame 9    size 189    
mpp[4146]: mpi_enc_test: 0x31343a00 encode max 10 frames
mpp[4146]: mpi_enc_test: 0x31343a00 mpi_enc_test success total frame 10 bps 44448

HermanChen, 2022-02-25:
应该是用户态的 mpp 库和内核不匹配吧

Csf-mycount, 2022-02-25:
那么手册上有讲内核需要怎么配置吗

Csf-mycount, 2022-02-28:
如何查看是否匹配呢?

Csf-mycount, 2022-03-07:
在同一套sdk的mpp编译后在同个环境下运行没有这个问题了

closed 2022-03-07 by Csf-mycount