rockchip-linux/mpp #281: Any example code demonstrating multi-stream decoding/encoding without display?

changtimwu, 2022-03-17:
Here is one of the RK3588 feature highlights 
[image]

While 32-stream decoding and 16-stream encoding might look powerful, I couldn't find any code clip demonstrating more than 2 stream decoding.

I'd like to know if it's doable in RK 35xx series.  Thanks a lot.

changtimwu, 2022-03-17:
Sorry, maybe I should try standard gst command first.  Still waiting for my RK board arrival. Thanks,though.

closed 2022-03-17 by changtimwu

HermanChen, 2022-03-18:
Use mpi_dec_multi_test or mpi_dec_test with -s option you can decode multi stream at one time.