rockchip-linux/mpp #282: 测试rv1126编码性能问题

liyungithub, 2022-03-22:
测试环境：rv1126板子
测试工具：mpi_enc_multi_test

使用mpi_enc_multi_test来测试rv1126的编码性能，目的想知道rv1126以H264编码能编码多少路1080P的视频？
目前的测试结果是：编码1080P视频成264平均帧率是30帧，但和官方给出性能参数不符，接着我又试了跑两个mpi_enc_multi_test程序，结果两个程序输出的结果平均帧率都是30左右，所有我就有疑问了，怎么去理解mpi_enc_multi_test这个程序的测试结果？我现在想测试rv1126编码性能应该怎么做？
谢谢！

HermanChen, 2022-03-22:
要分析mpi_enc_multi_test里做了什么，里面有 cpu 画图，有编码，瓶颈估计是 cpu 画图的时间，要评估时间的话要看硬件运行时间：
echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug
然后跑用例看下打印

closed 2022-04-01 by liyungithub

liyungithub, 2022-04-13:
再请教你几个问题 @HermanChen ：
1、针对ROI来设置QP值是不是只有H265编码才有？H264编码有吗？
2、如果多个ROI重叠，那设置的QP值以哪个为准？
3、设置ROI参数来编码时，可以针对每一张图片都设置不同的ROI参数吗？还是说编码前就要设置好ROI参数，编码过程不能改

liyungithub, 2022-04-18:
@HermanChen 麻烦解答下上面问题？谢谢了
再补充一个：
4、roi的数量上限是多少？

HermanChen, 2022-04-18:
1. ROI 是通用接口 H.264 和 H.265 都有
2. ROI 有场次，index 最大，优先级越高，接口上最多 8 个，KEY_ROI_DATA 来配置
3. ROI 是帧级配置的，每帧都是可以变化
4. ROI 在硬件上是全局 buffer 配置，一帧按块划分，数据写到 buffer 里，所以上限是每个块一个配置，通过 KEY_ROI_DATA2 来配置就可以在外部生成好数据，配置给编码器

liyungithub, 2022-04-18:
好的，谢谢了！

liyungithub, 2022-05-07:
在rv1126上解码jpg文件不成功，内核有unknown mpp ioctl cmd 403打印，是不是有什么库版本不一致问题?@HermanChen
[image]
[image]

liyungithub, 2022-05-09:
@HermanChen 麻烦看看这问题？谢谢！

HermanChen, 2022-05-09:
unknown mpp ioctl cmd 403
这个是兼容性检查打印，估计是内核版本比较旧
看下内核有没有办法更新下版本

liyungithub, 2022-05-09:
好的，我试下，谢谢了！

liyungithub, 2022-05-09:
> unknown mpp ioctl cmd 403 这个是兼容性检查打印，估计是内核版本比较旧 看下内核有没有办法更新下版本

下载了最新的固件烧录，这问题解决了。

liyungithub, 2022-05-09:
还遇到了一个新问题:
使用rga操作的时候没有成功，dmsg有rga2_reg_init, [880] set mmu info error 打印。
该测试demo在1808运行成功的，但在1126就出错了
[image]

liyungithub, 2022-05-09:
@HermanChen 再一次麻烦你了，谢谢！

HermanChen, 2022-05-09:
rga 找一下 github 上 librga 的人员处理，mpp 里默认是不使用 rga 的

liyungithub, 2022-05-10:
> rga 找一下 github 上 librga 的人员处理，mpp 里默认是不使用 rga 的

谢谢了，问题已解决

liyungithub, 2022-06-17:
请教下stride尺寸怎么算的，1080PH264解码后是stride[1920:1088]
[image]
1080PH264解码后是stride[2304:1080]
[image]
1080PH264编码后是stride[1920:1088]
[image]
1080PH264编码后是stride[1920:1080]
[image]
最开始我以为MPP_ALIGN(width, 16)算16字节对齐就行，但现在看来不是，我理解还是有问题的，那应该怎么理解stride尺寸或怎么算stride尺寸呢？ @HermanChen 麻烦解答下，谢谢！

liyungithub, 2022-06-22:
@HermanChen 麻烦解答下上面问题？谢谢

HermanChen, 2022-06-22:
https://www.doc88.com/p-1794847173765.html
参考下开发文档