rockchip-linux/mpp #289: 使用camera_source代码，mpi_enc_test -w 640 -h 480 -t 7 -f 8 -i /dev/video0 -o /userdata/1131.h264，画面有横细线

Csf-mycount, 2022-04-07:
[image]
请问怎么解决呢，测试过无横细线单独yuv编码成h264播放没问题，说明编码是没问题的，问题应该出在v4l2?

HermanChen, 2022-04-07:
是否有 cpu 拷贝？看起来是 cpu 拷贝的 cache 问题

Csf-mycount, 2022-04-07:
没有拷贝，单纯的从摄像头中获取图像编码成h264文件播放会如上图，然后测试了一下只拿yuv格式图像一帧一帧看又没问题，然后又测试用yuv格式视频通过mpi_enc_test编码成h264也没问题

HermanChen, 2022-04-08:
估计还是有 cpu 拷贝的过程，或者其他器件在做拷贝，这种的拉丝比较像是 cache 问题，或者是 buffer 管理有问题，同一个 buffer 被同时读写了

Csf-mycount, 2022-04-20:
我是直接应用mpi_enc_test.c来测试，请问你们有同样测试过吗?

Csf-mycount, 2022-04-20:
还有，假如我在这个位置延时了一下就不会有横线，但是播放的视频会有快进的感觉
[image]

Csf-mycount, 2022-04-20:
ok 解决了
[image]

Csf-mycount, 2022-04-20:
已解决，用usb摄像头时候不使用mpp_frame_set_buffer(frame, cam_buf)，重新memcpy(buf, p->cam_ctx->fbuf[cam_frm_idx].start, p->width * p->height *2)， 再使用mpp_frame_set_buffer(frame, p->frm_buf)就不会有横线

closed 2022-04-20 by Csf-mycount