rockchip-linux/mpp #292: mpi_dec_test failed with latest MPP source code

zhouwg, 2022-04-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Host Operating System & Version |   {Ubuntu 20.04.2 } | 
| Branch                           |  { develop }                          | 
| Board                | RK3568                             |
| Board OS                               | Linux |
| Board OS Kernel Version    |  (4.19.161)                                         | 

### Issue Description

mpi_dec_test failed with latest MPP source code

### How to reproduce the issue

(1) mpi_dec_test sanity check with original librockchip_mpp.so.0

```
# uname -a
Linux RK356X 4.19.161 #4 SMP Tue Dec 14 14:56:37 CST 2021 aarch64 GNU/Linux

# ls -l /usr/lib/librockchip_*
lrwxrwxrwx 1 root root      20 Apr 17 09:04 /usr/lib/librockchip_mpp.so -> librockchip_mpp.so.1
-rwxr-xr-x 1 root root 5470632 Apr 18 02:04 /usr/lib/librockchip_mpp.so.0
-rwxr-xr-x 1 root root 5470632 Apr 18 01:50 /usr/lib/librockchip_mpp.so.0.org
lrwxrwxrwx 1 root root      20 Apr 17 09:04 /usr/lib/librockchip_mpp.so.1 -> librockchip_mpp.so.0
lrwxrwxrwx 1 root root      20 Apr 17 09:04 /usr/lib/librockchip_vpu.so -> librockchip_vpu.so.1
-rwxr-xr-x 1 root root  183000 Apr 18 01:56 /usr/lib/librockchip_vpu.so.0
-rwxr-xr-x 1 root root  193896 Apr 18 01:54 /usr/lib/librockchip_vpu.so.0.org
lrwxrwxrwx 1 root root      20 Apr 17 09:04 /usr/lib/librockchip_vpu.so.1 -> librockchip_vpu.so.0

# ffprobe /vdi/video/h264.bin 
ffprobe version N-106509-g21ca221068 Copyright (c) 2007-2022 the FFmpeg developers
  libavutil      57. 24.101 / 57. 24.101
  libavcodec     59. 25.100 / 59. 25.100
  libavformat    59. 20.101 / 59. 20.101
  libavdevice    59.  6.100 / 59.  6.100
  libavfilter     8. 30.100 /  8. 30.100
  libswscale      6.  6.100 /  6.  6.100
  libswresample   4.  6.100 /  4.  6.100
  libpostproc    56.  5.100 / 56.  5.100
Input #0, h264, from '/vdi/video/h264.bin':
  Duration: N/A, bitrate: N/A
  Stream #0:0: Video: h264 (Main), yuv420p(tv, smpte170m/smpte170m/bt709, progressive), 320x240, 25 fps, 25 tbr, 1200k tbn

# ./mpi_dec_test -i /vdi/video/h264.bin -o /vdi/video/test.yuv -w 320 -h 240 -t 7
mpp[29883]: mpp_opt: setup:real node 18:22 info 11:11
mpp[29883]: mpi_dec_utils: input file /vdi/video/h264.bin size 94815
mpp[29883]: mpi_dec_utils: cmd parse result:
mpp[29883]: mpi_dec_utils: input  file name: /vdi/video/h264.bin
mpp[29883]: mpi_dec_utils: output file name: /vdi/video/test.yuv
mpp[29883]: mpi_dec_utils: width      :  320
mpp[29883]: mpi_dec_utils: height     :  240
mpp[29883]: mpi_dec_utils: type       :    7
mpp[29883]: mpi_dec_utils: max frames :    0
mpp[29883]: mpi_dec_test: mpi_dec_test start
mpp[29883]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[29883]: mpi_dec_test: 0x3dfab440 mpi_dec_test decoder test start w 320 h 240 type 7
mpp[29883]: mpp_rt: NOT found ion allocator
mpp[29883]: mpp_rt: found drm allocator
mpp[29883]: mpi_dec_test: 0x3dfab440 decode_get_frame get info changed found
mpp[29883]: mpi_dec_test: 0x3dfab440 decoder require buffer w:h [320:240] stride [320:240] buf_size 153600
mpp[29883]: mpi_dec_test: 0x3dfab440 decode get frame 0
mpp[29883]: mpi_dec_test: 0x3dfab440 decode get frame 1
mpp[29883]: mpi_dec_test: 0x3dfab440 decode get frame 2
mpp[29883]: mpi_dec_test: 0x3dfab440 decode get frame 3
mpp[29883]: mpi_dec_test: 0x3dfab440 decode get frame 4
mpp[29883]: mpi_dec_test: 0x3dfab440 decode get frame 5
mpp[29883]: mpi_dec_test: 0x3dfab440 decode get frame 6
[... 342 lines cut ...]
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 136 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 137 err 0 discard 1
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 138 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 139 err 0 discard 1
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 140 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 141 err 0 discard 1
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 142 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 143 err 0 discard 1
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 144 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 145 err 0 discard 1
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 146 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 147 err 0 discard 1
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 148 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 decode get frame 149 err 1 discard 0
mpp[29903]: mpi_dec_test: 0x26111440 found last packet
mpp[29903]: mpi_dec_test: decode 150 frames time 6404 ms delay 2100 ms fps 23.42
mpp[29903]: mpi_dec_test: test success max memory 1.03 MB

```

### Help wanted

(1)pls help to clarify why this issue happened;
(2)shoul I use the latest MPP source code for the product?
(3)why /dev/vpu_service was not founded on RK3568 board?

HermanChen, 2022-04-18:
(3) /dev/vpu_service is the old device used before kernel 3.10. It is not used on kernel 4.4 and later.
(2) The ubunut code may need some patches?
(1) Can you add some log to trace the initial process or strace the check the drm buffer open / close process?

zhouwg, 2022-04-18:
mpi_dec_test sanity check with original librockchip_mpp.so:

```
[root@RK356X:/vdi/vendor/rk3568/rockchip-linux-mpp/out/arm64-v8a/test]# strace mpi_dec_test -i /vdi/video/h264.bin -o /vdi/
video/test.yuv -w 320 -h 240 -t 7
execve("/usr/bin/mpi_dec_test", ["mpi_dec_test", "-i", "/vdi/video/h264.bin", "-o", "/vdi/video/test.yuv", "-w", "320", "-h", "240", "-t", "7"], 0x7fce330650 /* 20 vars */) = 0
brk(NULL)                               = 0x13a7c000
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=33734, ...}) = 0
mmap(NULL, 33734, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fbcb73000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/librockchip_mpp.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\2400\3\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=5470632, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbcb71000
mmap(NULL, 2065528, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fbc956000
mprotect(0x7fbcb2c000, 65536, PROT_NONE) = 0
mmap(0x7fbcb3c000, 73728, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1d6000) = 0x7fbcb3c000
mmap(0x7fbcb4e000, 1144, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fbcb4e000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0 d\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=2813968, ...}) = 0
mmap(NULL, 185360, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fbc928000
mprotect(0x7fbc941000, 61440, PROT_NONE) = 0
mmap(0x7fbc950000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x18000) = 0x7fbc950000
mmap(0x7fbc952000, 13328, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fbc952000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libstdc++.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\360-\f\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=10190336, ...}) = 0
mmap(NULL, 2359480, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fbc6e7000
mprotect(0x7fbc905000, 61440, PROT_NONE) = 0
mmap(0x7fbc914000, 65536, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x21d000) = 0x7fbc914000
mmap(0x7fbc924000, 12472, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fbc924000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\320\267\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=2912056, ...}) = 0
mmap(NULL, 712816, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fbc638000
mprotect(0x7fbc6d6000, 61440, PROT_NONE) = 0
mmap(0x7fbc6e5000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9d000) = 0x7fbc6e5000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libgcc_s.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0 .\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=538192, ...}) = 0
mmap(NULL, 214120, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fbc603000
mprotect(0x7fbc627000, 61440, PROT_NONE) = 0
mmap(0x7fbc636000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x23000) = 0x7fbc636000
close(3)                                = 0
openat(AT_FDCWD, "/lib/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0000\17\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=17813704, ...}) = 0
mmap(NULL, 1491048, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fbc496000
mprotect(0x7fbc5e9000, 65536, PROT_NONE) = 0
mmap(0x7fbc5f9000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x153000) = 0x7fbc5f9000
mmap(0x7fbc5ff000, 12392, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fbc5ff000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbcb6f000
mprotect(0x7fbc5f9000, 16384, PROT_READ) = 0
mprotect(0x7fbc636000, 4096, PROT_READ) = 0
mprotect(0x7fbc6e5000, 4096, PROT_READ) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbc494000
mprotect(0x7fbc914000, 45056, PROT_READ) = 0
mprotect(0x7fbc950000, 4096, PROT_READ) = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fbc491000
mprotect(0x7fbcb3c000, 12288, PROT_READ) = 0
mprotect(0x41d000, 4096, PROT_READ)     = 0
mprotect(0x7fbcb7e000, 4096, PROT_READ) = 0
[... 782 lines cut ...]
futex(0x322d8eb4, FUTEX_WAKE_PRIVATE, 1) = 1
futex(0x322d8e58, FUTEX_WAKE_PRIVATE, 1) = 1
munmap(0x7fa2186000, 2101248)           = 0
munmap(0x7fa2085000, 1052672)           = 0
munmap(0x7fa1e84000, 2101248)           = 0
close(6)                                = 0
munmap(0x7fa016c000, 159744)            = 0
munmap(0x7fa0193000, 159744)            = 0
munmap(0x7fa022f000, 159744)            = 0
munmap(0x7fa01e1000, 159744)            = 0
munmap(0x7fa0458000, 159744)            = 0
munmap(0x7fa01ba000, 159744)            = 0
[... cut ...]
close(5)                                = 0
getpid()                                = 790
writev(2, [{iov_base="mpp[790]: mpi_dec_test: test suc"..., iov_len=56}], 1mpp[790]: mpi_dec_test: test success max memory 1.03 MB
) = 56
sendto(4, "<14>Apr 18 10:07:41 mpp[790]: mp"..., 76, MSG_NOSIGNAL, NULL, 0) = 76
close(3)                                = 0
close(4)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++
[root@RK356X:/vdi/vendor/rk3568/rockchip-linux-mpp/out/arm64-v8a/test]# 
```

HermanChen, 2022-04-18:
确认一下最新代码编译的时候，有没有加上 HAVE_DRM 配置？

HermanChen, 2022-04-18:
看到少了 DRM_IOCTL_MODE_CREATE_DUMB 的分配内存的调用，可以在 drm_alloc 里加下打印看看，有没有走进去

HermanChen, 2022-04-18:
新版本的库只打开一次 /dev/dri/card0 设备，在同一个进程使用一个 drm fd 来做为主设备

zhouwg, 2022-04-18:
> 确认一下最新代码编译的时候，有没有加上 HAVE_DRM 配置？

谢谢！

zhouwg, 2022-04-18:
> 新版本的库只打开一次 /dev/dri/card0 设备，在同一个进程使用一个 drm fd 来做为主设备

您好，MPP的这个linux demo 

[https://github.com/WainDing/mpp_linux_cpp](https://github.com/WainDing/mpp_linux_cpp)

在rk3568开发板上无法正常运行。

按照您的回复（添加HAVE_DRM=on)重新编译了MPP代码生成mpp库，以及改了mpp_linux_cpp的代码（主要是将RGA相关的全部注释掉，使用drm_show_frame这个函数来显示解码后的 video frame)，可以正常运行，但是屏幕黑屏，请问这是什么原因？是需要修改drm_show_frame的实现吗？ 贵公司有这样的example吗（使用drm将解码后的video_frame显示在video layer上）？

HermanChen, 2022-04-19:
那个库也是我们同事的仓库，可以直接问下他

zhouwg, 2022-04-19:
> 那个库也是我们同事的仓库，可以直接问下他

感谢！

closed 2022-04-19 by zhouwg