rockchip-linux/mpp #296: 通过交叉编译MPP库，用到的工具链是gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf.tar，移植编译出来的动态库和test文件夹到rv1126，测试解码mpi_dec_test返回ret -1，烦请解决。

huaxin233, 2022-05-20:
root@debian10:/usr/local/lib# mpi_dec_test -i /home/toybrick/test.h264 -t 7 -n 10
mpp[813]: mpi_dec_utils: input file /home/toybrick/test.h264 size 3032275
mpp[813]: mpi_dec_utils: cmd parse result:
mpp[813]: mpi_dec_utils: input  file name: /home/toybrick/test.h264
mpp[813]: mpi_dec_utils: output file name: 
mpp[813]: mpi_dec_utils: width      :    0
mpp[813]: mpi_dec_utils: height     :    0
mpp[813]: mpi_dec_utils: type       :    7
mpp[813]: mpi_dec_utils: max frames :   10
mpp[813]: mpi_dec_test: mpi_dec_test start
mpp[813]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[813]: mpi_dec_test: 0x351c0 mpi_dec_test decoder test start w 0 h 0 type 7
mpp[813]: mpp_rt: NOT found ion allocator
mpp[813]: mpp_rt: found drm allocator
mpp[813]: mpp: command 310202 param 0x45520 ret -1
mpp[813]: mpi_dec_test: 0x351c0 failed to get decoder cfg ret -1
mpp[813]: mpi_dec_test: test failed ret -1

HermanChen, 2022-05-20:
更新下代码看看
另外，mpp_info: mpp version: unknown mpp version for missing VCS info 这里看不到 mpp 的版本信息

huaxin233, 2022-05-20:
会不会动态库没有起来，在/home/toybrick/.bashrc最后加了如下两句，然后source一下
export PATH=/usr/local/bin/mpp_bin:$PATH
export LD_LIBRARY_PATH=/usr/local/lib/mpp_lib:$LD_LIBRARY_PATH

huaxin233, 2022-05-20:
> 更新下代码看看 另外，mpp_info: mpp version: unknown mpp version for missing VCS info 这里看不到 mpp 的版本信息

更新代码指重新下载MPP库编译吗？

huaxin233, 2022-05-23:
> 更新下代码看看 另外，mpp_info: mpp version: unknown mpp version for missing VCS info 这里看不到 mpp 的版本信息
编译了新库提示这个
root@debian10:/home/toybrick# mpi_dec_test 
mpi_dec_test: symbol lookup error: mpi_dec_test: undefined symbol: _mpp_log_l

Bro-Zero, 2022-06-23:
> > 更新下代码看看 另外，mpp_info: mpp version: unknown mpp version for missing VCS info 这里看不到 mpp 的版本信息
> > 编译了新库提示这个
> > root@debian10:/home/toybrick# mpi_dec_test
> > mpi_dec_test: symbol lookup error: mpi_dec_test: undefined symbol: _mpp_log_l

可以试一下把编译后的 librockchip_mpp.so* 和 librockchip_vpu.so* 文件放到板子的 /usr/lib/ 目录下（注意.so*文件间有链接关系）

ps2chiper, 2022-06-30:
I also have this issue.