rockchip-linux/mpp #299: video decoding on RK3128 doesn't work

sergei-gagarin, 2022-05-24:
I have compiled mpp and vpu libraries from this repository on RK3128 board (kernel 3.10, Debian 10, armhf), compiled my video decoding test application ([ffrtsp.zip](https://github.com/rockchip-linux/mpp/files/8766832/ffrtsp.zip)) with these libraries and ffmpeg libraries. But it does not work:

1) When I try to play H.264 file then function decode() of VPU context only 1 time returns empty VPU_FRAME and then stucks. Using debugger I found that it stucks in dead loop in function mpi_decode() (mpp/mpi.cpp). Also I noted that in function try_proc_dec_task() (mpp/codec/mpp_dec.cpp) execution never continues after:
    task->wait.info_change = mpp_buf_slot_is_changed(frame_slots);
    if (task->wait.info_change) {
        return MPP_ERR_STREAM;
    } else {
because task->wait.info_change always is not 0. So VPU's ioctl VPU_IOC_SET_REG is never called and no mesage "vpu_service: power on" in kernel log.

2)  When I try to play MPEG4 file then it looks like to be working but looses many frames (for example it decodes only 40 frames from 100) and frames are often damaged (have square block artifacts).

When I use libvpu.so from Android distro (with machybris package) then my application works properly (almost all frames are decoded and they does not have any artifact).

Any ideas what is wrong?

HermanChen, 2022-05-25:
1. This is better to use mpi interface rather than old vpu interface. Try mpi_dec_test to decode the raw stream.
2. mpp can not accept mp4 file. Use ffmpeg to extract the raw stream and use mpi_dec_test to decode it.

The 3.10 kernel and 3128 is quite old and may have compatible issue on latest mpp. But if it works fine on android then it should works fine with mpi_dec_test.

sergei-gagarin, 2022-05-25:
OK, thanks!
I have reworked program to use mpi interface and it works.
It seems that libvpu interface is broken.