rockchip-linux/mpp #300: rk3399 kernel panic when doing decode

Allenhe123, 2022-05-25:
Application Scenario:
RK3399 receive 4 camera's rtsp stream, do decode, do scale using RGA. 
LOG：

0410_03-08-12.384SOM-RK3399v2 login: [   48.042123] Unable to handle kernel NULL pointer dereference at virtual address 00000000
0410_03-08-12.384[   48.042887] pgd = ffffffc0e270c000
0410_03-08-12.384[   48.043205] [00000000] *pgd=0000000000000000, *pud=0000000000000000
0410_03-08-12.384[   48.043801] Internal error: Oops: 96000045 [#1] SMP
0410_03-08-12.384[   48.044240] Modules linked in: pcie_com1 pcie_com0 rk3399_gpio_heart bcmdhd ip_tables x_tables
0410_03-08-12.384[   48.045067] CPU: 1 PID: 904 Comm: quadways_decode Not tainted 4.4.179-00001-gb8c28354 #6
0410_03-08-12.384[   48.045783] Hardware name: FriendlyElec SOM-RK3399 (DT)
0410_03-08-12.384[   48.046250] task: ffffffc0e5754600 task.stack: ffffffc0e98cc000
0410_03-08-12.384[   48.046788] PC is at rb_erase+0x120/0x3a8
0410_03-08-12.384[   48.047153] LR is at timerqueue_del+0x38/0x88
0410_03-08-12.384[   48.047538] pc : [<ffffff800842ac70>] lr : [<ffffff800842d628>] pstate: a00001c5
0410_03-08-12.385[   48.048184] sp : ffffffc0f7ee1e00
0410_03-08-12.385[   48.048487] x29: ffffffc0f7ee1e00 x28: ffffffc0e5754600 
0410_03-08-12.385[   48.048981] x27: ffffff80091d2168 x26: ffffff8009090000 
0410_03-08-12.385[   48.049474] x25: 0000000000000001 x24: ffffff8009090000 
0410_03-08-12.385[   48.049968] x23: 0000000000000000 x22: ffffffc0f7ee3c80 
0410_03-08-12.385[   48.050462] x21: 0000000000000000 x20: ffffffc0f7ee3d10 
0410_03-08-12.385[   48.050953] x19: ffffffc0f7f23140 x18: 0000007f627fb4c7 
0410_03-08-12.385[   48.051443] x17: 0000007f78614f90 x16: ffffff8008119870 
0410_03-08-12.385[   48.051933] x15: 000000000000000a x14: 0000000000000008 
0410_03-08-12.385[   48.052423] x13: 7f807f807f807f80 x12: 7f807f807f807f80 
0410_03-08-12.385[   48.052912] x11: 7f807f807f807f80 x10: 0000000000000a20 
0410_03-08-12.385[   48.053401] x9 : 0000000000000000 x8 : ffffffc0e60000a0 
0410_03-08-12.385[   48.053890] x7 : ffffffc0e6000000 x6 : ffffffc0f7f23140 
0410_03-08-12.385[   48.054380] x5 : 0000000000000000 x4 : 0000000000000000 
0410_03-08-12.385[   48.054869] x3 : ffffffc0f7ee4141 x2 : ffffffc0f7ee4140 
0410_03-08-12.385[   48.055358] x1 : ffffffc0f7ee3d10 x0 : ffffffc0f7ee4140 
0410_03-08-12.385[   48.055850] 
0410_03-08-12.385[   48.055850] PC: 0xffffff800842abf0:
0410_03-08-12.385[   48.056292] abf0  aa0303e2 b4001423 f9400866 eb06001f 54001400 f9000465 b4001325 f90000a4
0410_03-08-12.386[   48.057056] ac10  d65f03c0 d503201f f9400002 f27ef443 f9000082 54001320 f9400861 eb01001f
0410_03-08-12.386[   48.057818] ac30  54001300 f9000464 d65f03c0 d503201f f9400060 f9000064 36000900 d2800003
0410_03-08-12.386[   48.058581] ac50  14000022 d503201f f9400003 37000203 f9400804 b2400043 f9000444 f9000802
0410_03-08-12.386[   48.059342] ac70  f9000083 f9400043 f9000003 f27ef463 f9000040 540007a0 f9400865 eb05005f
0410_03-08-12.386[   48.060104] ac90  540007c0 f9000460 aa0403e0 f9400404 b4000064 f9400083 36000903 f9400803
0410_03-08-12.386[   48.060866] acb0  b4000063 f9400064 36000b84 f9000002 f9400040 360006a0 f27ef400 54000520
0410_03-08-12.386[   48.061626] acd0  aa0203e3 aa0003e2 f9400440 eb03001f 54fffbc1 f9400840 f9400003 37000203
0410_03-08-12.386[   48.062389] 
0410_03-08-12.386[   48.062389] LR: 0xffffff800842d5a8:
0410_03-08-12.386[   48.062831] d5a8  a94153f3 a8c27bfd d65f03c0 90006d20 3944d001 35fffbe1 d4210000 52800021
0410_03-08-12.386[   48.063594] d5c8  3904d001 17ffffdb b40000c0 a9bf7bfd 910003fd 97fff71f a8c17bfd d65f03c0
0410_03-08-12.386[   48.064355] d5e8  d2800000 d65f03c0 a9be7bfd 910003fd a90153f3 aa0003f4 aa0103f3 f9400020
0410_03-08-12.386[   48.065116] d608  eb01001f 54000260 f9400680 eb00027f 54000180 aa1303e0 aa1403e1 97fff54b
0410_03-08-12.386[   48.065878] d628  f9000273 f9400680 a94153f3 f100001f 1a9f07e0 a8c27bfd d65f03c0 d503201f
0410_03-08-12.386[   48.066644] d648  aa1303e0 97fff703 f9000680 17fffff2 90006d20 9104d000 39400401 35fffd61
0410_03-08-12.387[   48.067409] d668  d4210000 52800021 39000401 17ffffe7 f9400001 b0003ea5 910460a5 91000421
0410_03-08-12.387[   48.068175] d688  52800002 52800146 aa0103e3 f9000001 91000421 39400063 385fe024 1b061042
0410_03-08-12.387[   48.068944] 
0410_03-08-12.387[   48.068944] SP: 0xffffffc0f7ee1d80:
0410_03-08-12.387[   48.069388] 1d80  f7ee3c80 ffffffc0 00000000 00000000 09090000 ffffff80 00000001 00000000
0410_03-08-12.387[   48.070154] 1da0  09090000 ffffff80 091d2168 ffffff80 e5754600 ffffffc0 f7ee1e00 ffffffc0
0410_03-08-12.387[   48.070917] 1dc0  0842d628 ffffff80 f7ee1e00 ffffffc0 0842ac70 ffffff80 a00001c5 00000000
0410_03-08-12.387[   48.071680] 1de0  f7ee1e00 ffffffc0 08128b98 ffffff80 00000000 00000080 00000000 00000000
0410_03-08-12.387[   48.072444] 1e00  f7ee1e20 ffffffc0 081186a4 ffffff80 f7f23140 ffffffc0 f7ee3d00 ffffffc0
0410_03-08-12.387[   48.073206] 1e20  f7ee1e60 ffffffc0 08118950 ffffff80 f7f23140 ffffffc0 f7ee3c80 ffffffc0
0410_03-08-12.394[   48.073970] 1e40  f7ee3d00 ffffffc0 ffffffff 7fffffff 090c5000 ffffff80 0888c348 ffffff80
0410_03-08-12.417[   48.074733] 1e60  f7ee1ee0 ffffffc0 08119238 ffffff80 f7ee3c80 ffffffc0 00000003 00000000
0410_03-08-12.417[   48.075499] 
0410_03-08-12.417[   48.075499] X0: 0xffffffc0f7ee40c0:
0410_03-08-12.417[   48.075940] 40c0  00000000 dead4ead ffffffff 00000000 ffffffff ffffffff f7ee40d8 ffffffc0
0410_03-08-12.417[   48.076705] 40e0  f7ee40d8 ffffffc0 00000000 00000000 00000000 00000000 00000000 00000000
0410_03-08-12.417[   48.077467] 4100  00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
0410_03-08-12.417[   48.078231] 4120  00000000 00000000 00000000 00000000 f7eebb40 ffffffc0 00000001 00000000
0410_03-08-12.417[   48.078996] 4140  f7ee4140 ffffffc0 00000000 00000000 f7ee4140 ffffffc0 2a9443c0 0000000b
0410_03-08-12.417[   48.079759] 4160  2a9443c0 0000000b 08128bc8 ffffff80 f7ee3d00 ffffffc0 00000001 00000000
0410_03-08-12.417[   48.080524] 4180  081285a0 ffffff80 70617773 2f726570 00000031 00000000 00000000 00000000
[... 363 lines cut ...]
0410_03-08-13.756[   49.417342] 000000c0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
0410_03-08-13.756[   49.418096] 000000e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
0410_03-08-13.756[   49.418849] 00000100: 00000efc 0000fb80 00000000 00000000 00000024 00000000 00000000 00000000
0410_03-08-13.756[   49.419600] 00000120: 00000000 00000000 00000000 00000000 00000003
0410_03-08-13.756[   49.420143] PMU:
0410_03-08-13.757[   49.420315] 00000000: 00000000 00000000 00000000 00000000 00000000 00198100 00198100 00000000
0410_03-08-13.757[   49.421069] 00000020: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
0410_03-08-13.757[   49.421822] 00000040: 00000000 00000000 00000000 00000000 00000000 881f833f 00000000 00000000
0410_03-08-13.757[   49.422576] 00000060: 000001c9 000001c9 000001c9 00000010 00000000 00007e00 00000000 28800000
0410_03-08-13.757[   49.423329] 00000080: 000003c0 00011940 00005dc0 000003c0 00005dc0 00000060 0000000f 00000018
0410_03-08-13.757[   49.424083] 000000a0: 00000018 00000018 00000018 00000006 00000006 00005dc0 00005dc0 00005dc0
0410_03-08-13.757[   49.424836] 000000c0: 00000000 00000000 00000000 00000000 00000000 00000000 3fffffff 00000000
0410_03-08-13.757[   49.425589] 000000e0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
0410_03-08-13.757[   49.426336] Kernel Offset: disabled
0410_03-08-13.757[   49.426649] Memory Limit: none
0410_03-08-13.757[   49.426952] rockchip-thermal ff260000.tsadc: channal 0: temperature(50 C)
0410_03-08-13.757[   49.427550] THERMAL REGS:
0410_03-08-13.757[   49.427789] 00000000: 00000200 00030133 00000031 00000002 00000000 00000000 00000000 00000000
0410_03-08-13.757[   49.428542] 00000020: 00000232 00000225 00000000 00000000 0000024e 00000000 00000000 00000000
0410_03-08-13.758[   49.429294] 00000040: 000002a5 000002a5 00000000 00000000 00000000 00000000 00000000 00000000
0410_03-08-13.771[   49.430047] 00000060: 00000004 00000004 00000753 00000753 00000000 00000000 00000000 00000000
0410_03-08-13.771[   49.430795] 00000080: 00000000 00000000
0410_03-08-13.798[   49.446861] ---[ end Kernel panic - not syncing: Fatal exception in interrupt
0410_03-08-33.673DDR Version 1.24 20191016
0410_03-08-33.673In

Allenhe123, 2022-05-25:
From the log, i can see below info:
0410_03-08-12.517[ 48.202025] [] el1_irq+0xb4/0x140
0410_03-08-12.517[ 48.202456] [] __dma_flush_range+0x1c/0x30
0410_03-08-12.517[ 48.202958] [] rga2_MapUserMemory+0x23c/0x2b0
0410_03-08-12.517[ 48.203479] [] rga2_set_mmu_info+0x998/0xaa8
0410_03-08-12.518[ 48.203999] [] rga2_blit+0x14c/0x360
0410_03-08-12.529[ 48.204451] [] rga2_blit_sync+0x7c/0x698
0410_03-08-12.529[ 48.204938] [] rga_ioctl+0x4e4/0x840
0410_03-08-12.550[ 48.205393] [] do_vfs_ioctl+0x33c/0x5c8
0410_03-08-12.550[ 48.205868] [] SyS_ioctl+0x94/0xa8
0410_03-08-12.550[ 48.206309] [] el0_svc_naked+0x24/0x28
0410_03-08-12.550[ 48.206784] Code: f9400804 b2400043 f9000444 f9000802 (f9000083)
0410_03-08-12.550[ 48.207329] ---[ end trace 27067e106eb5be06 ]---

Is this issue caused by RGA module?
Please help to check, Thanks.

Allenhe123, 2022-05-25:
Another log:

0525_13-46-27.928   18 root      20   0       0      0      0 S   0.0  0.0   0:00.31 ksoftirqd/2 
0525_13-46-28.564[55498.135156] 7f40: 0000007fab33f0b8 0000007fab1e0b10 000000000000081d 0000005571b856d8
0525_13-46-29.044pi@SOM-RK3399v2:~$ 
0525_13-46-29.541pi@SOM-RK3399v2:~$ : 0000000000000001 0000005571b84000 0000000000000000 0000005571b851a8
0525_13-46-29.572pi@SOM-RK3399v2:~$ : 0000000000000001 0000007fa5d042e0 0000007fab32cea8 0000007fa5d042e0
0525_13-46-29.608pi@SOM-RK3399v2:~$ : 0000007fab1e0b1c 0000000060000000 000000000000003a 000000000000001d
0525_13-46-29.639pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 0000000000000000 0000000000000000
0525_13-46-29.669pi@SOM-RK3399v2:~$  trace:
0525_13-46-29.702pi@SOM-RK3399v2:~$ ption stack(0xffffffc0e28c7430 to 0xffffffc0e28c7560)
0525_13-46-29.735pi@SOM-RK3399v2:~$ :                                   ffffffbdc0000000 0000008000000000
0525_13-46-29.767pi@SOM-RK3399v2:~$ : ffffffc0e28c7600 ffffff8008096418 ffffffc0f7efd240 0000000000000001
0525_13-46-29.797pi@SOM-RK3399v2:~$ : ffffff80090c6000 0000000000000000 ffffffc0e838b000 0000000000000002
0525_13-46-29.847pi@SOM-RK3399v2:~$ : ffffffc0e28c7490 ffffff80080cd4d8 ffffffc0e28c74a0 ffffff8008423ed8
0525_13-46-29.877pi@SOM-RK3399v2:~$ : ffffffc0e28c74d0 ffffff80080e6058 ffffffc0e838b220 0000000000000000
0525_13-46-29.908pi@SOM-RK3399v2:~$ : ffffffc0e28c74e0 ffffff80081c6f58 ffffffc000000000 ffffffc000001000
0525_13-46-29.941pi@SOM-RK3399v2:~$ : 0000000000000040 000000000000003f ffffffc0e2972d00 ffffffc0e4080000
0525_13-46-29.973pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 ffffffc0e8ffb200 0000000000000000
0525_13-46-30.005pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 0000000000000000 0000000000000000
0525_13-46-30.037pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 0000000000000000 0000000000000000
0525_13-46-30.068pi@SOM-RK3399v2:~$ ffff8008096418>] __dma_flush_range+0x18/0x30
0525_13-46-30.101pi@SOM-RK3399v2:~$ ffff80084c0ba4>] rga2_MapUserMemory+0x23c/0x2b0
0525_13-46-30.133pi@SOM-RK3399v2:~$ ffff80084c1828>] rga2_set_mmu_info+0x998/0xaa8
0525_13-46-30.165pi@SOM-RK3399v2:~$ ffff80084be3cc>] rga2_blit+0x14c/0x360
0525_13-46-30.196pi@SOM-RK3399v2:~$ ffff80084be6d4>] rga2_blit_sync+0x7c/0x698
0525_13-46-30.229pi@SOM-RK3399v2:~$ ffff80084bf9e4>] rga_ioctl+0x4e4/0x840
0525_13-46-30.264pi@SOM-RK3399v2:~$ ffff80082128b4>] do_vfs_ioctl+0x33c/0x5c8
0525_13-46-30.293pi@SOM-RK3399v2:~$ ffff8008212bd4>] SyS_ioctl+0x94/0xa8
0525_13-46-30.327pi@SOM-RK3399v2:~$ ffff8008083730>] el0_svc_naked+0x24/0x28
0525_13-46-30.376pi@SOM-RK3399v2:~$ : d2800082 9ac32042 d1000443 8a230000 (d50b7e20) 
0525_13-46-30.405pi@SOM-RK3399v2:~$  end trace f09750031cf9e507 ]---

Allenhe123, 2022-05-25:
Another Log:

0525_13-46-29.541pi@SOM-RK3399v2:~$ : 0000000000000001 0000005571b84000 0000000000000000 0000005571b851a8
0525_13-46-29.572pi@SOM-RK3399v2:~$ : 0000000000000001 0000007fa5d042e0 0000007fab32cea8 0000007fa5d042e0
0525_13-46-29.608pi@SOM-RK3399v2:~$ : 0000007fab1e0b1c 0000000060000000 000000000000003a 000000000000001d
0525_13-46-29.639pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 0000000000000000 0000000000000000
0525_13-46-29.669pi@SOM-RK3399v2:~$  trace:
0525_13-46-29.702pi@SOM-RK3399v2:~$ ption stack(0xffffffc0e28c7430 to 0xffffffc0e28c7560)
0525_13-46-29.735pi@SOM-RK3399v2:~$ :                                   ffffffbdc0000000 0000008000000000
0525_13-46-29.767pi@SOM-RK3399v2:~$ : ffffffc0e28c7600 ffffff8008096418 ffffffc0f7efd240 0000000000000001
0525_13-46-29.797pi@SOM-RK3399v2:~$ : ffffff80090c6000 0000000000000000 ffffffc0e838b000 0000000000000002
0525_13-46-29.847pi@SOM-RK3399v2:~$ : ffffffc0e28c7490 ffffff80080cd4d8 ffffffc0e28c74a0 ffffff8008423ed8
0525_13-46-29.877pi@SOM-RK3399v2:~$ : ffffffc0e28c74d0 ffffff80080e6058 ffffffc0e838b220 0000000000000000
0525_13-46-29.908pi@SOM-RK3399v2:~$ : ffffffc0e28c74e0 ffffff80081c6f58 ffffffc000000000 ffffffc000001000
0525_13-46-29.941pi@SOM-RK3399v2:~$ : 0000000000000040 000000000000003f ffffffc0e2972d00 ffffffc0e4080000
0525_13-46-29.973pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 ffffffc0e8ffb200 0000000000000000
0525_13-46-30.005pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 0000000000000000 0000000000000000
0525_13-46-30.037pi@SOM-RK3399v2:~$ : 0000000000000000 0000000000000000 0000000000000000 0000000000000000
0525_13-46-30.068pi@SOM-RK3399v2:~$ ffff8008096418>] __dma_flush_range+0x18/0x30
0525_13-46-30.101pi@SOM-RK3399v2:~$ ffff80084c0ba4>] rga2_MapUserMemory+0x23c/0x2b0
0525_13-46-30.133pi@SOM-RK3399v2:~$ ffff80084c1828>] rga2_set_mmu_info+0x998/0xaa8
0525_13-46-30.165pi@SOM-RK3399v2:~$ ffff80084be3cc>] rga2_blit+0x14c/0x360
0525_13-46-30.196pi@SOM-RK3399v2:~$ ffff80084be6d4>] rga2_blit_sync+0x7c/0x698
0525_13-46-30.229pi@SOM-RK3399v2:~$ ffff80084bf9e4>] rga_ioctl+0x4e4/0x840
0525_13-46-30.264pi@SOM-RK3399v2:~$ ffff80082128b4>] do_vfs_ioctl+0x33c/0x5c8
0525_13-46-30.293pi@SOM-RK3399v2:~$ ffff8008212bd4>] SyS_ioctl+0x94/0xa8
0525_13-46-30.327pi@SOM-RK3399v2:~$ ffff8008083730>] el0_svc_naked+0x24/0x28
0525_13-46-30.376pi@SOM-RK3399v2:~$ : d2800082 9ac32042 d1000443 8a230000 (d50b7e20) 
0525_13-46-30.405pi@SOM-RK3399v2:~$  end trace f09750031cf9e507 ]---

HermanChen, 2022-05-25:
The rga in mpp is not the formal lib. Try the librga https://github.com/AngerLee/librga

closed 2022-05-25 by HermanChen

Allenhe123, 2022-05-25:
got it, thank you @HermanChen