rockchip-linux/mpp #307: 硬解码关联surface与不关联surface的性能差异

chfdeng, 2022-07-03:
请假下，我用android实际测试rk3568时，在硬解码关联surface时勉强可以9路，不关联suface时只能勉强6路，这是为什么呢，是因为关联surface时硬解码后直接调用gpu渲染，而不关联surface时需要复制到内存中导致额外的消耗吗？望赐教！！

HermanChen, 2022-07-04:
有显示的话会消耗 DDR 带宽，影响解码器性能

chfdeng, 2022-07-04:
可是我实际测试结果是有显示时能流畅解码的路数比没显示时的路数还多，这是为什么呢@[HermanChen](https://github.com/HermanChen)

HermanChen, 2022-07-04:
是不是有关联surface的时候走的零拷贝路径，没关联surface时有cpu数据拷贝？top 看下负载

chfdeng, 2022-07-04:
这是不关联surface的：
[image]
这是关联surface的：
[image]
可以看出在不关联surface的情况下cpu和内存占用都比较大

HermanChen, 2022-07-04:
总的cpu用量区别不大
echo 0x100 > /sys/module/rk_vcodec/parameters/mpp_dev_debug
看看硬件运行时间，这个路径是手打的，不一定对，需要在板上验证下目录名

chfdeng, 2022-07-04:
这是不关联surface的：
[image]
这是关联surface的：
[image]

chfdeng, 2022-07-04:
是不是不关联的话需要多一次数据copy导致的额外消耗呢@[HermanChen](https://github.com/HermanChen)

HermanChen, 2022-07-04:
嗯，应该是，看硬件时间是差不多的

chfdeng, 2022-07-04:
好的，非常感谢

closed 2022-07-04 by chfdeng

xuhaijia, 2023-04-26:
@chfdeng 你好，请问您是如何在android上实现关联surface的，最近想开发这个功能，能否借鉴下您的demo