rockchip-linux/mpp #309: RK3399使用v4l2驱动读取相机的输出为JPEG格式的buf，使用mpp解码RGB格式时无法获取frame

aitazhixin, 2022-07-10:
初始化：
`    if (mpp_packet_init(&(capture->mPkt), buf, length) != 0)
    {
        LOGW(TAG) << "mpp packet init failed";
        return -1;
    }
    if (mpp_create(&(capture->mCtx), &(capture->mApi)) != MPP_OK) {
        LOGW(TAG) << "mpp create failed";
        return -1;
    }

    if (mpp_init(capture->mCtx, MPP_CTX_DEC, MppCodingType::MPP_VIDEO_CodingMJPEG) != MPP_OK) {
        LOGW(TAG) << "mpp init decode mjpeg failed";
        return -1;
    }`

输出：
`        mpp_packet_write(capture->mPkt, 0, src, length);
        mpp_packet_set_pos(capture->mPkt, src);
        mpp_packet_set_length(capture->mPkt, length);
        mpp_packet_set_eos(capture->mPkt);
        int ret = 0;
        MppFrame srcFrm  = NULL;
        ret = capture->mApi->decode_put_packet(capture->mCtx, capture->mPkt);
        if (ret != MPP_OK) {
            LOGW(TAG) << "decode failed " << ret;
            return false;
        }

        RK_S32 times = 5;
        {
            try_again:
            ret = capture->mApi->decode_get_frame(capture->mCtx, &srcFrm);
            if (ret == MPP_ERR_TIMEOUT) {
                if (times > 0) {
                    times--;
                    usleep(2000);
                    goto try_again;
                }
            }

            if (ret != MPP_OK) {
                LOGW(TAG) << "get frame failed";
                return false;
            }

            if (srcFrm) {
                if (mpp_frame_get_info_change(srcFrm)) {
                    RK_U32 width = mpp_frame_get_width(srcFrm);
                    RK_U32 height = mpp_frame_get_height(srcFrm);
                    LOGD(TAG) << "get frame width " << width << " height " << height;
                }
                mpp_frame_deinit(&srcFrm);
            }
            else {
                LOGE(TAG) << "src frame is null, ret is " << ret;
                usleep(2000);
                goto try_again;
            }
        }`

aitazhixin, 2022-07-11:
@HermanChen 可以帮忙解答一下问题吗？我这里的使用方法是否有问题？

HermanChen, 2022-07-11:
mpp_packet_set_eos 这个不需要
jpeg 解码目前还不好用 put_packet / get_frame 接口来处理……
需要用到 poll / enqueue / dequeue 接口，有点麻烦，需要参考旧版本的 mpi_dec_test 代码

aitazhixin, 2022-07-11:
> mpp_packet_set_eos 这个不需要 jpeg 解码目前还不好用 put_packet / get_frame 接口来处理…… 需要用到 poll / enqueue / dequeue 接口，有点麻烦，需要参考旧版本的 mpi_dec_test 代码

如果不设置mpp_packet_set_eos会导致put_packet()中的判断if (mPackets->list_size() < 4 || eos) 为false；

旧版本的mpi_dec_test怎么可以找到？有对应分支或tag吗？

aitazhixin, 2022-07-11:
> mpp_packet_set_eos 这个不需要 jpeg 解码目前还不好用 put_packet / get_frame 接口来处理…… 需要用到 poll / enqueue / dequeue 接口，有点麻烦，需要参考旧版本的 mpi_dec_test 代码

旧版本是static int decode_advanced(MpiDecCtx *data)这个函数吗？

HermanChen, 2022-07-11:
对，用那个advanced的流程

aitazhixin, 2022-07-14:
> 对，用那个advanced的流程

> 对，用那个advanced的流程

在使用这个的时候遇到了崩溃
`
#define list_entry(ptr, type, member) \
    ((type *)((char *)(ptr)-(unsigned long)(&((type *)0)->member)))
`
这里((type *)0)->member空指针取member会崩溃

HermanChen, 2022-07-14:
具体给下出错 log

aitazhixin, 2022-07-14:
> 具体给下出错 log

[image]
出错是在map_buffer_get_unused函数里面

aitazhixin, 2022-07-14:
> 具体给下出错 log

这个崩溃已经找到问题了。现在遇到的一个问题是mpp_init后，再调用control函数设置输出图像格式为RGB888，会导致mpp_dev_deinit调用，然后重新调用ret = mpp_dev_init(&JpegHalCtx->dev, VPU_CLIENT_VDPU2_PP);可是RK3399好像是不支持此类型的：
static const MppVpuType mpp_vpu_version[] = {
    { "rk3036",  ROCKCHIP_SOC_RK3036,   HAVE_VDPU1 | HAVE_VEPU1 | HAVE_HEVC_DEC, },
    { "rk3066",  ROCKCHIP_SOC_RK3066,   HAVE_VDPU1 | HAVE_VEPU1,                 },
    { "rk3188",  ROCKCHIP_SOC_RK3188,   HAVE_VDPU1 | HAVE_VEPU1,                 },
    { "rk3288",  ROCKCHIP_SOC_RK3288,   HAVE_VDPU1 | HAVE_VEPU1 | HAVE_HEVC_DEC, },
    { "rk3126",  ROCKCHIP_SOC_RK312X,   HAVE_VDPU1 | HAVE_VEPU1 | HAVE_HEVC_DEC, },
    { "rk3128h", ROCKCHIP_SOC_RK3128H,  HAVE_VDPU2 | HAVE_VEPU2 | HAVE_RKVDEC,   },
    { "rk3128",  ROCKCHIP_SOC_RK312X,   HAVE_VDPU1 | HAVE_VEPU1 | HAVE_HEVC_DEC, },
    { "rk3368",  ROCKCHIP_SOC_RK3368,   HAVE_VDPU1 | HAVE_VEPU1 | HAVE_HEVC_DEC, },
    { "rk3399",  ROCKCHIP_SOC_RK3399,   HAVE_VDPU2 | HAVE_VEPU2 | HAVE_RKVDEC,   },
……

这种情况，是不是就不能设置输出图片的格式？还是可以把上面的mpp_dev_init输入的type改为HAVE_VEPU2？

Shining-Tiga, 2024-04-02:
@HermanChen 请问一下，现在的develop版本的支持put_packet / get_frame 接口，是否支持解码v4l2驱动读取相机的输出为JPEG格式的buf流吗？

patstar123, 2024-11-30:
> 这个崩溃已经找到问题了。现在遇到的一个问题是mpp_init后，再调用control函数设置输出图像格式为RGB888，会导致mpp_dev_deinit调用，然后重新调用ret = mpp_dev_init(&JpegHalCtx->dev, VPU_CLIENT_VDPU2_PP);可是RK3399好像是不支持此类型的： static const MppVpuType [...]

请问这个【#define list_entry(ptr, type, member) \ ((type *)((char *)(ptr)-(unsigned long)(&((type *)0)->member)))】这个崩溃是什么原因，看着是必然崩的呢，我也程序也在这崩了

HermanChen, 2024-12-02:
> @HermanChen 请问一下，现在的develop版本的支持put_packet / get_frame 接口，是否支持解码v4l2驱动读取相机的输出为JPEG格式的buf流吗？

刚刚在支持，内部补丁还在测试