rockchip-linux/mpp #310: mpp解码 h264 MVC

tianqu308, 2022-07-27:
请问 libmpp 能够解码 h264 MVC吗？有没有例子可以参考的？

HermanChen, 2022-07-27:
MVC 解码和 AVC 过程是一样的，只是解出来的帧在显示的时候要看情况处理
解码部分是一样的

closed 2022-07-27 by HermanChen

tianqu308, 2022-07-27:
我在libmpp  h264d_api.c 中发现有如下设置
 mpp_env_get_u32("rkv_h264d_mvc_disable", &p_Inp->mvc_disable, 1); 
默认h264 mvc 解码是关闭的吗？
是不是需要设置 setprop rkv_h264d_mvc_disable 0 将mvc解码使能

HermanChen, 2022-07-27:
这个是解析码流路径的开关，关了 mvc 路径之后效率会高一些，是要把 disable 环境变量配置为 0

tianqu308, 2022-07-27:
> 这个是解析码流路径的开关，关了 mvc 路径之后效率会高一些，是要把 disable 环境变量配置为 0

也就是说，我不需要设置什么，mvc 解码默认是支持的吧？

另外，我用的 ffmpeg demux,

tianqu308, 2022-07-27:
> 这个是解析码流路径的开关，关了 mvc 路径之后效率会高一些，是要把 disable 环境变量配置为 0

libmpp默认是支持 mvc解码的，不需要额外设置什么吧

另外：
我使用 ffmpeg demux ，将mvc数据直接送到mpp decode，解码能够成功，如何获取mvc的左右两个frame ，是连续执行两次decode_get_frame 吗？
伪代码如下：
`
AVPacket pkt;
av_read_frame(m_pFormatContext, &pkt);
...

MppPacket mppPacket = nullptr;
mpp_packet_init(&mppPacket, pkt.data, pkt.size);

ret = mpi->decode_put_packet(ctx, mppPacket);
...

MppFrame m_pFrame = nullptr;
ret = mpi->decode_get_frame(ctx, &m_pFrame);
`

HermanChen, 2022-07-28:
对，MVC 基本上两个视角的数据交替来的。

tianqu308, 2022-07-28:
libmpp有api 来区分两个frame吗？

tianqu308, 2022-07-28:
libmpp有api 来区分两个视角的frame吗？

HermanChen, 2022-07-28:
mpp_frame_get_viewid 可以得到

tianqu308, 2022-07-28:
非常感谢您的回复。
我用如下代码获取到的两个视角的frame 的viewid 均为0，请问是调用逻辑上哪里有问题吗？
`
AVPacket pkt;
av_read_frame(m_pFormatContext, &pkt);

...

MppPacket mppPacket = nullptr;
mpp_packet_init(&mppPacket, pkt.data, pkt.size);

ret = mpi->decode_put_packet(ctx, mppPacket);

...

MppFrame m_pFrame = nullptr;
MppFrame m_pSubFrame = nullptr;

// 此处连续调用两次decode_get_frame 分别获取两个视角frame
// 获得的 viewid 和viewid_sub 均为0
 
ret = mpi->decode_get_frame(ctx, &m_pFrame);
RK_U32 viewid = mpp_frame_get_viewid(m_pFrame);

ret = mpi->decode_get_frame(ctx, &m_pSubFrame);
RK_U32 viewid_sub = mpp_frame_get_viewid(m_pSubFrame);

`
这是我们的测试源
链接：https://pan.baidu.com/s/1yUq_MsqvKyXKEa2SRIot0A 
提取码：p2ht

HermanChen, 2022-07-29:
可以在调用 mpp_frame_set_viewid 的 write_picture 里加一下打印看看

tianqu308, 2022-07-29:
我使用的mpp develop分支最新代码。
打开了 H264D_DBG_DPB_INFO ，下面是mpp.log 
[mpp.log](https://github.com/rockchip-linux/mpp/files/9215564/mpp.log)

下面这句是加的打印，
07-29 02:19:53.316  2022  2086 I h264d_dpb: [write_picture] p->layer_id=0

另外 看到log中 有 mvc_flag=0 的字样，是否没识别出 是mvc源？

HermanChen, 2022-07-29:
跟踪到 set_slice_user_parmeters 看看，active_mvc_sps_flag 没被置起来，估计是没解析到 mvcExt？

tianqu308, 2022-07-29:
[10.10.8.121_2022-07-29_12-15-33.log](https://github.com/rockchip-linux/mpp/files/9216019/10.10.8.121_2022-07-29_12-15-33.log)
分别在init_slice_parmeters 和 set_slice_user_parmeters 增加了打印