rockchip-linux/mpp #311: can not found match soc name: rk3566

Wazirri, 2022-07-29:
mpi_dec_test -i 2022-06-18_15-15-54.mkv -t 7 -f 7 -w 10 -h 10 
mpi_dec_test: cmd parse result:
mpi_dec_test: input  file name: 2022-06-18_15-15-54.mkv
mpi_dec_test: output file name: 
mpi_dec_test: config file name: 
mpi_dec_test: width      :   10
mpi_dec_test: height     :   10
mpi_dec_test: type       : 7
mpi_dec_test: debug flag : 0
mpi_dec_test: max frames : 0
mpi_dec_test: mpi_dec_test start
mpi_dec_test: input file size 102681644
mpi_dec_test: mpi_dec_test decoder test start w 10 h 10 type 7
mpi: mpp version: 5ce1cb85 author: Caesar Wang debian: add rules for mpp with 20190626
mpp_log: can not found match soc name: radxa,radxa-cm3-io rockchip,rk3566
hal_h264d_api: Assertion vcodec_type & ((0x00000200) | (0x00000001) | (0x00000002)) failed at hal_h264d_init:104
hal_h264d_api: hal_h264d_init hard mode error, value=0
hal_h264d_api: Assertion 0 failed at hal_h264d_init:154
mpp_device: mpp_device_init failed to find device for coding 7 type 0
mpp_rt: NOT found ion allocator
mpp_rt: found drm allocator
Segmentation fault

Wazirri, 2022-07-29:
@HermanChen  @Caesar-github  please help about this. We could not make VPU mpp work with RK3566, and we have important project. If you can guide us how to do that, we can work and try to help your project. Thanks.

Wazirri, 2022-07-30:
Now, I build @HermanChen repository. It gives Seg fault directly.

root@radxa-cm3-io:/home/rock/mpp/build/linux/aarch64/test# ./mpi_dec_multi_test /home/rock/h264.h264 -w 10 -h 10 -t 7
Segmentation fault

HermanChen, 2022-08-01:
radxa,radxa-cm3-io rockchip,rk3566 这个找到不匹配的 soc 名字，可以更新 下 mpp 看看

Wazirri, 2022-08-04:
When I compiled your repo it worked, thanks.