rockchip-linux/mpp #315: mpi_dec_test 解码h265文件无法输出yuv

imxys, 2022-08-04:
```shell
[root@root:/app_data]# /app_data/mpi_dec_test -i /app_data/4k_h265_60_5s.h265 -o /app_data/maou.yuv -t 16777220 -w 3840 -h 2160
mpp[14793]: mpi_dec_utils: input file /app_data/4k_h265_60_5s.h265 size 778076
mpp[14793]: mpi_dec_utils: cmd parse result:
mpp[14793]: mpi_dec_utils: input  file name: /app_data/4k_h265_60_5s.h265
mpp[14793]: mpi_dec_utils: output file name: /app_data/maou.yuv
mpp[14793]: mpi_dec_utils: width      : 3840
mpp[14793]: mpi_dec_utils: height     : 2160
mpp[14793]: mpi_dec_utils: type       : 16777220
mpp[14793]: mpi_dec_utils: max frames :    0
mpp[14793]: mpi_dec_test: mpi_dec_test start
mpp[14793]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[14793]: mpi_dec_test: 0x3e46d200 mpi_dec_test decoder test start w 3840 h 2160 type 16777220
mpp[14793]: mpp_rt: NOT found ion allocator
mpp[14793]: mpp_rt: found drm allocator
mpp[14793]: mpi_dec_test: 0x3e46d200 decode_get_frame get info changed found
mpp[14793]: mpi_dec_test: 0x3e46d200 decoder require buffer w:h [3840:2160] stride [4864:2160] buf_size 18911232
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 0
mpp[14793]: utils: not supported format 1
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 1
mpp[14793]: utils: not supported format 1
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 2
mpp[14793]: utils: not supported format 1
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 3
mpp[14793]: utils: not supported format 1
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 4
mpp[14793]: utils: not supported format 1
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 5
mpp[14793]: utils: not supported format 1
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 6
mpp[14793]: utils: not supported format 1
mpp[14793]: mpi_dec_test: 0x3e46d200 decode get frame 7
```

电脑播放该文件情况为
```shell
ffplay 4k_h265_60_5s.h265 
ffplay version 4.2.7-0ubuntu0.1 Copyright (c) 2003-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Input #0, hevc, from '4k_h265_60_5s.h265':    0KB sq=    0B f=0/0   
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: hevc (Main 10), yuv420p10le(tv), 3840x2160 [SAR 1:1 DAR 16:9], 60 fps, 60 tbr, 1200k tbn, 60 tbc
    nan M-V:    nan fd=   2 aq=    0KB vq=    0KB sq=    0B f=0/0 
```

请问无法正常输出文件可以确认解码成功吗？

imxys, 2022-08-04:
更新，找到一个h265文件是Main Profile yuv420p，解码正常
```shell
19:41:40 ~/Videos > ffplay 1080_h265_60_Main_yuv420p_5s.h265 
ffplay version 4.2.7-0ubuntu0.1 Copyright (c) 2003-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Input #0, hevc, from '1080_h265_60_Main_yuv420p_5s.h265':0B f=0/0   
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: hevc (Main), yuv420p(tv), 1920x1080 [SAR 1:1 DAR 16:9], 60 fps, 60 tbr, 1200k tbn, 60 tbc
    nan M-V:    nan fd=   2 aq=    0KB vq=   23KB sq=    0B f=0/0   
```

```shell
[root@root:/app_data]# /app_data/mpi_dec_test -i /app_data/1080_h265_60_Main_yuv420p_5s.h265 -o /app_data/maou.yuv -t 16777220 -w 1920 -h 1080
mpp[1414]: mpi_dec_utils: input file /app_data/1080_h265_60_Main_yuv420p_5s.h265 size 180632
mpp[1414]: mpi_dec_utils: cmd parse result:
mpp[1414]: mpi_dec_utils: input  file name: /app_data/1080_h265_60_Main_yuv420p_5s.h265
mpp[1414]: mpi_dec_utils: output file name: /app_data/maou.yuv
mpp[1414]: mpi_dec_utils: width      : 1920
mpp[1414]: mpi_dec_utils: height     : 1080
mpp[1414]: mpi_dec_utils: type       : 16777220
mpp[1414]: mpi_dec_utils: max frames :    0
mpp[1414]: mpi_dec_test: mpi_dec_test start
mpp[1414]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[1414]: mpi_dec_test: 0x1bb00d80 mpi_dec_test decoder test start w 1920 h 1080 type 16777220
mpp[1414]: mpp_rt: NOT found ion allocator
mpp[1414]: mpp_rt: found drm allocator
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode_get_frame get info changed found
mpp[1414]: mpi_dec_test: 0x1bb00d80 decoder require buffer w:h [1920:1080] stride [2304:1080] buf_size 4478976
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode get frame 0
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode get frame 1
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode get frame 2
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode get frame 3
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode get frame 4
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode get frame 5
mpp[1414]: mpi_dec_test: 0x1bb00d80 decode get frame 6
```

imxys, 2022-08-04:
另外请问h263解码是否正常？ 我输出解码yuv文件结果画面是绿色的

原视频信息如下
```shell
~/Videos > ffplay 704x576_h263.h263 
ffplay version 4.2.7-0ubuntu0.1 Copyright (c) 2003-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
Input #0, h263, from '704x576_h263.h263':=    0KB sq=    0B f=0/0   
  Duration: N/A, bitrate: N/A
    Stream #0:0: Video: h263, yuv420p, 704x576 [SAR 12:11 DAR 4:3], 29.97 fps, 29.97 tbr, 1200k tbn, 29.97 tbc
   1.34 M-V:  0.031 fd=   0 aq=    0KB vq=   81KB sq=    0B f=0/0 
```

```shell
[root@root:/app_data]# /app_data/mpi_dec_test -t 3 -i /app_data/704x576_h263.h263 -o /app_data/704x576_h263.yuv -w 704 -h 576
mpp[449]: mpi_dec_utils: input file /app_data/704x576_h263.h263 size 642153
mpp[449]: mpi_dec_utils: cmd parse result:
mpp[449]: mpi_dec_utils: input  file name: /app_data/704x576_h263.h263
mpp[449]: mpi_dec_utils: output file name: /app_data/704x576_h263.yuv
mpp[449]: mpi_dec_utils: width      :  704
mpp[449]: mpi_dec_utils: height     :  576
mpp[449]: mpi_dec_utils: type       :    3
mpp[449]: mpi_dec_utils: max frames :    0
mpp[449]: mpi_dec_test: mpi_dec_test start
mpp[449]: mpp_info: mpp version: unknown mpp version for missing VCS info
mpp[449]: mpi_dec_test: 0x2e850100 mpi_dec_test decoder test start w 704 h 576 type 3
mpp[449]: mpp_rt: NOT found ion allocator
mpp[449]: mpp_rt: found drm allocator
mpp[449]: mpp_log: the VDPU2_MODE is used currently!
mpp[449]: mpi_dec_test: 0x2e850100 decode_get_frame get info changed found
mpp[449]: mpi_dec_test: 0x2e850100 decoder require buffer w:h [704:576] stride [704:576] buf_size 729907
mpp[449]: mpp_log: mpp_h263_parser_decode can not found start code in len 4096 packet
mpp[449]: mpp_log: mpp_h263_parser_decode can not found start code in len 4096 packet
mpp[449]: mpp_log: mpp_h263_parser_decode can not found start code in len 4096 packet
mpp[449]: mpi_dec_test: 0x2e850100 decode get frame 0
mpp[449]: mpi_dec_test: 0x2e850100 decode get frame 1
mpp[449]: mpi_dec_test: 0x2e850100 decode get frame 2
mpp[449]: mpp_log: mpp_h263_parser_decode can not found start code in len 4096 packet
mpp[449]: mpp_log: mpp_h263_parser_decode can not found start code in len 4096 packet
mpp[449]: mpp_log: mpp_h263_parser_decode can not found start code in len 4096 packet
mpp[449]: mpi_dec_test: 0x2e850100 decode get frame 3
mpp[449]: mpi_dec_test: 0x2e850100 decode get frame 4
```
从日志上看不出问题在哪里

HermanChen, 2022-08-05:
mpp[449]: mpi_dec_test: 0x2e850100 decode get frame 0
解码是正常的

mpp[449]: mpp_log: mpp_h263_parser_decode can not found start code in len 4096 packet
这个只是 warning log，说明有个 4K 分段没找到起始码，可忽略

HermanChen, 2022-08-05:
YUV是绿屏的话，麻烦提供下 263 码流

imxys, 2022-08-05:
[704x576_h263p.h263.zip](https://github.com/rockchip-linux/mpp/files/9269077/704x576_h263p.h263.zip)

不好意思早上没看消息。另外
我发现我这文件解码出来才48.7M，在设备上用ffmpeg软解，指定 -c h263 解码有175M，画面正常

imxys, 2022-08-05:
h265那边，yuv420p10le的视频，也是解码正常，只是yuv无法保存文件吗？

HermanChen, 2022-08-06:
10bit 的yuv保存有点麻烦，不一定能存得下来
263 的问题我们看一下

imxys, 2022-08-21:
> 10bit 的yuv保存有点麻烦，不一定能存得下来 263 的问题我们看一下

请问有进展吗？
另外我想问，h265的stride是怎么算的？ 我看源码，似乎默认的水平stride是256，垂直的stride是8
这个在解码yuv420p的 h265的时候好像没有错
```sh
mpp[30284]: mpi_dec_test: 0x3e2fae0 decoder require buffer w:h [1920:1080] stride [2304:1080] buf_size 4478976
```
但是在解码 yuv420p10le 的时候完全对不上
```sh
mpp[29081]: mpi_dec_test: 0x29b46fe0 decoder require buffer w:h [1920:1080] stride [2816:1080] buf_size 5474304
```
想不出为啥

HermanChen, 2022-08-22:
265 10bit 的水平 stride 是 256 奇数倍对齐，看下 fetch_data 这个函数

imxys, 2022-08-22:
> 265 10bit 的水平 stride 是 256 奇数倍对齐，看下 fetch_data 这个函数

那一开头
```shell
mpp[14793]: mpi_dec_test: 0x3e46d200 decoder require buffer w:h [3840:2160] stride [4864:2160] buf_size 18911232
```
这个是为什么？ 256 * 15 = 3840,  256 * 19 = 4864，3840已经对齐了才对

Sherman86, 2022-08-23:
你好，263的分帧逻辑已确认是有问题的，感谢指正。近期将提交补丁修正。

Sherman86, 2022-08-23:
[librockchip_mpp.zip](https://github.com/rockchip-linux/mpp/files/9399404/librockchip_mpp.zip)
你好，附件是linux对应的库，push到/lib/路径中即可解决h263文件解码异常问题。

Sherman86, 2022-08-23:
[librockchip_mpp.zip](https://github.com/rockchip-linux/mpp/files/9399888/librockchip_mpp.zip)

你好，附件是
linux对应的库，push到/lib/路径中即可解决h263文件解码异常问题。

imxys, 2022-08-23:
> [librockchip_mpp.zip](https://github.com/rockchip-linux/mpp/files/9399888/librockchip_mpp.zip)
> 
> 你好，附件是 linux对应的库，push到/lib/路径中即可解决h263文件解码异常问题。

您好，您发的两份文件都是32位的so库，麻烦发一份rk3399上能运行的，感谢

Sherman86, 2022-08-23:
> 

请问您手里有mpp代码吗？我直接发给您补丁，您编译之后push到板子里可能更方便些。不同环境对应的交叉编译工具有所不同，我板子的环境应该和您的不太一样。

imxys, 2022-08-23:
> > 
> 
> 请问您手里有mpp代码吗？我直接发给您补丁，您编译之后push到板子里可能更方便些。不同环境对应的交叉编译工具有所不同，我板子的环境应该和您的不太一样。

有的

Sherman86, 2022-08-24:
[h263d.zip](https://github.com/rockchip-linux/mpp/files/9411293/h263d.zip)
git apply h263d.diff就可以打上了。

imxys, 2022-08-24:
> [h263d.zip](https://github.com/rockchip-linux/mpp/files/9411293/h263d.zip) git apply h263d.diff就可以打上了。

已验证h263硬解yuv正常，谢谢。

顺便想咨询一下，有没有什么yuv查看工具推荐？7yuv只有水平跨距，mpp硬解的buffer在垂直方向还有跨距，不设置这个，保存整个MppFrame而不是只保存有效数据时yuv没法正常显示。如果没有我就只保存有效数据了。

Sherman86, 2022-08-24:
我一般使用的是yuview工具，一款开源免费的yuv查看工具。

其中要注意的是，rk芯片解码出的yuv默认为yuv420sp格式的。format->custom->开启UV Interleaved。

imxys, 2022-08-24:
了解，我最近也有找到这个。谢谢

imxys, 2022-08-24:
我需要关闭该issue吗？还是你们提交修改后关闭

Sherman86, 2022-08-24:
> 我需要关闭该issue吗？还是你们提交修改后关闭

如果没有其他疑问的话，关闭本issue就可以了。

HermanChen, 2022-08-24:
This commit https://github.com/rockchip-linux/mpp/commit/c00f55111fff01fc18ff595f3bbaa029c114323c fixes the issue

closed 2022-08-24 by HermanChen