rockchip-linux/mpp #324: about mpp's zero-copy display path

docterling, 2022-09-19:
1、according to the readme,mpp support 3 types memory usage,because i use drm to display the yuv data,so i choose the third method:
CREATE_DUMB-->mpp_buffer_commit-->drmModeAddFB2
for 4K res，malloc 30 buffers，it takes about 2000 milisecond；for 1080P，24 buffers，600 milisecond，is it normal for that？and the decode_put_packet return lots of MPP_ERR_BUFFER_FULL in the very beginning.in my use case,the res will be 720p to 4K,so i must dynamicly to create and destroy drm dump buf.so i can't use this zero-copy mode?

HermanChen, 2022-09-19:
The drm buffer alloc and map will take a lot of time. That is quite normal...
The ion allocator and dma-heap allocator will take less time.
And you can make the external buffer creator's buffer creation parellely running with mpp working thread. Do not be blocked by all buffer creation, create one buffer and commit one.

docterling, 2022-09-20:
1.
thx for your advice
i've try create one buffer and commit one in the mpp putting packet and getting frame thread,1080p res would be ok,but 4k res can see that video is not so smooth,cause in my use case starting mpp dec and then ending mpp dec would be in milisecond,so if i use another thread to do the create and commit would be difficult to sync
i see the mpi_dec_test demo use the ION,so can i use it in linux sdk?
2.
and i had found that in rk3566 platform,the apple m1 airplay streaming mpp dec would frozen.can you try this bitstream to see what's happening.
[m1_streaming.zip](https://github.com/rockchip-linux/mpp/files/9603399/m1_streaming.zip)

Sherman86, 2022-09-27:
1. You can use MPP_BUFFER_TYPE_ION command in linux sdk. Actually, input ION doesn't mean that you can finally get ion buffer, which is decided by driver. The allocator priority of MPP_BUFFER_TYPE_ION command: dma_heap>ion>drm.
2. Mpp is quite strict with h.264 protocol. The stream loses parts of frames result in frame_num discontinuous. In this case, sps's gaps_in_frame_num_value_allowed_flag should be equal to 1 in h.264 protocol but 0 in m1_streaming. So mpp marks error flag and then stops.
There are some methods can be refered. 
1. amend gaps_in_frame_num_value_allowed_flag to 1 in m1 airplay. 
2. enable disable_error flag in mpp. //mpp_dec_cfg_set_u32(cfg, "base:disable_error", disable_error);
3. remove the discontinuous checking in mpp. //check_dpb_discontinuous() in h264d_init

docterling, 2022-09-27:
Big thanks for your reply and advices.
i've tried and it works fine.
thanks again.

closed 2022-09-27 by docterling