rockchip-linux/mpp #325: RV1108  20201029_1.4.4 SDK USB camera mjpeg preview decode Error

dasimaznG, 2022-09-22:
RV1108  20201029_1.4.4 SDK USB camera mjpeg preview decode Error

RV1108 的官方开发板 RK_MINIEVB_RV1108_V10 连接USB MJPEG 格式的720P摄像头，编译SDK烧录后，连接摄像头，官方video程序PREIVEW时出现如下错误，图面时而卡顿，时而正常，卡顿时出现如下log， 麻烦看一下要怎么解决

MPP 版本是 video: mpp_info: mpp version: 5635027 author: Hertz Wang     2020-06-09 skip unhandled jpeg section

video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.481697] video: vpu_decode_jpeg_doing err
[    9.482735] video: Debug: NV12_Encode inBuf datasize get 0
[    9.510988] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[    9.511429] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.512232] video: vpu_decode_jpeg_doing err
[    9.513854] video: Debug: NV12_Encode inBuf datasize get 0
[    9.546649] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[    9.547148] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.548617] video: vpu_decode_jpeg_doing err
[    9.549532] video: Debug: NV12_Encode inBuf datasize get 0
[    9.578197] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[    9.578321] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.580216] video: vpu_decode_jpeg_doing err
[    9.580293] video: Debug: NV12_Encode inBuf datasize get 0
[    9.609544] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[    9.609668] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.611556] video: vpu_decode_jpeg_doing err
[    9.611631] video: Debug: NV12_Encode inBuf datasize get 0
[    9.641244] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[    9.641371] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.643123] video: vpu_decode_jpeg_doing err
[    9.643199] video: Debug: NV12_Encode inBuf datasize get 0
[    9.676522] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[    9.676671] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.678427] video: vpu_decode_jpeg_doing err
[    9.678503] video: Debug: NV12_Encode inBuf datasize get 0
[    9.708039] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[    9.708154] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[    9.710112] video: vpu_decode_jpeg_doing err

HermanChen, 2022-09-22:
可能 USB 传输的数据有问题，码率高了会丢？保存一下输入数据，用ffmpeg解码看看有没有问题。

dasimaznG, 2022-09-22:
有可能 ，我换了一款USB 摄像头，重新测试：
   [   46.347266] video: CAMHALInfo:   alloc: format mjpeg 1280x960, usage 0x0000000a
  [   46.348155] video: CAMHALInfo:   ProxyCameraBuffer: vaddr (nil), camPixFmt mjpeg, stride 1280
  [   46.349169] video: CAMHALInfo:   alloc: format mjpeg 1280x960, usage 0x0000000a
  [   46.350022] video: CAMHALInfo:   ProxyCameraBuffer: vaddr (nil), camPixFmt mjpeg, stride 1280
  [   46.351194] video: CAMHALInfo:   alloc: format mjpeg 1280x960, usage 0x0000000a
  [   46.352059] video: CAMHALInfo:   ProxyCameraBuffer: vaddr (nil), camPixFmt mjpeg, stride 1280
  [   46.353048] video: CAMHALInfo:   alloc: format mjpeg 1280x960, usage 0x0000000a
  [   46.353255] video: CAMHALInfo:   ProxyCameraBuffer: vaddr (nil), camPixFmt mjpeg, stride 1280
  [   46.367151] video: CAMHALInfo:   addBufferNotifier:24 
  [   46.367833] video: CAMHALInfo:   addBufferNotifier:30 
  [   46.368659] video: CAMHALInfo:   addBufferNotifier:24 
  [   46.369321] video: CAMHALInfo:   addBufferNotifier:30 
并未出现JPEG decode 错误，，USB录像的视频在板下正常播放，但是在查看照片时，解码USB 摄像头拍照下的MJPEG照片时，多款USB拍照下的照片都无法被解码，放在PC上，照片都能正常打开
报如下错误：
 [  279.738159] video: CameraWinProc MSG_KEYDOWN SetMode = 3, key = 108
[  279.738268] video: videopreview_pre
[  279.738333] video: videopreview_show_decode file = /mnt/sdcard/PHOTO-REAR/20160121_085513_B.jpg, previewname = 20160121_085513_B.jpg(10/11)
[  279.738395] video: videoplay_decode_jpeg(/mnt/sdcard/PHOTO-REAR/20160121_085513_B.jpg)
[  279.791249] video: mpp_info: mpp version: 5635027 author: Hertz Wang     2020-06-09 skip unhandled jpeg section
[  279.796329] video: mpp_buf_slot: set frame info: w 1280 h  960 hor 1280 ver  960
[  279.797656] video: mpp_dec: setting default w 1280 h  960 h_str 1280 v_str  960
[  279.804285] video: --- ion alloc, get fd: 26
[  279.812744] video: jpegd_parser: jpegd_decode_frame EOI marker not found!
[  279.813672] video: mpp_dec: mpp_dec_advanced_thread something wrong with mpp_parser_parse!
[  279.815235] video: vpu_decode_jpeg_doing err
请帮忙看看这个问题