rockchip-linux/mpp #326: mpp memory usage problem

docterling, 2022-09-22:
1.
i use the Mode 2: Half internal mode
first i send 4k h265 stream to mpp,and then 1080p h264,next i will switch the bitstream on the fly,4k-1080-4k-1080......(time interval from a few milisecond to seconds)
and i see mpp will be detect the change from the api "mpp_frame_get_info_change",and i call the api "mpp_buffer_group_put" to free previous MppBufferGroup.but in htop i see the memory usage keep continues growing.finally my program will be killed by oom.
so how to correct freeing previous memory used by mpp in this situation.

HermanChen, 2022-09-23:
It is better to use mpp_buffer_group_clear to reset the buffer group on info change. And each frame output from mpp should be released.

docterling, 2022-09-23:
thx for you reply.
yeah,every decode_get_frame if return checking is ok will follow mpp_frame_deinit to free the mppframe.
i've changed the reset api to mpp_buffer_group_clear,the memory usage much better than before,now it's only grew 1~2MB each time,maybe the other causes.
thanks again

closed 2022-09-23 by HermanChen