rockchip-linux/mpp #327: 我在编码H264时，只有第一个IDR帧

GrizaJ, 2022-10-09:
开发环境：Firefly RK3399 v15
我在使用Mpp硬件编码时只有最开始的一帧有SPS和PPS 整段视频的结构是这样的：
00 00 00 01 67……00 00 00 01 68……00 00 00 01 06 05……00 00 00 01 06 05……00 00 00 01 25……00 00 00 01 21xN
后面的H264都是00 00 00 01 21 的帧。这导致我的视频流不能从中间开始播放。

代码中编码设置是这样的：

    mpp_enc_cfg_set_s32(p_cfg, "prep:width", in_stream_info->width);
    mpp_enc_cfg_set_s32(p_cfg, "prep:height", in_stream_info->height);
    mpp_enc_cfg_set_s32(p_cfg, "prep:hor_stride", data_hor_stride);
    mpp_enc_cfg_set_s32(p_cfg, "prep:ver_stride", data_ver_stride);
    mpp_enc_cfg_set_s32(p_cfg, "prep:format", MPP_FMT_YUV420P);

    mpp_enc_cfg_set_s32(p_cfg, "rc:fps_in_flex", 0);
    mpp_enc_cfg_set_s32(p_cfg, "rc:fps_in_num", 20);
    mpp_enc_cfg_set_s32(p_cfg, "rc:fps_in_denorm", 1);
    mpp_enc_cfg_set_s32(p_cfg, "rc:fps_out_flex", 0);
    mpp_enc_cfg_set_s32(p_cfg, "rc:fps_out_num", 20);
    mpp_enc_cfg_set_s32(p_cfg, "rc:fps_out_denorm", 1);
    mpp_enc_cfg_set_s32(p_cfg, "rc:gop", 20);
    mpp_enc_cfg_set_s32(p_cfg, "rc:skip_cnt", 0);

    MppEncRcMode t_rc_mode = MPP_ENC_RC_MODE_CBR;
    mpp_enc_cfg_set_s32(p_cfg, "rc:mode", t_rc_mode);
    switch (t_rc_mode) {
        case MPP_ENC_RC_MODE_CBR : { //固定码率
            /* CBR mode has narrow bound */
            mpp_enc_cfg_set_s32(p_cfg, "rc:bps_max", in_stream_info->bit_rate * 17 / 16);
            mpp_enc_cfg_set_s32(p_cfg, "rc:bps_min", in_stream_info->bit_rate * 15 / 16);
        } break;
        case MPP_ENC_RC_MODE_VBR ://动态码率
        case MPP_ENC_RC_MODE_AVBR : {
            /* VBR mode has wide bound */
            mpp_enc_cfg_set_s32(p_cfg, "rc:bps_max", in_stream_info->bit_rate * 17 / 16);
            mpp_enc_cfg_set_s32(p_cfg, "rc:bps_min", in_stream_info->bit_rate * 1 / 16);
        } break;
        default : {
            /* default use CBR mode */
            mpp_enc_cfg_set_s32(p_cfg, "rc:bps_max", in_stream_info->bit_rate * 17 / 16);
            mpp_enc_cfg_set_s32(p_cfg, "rc:bps_min", in_stream_info->bit_rate * 15 / 16);
        } break;
    }

    mpp_enc_cfg_set_s32(p_cfg, "rc:quality", MPP_ENC_RC_QUALITY_MEDIUM);

    mpp_enc_cfg_set_s32(p_cfg, "rc:qp_init", -1);
    mpp_enc_cfg_set_s32(p_cfg, "rc:qp_max", 51);
    mpp_enc_cfg_set_s32(p_cfg, "rc:qp_min", 10);
    mpp_enc_cfg_set_s32(p_cfg, "rc:qp_max_i", 51);
    mpp_enc_cfg_set_s32(p_cfg, "rc:qp_min_i", 10);
    mpp_enc_cfg_set_s32(p_cfg, "rc:qp_ip", 2);

/*
         * H.264 profile_idc parameter
         * 66  - Baseline profile
         * 77  - Main profile
         * 100 - High profile
         */
    mpp_enc_cfg_set_s32(p_cfg, "h264:profile", 77);
    /*
     * H.264 level_idc parameter
     * 10 / 11 / 12 / 13    - qcif@15fps / cif@7.5fps / cif@15fps / cif@30fps
     * 20 / 21 / 22         - cif@30fps / half-D1@@25fps / D1@12.5fps
     * 30 / 31 / 32         - D1@25fps / 720p@30fps / 720p@60fps
     * 40 / 41 / 42         - 1080p@30fps / 1080p@30fps / 1080p@60fps
     * 50 / 51 / 52         - 4K@30fps
     */
    mpp_enc_cfg_set_s32(p_cfg, "h264:level", 21);
    mpp_enc_cfg_set_s32(p_cfg, "h264:cabac_en", 1);
    mpp_enc_cfg_set_s32(p_cfg, "h264:cabac_idc", 0);
    mpp_enc_cfg_set_s32(p_cfg, "h264:trans8x8", 1);

    ret = mppApi->control(mppCtx, MPP_ENC_SET_CFG, p_cfg);
    if (ret) {
        LOGE("mpi control enc set cfg failed ret %d", ret);
        return RET_ERROR;
    }

    MppEncSeiMode sei_mode;
    sei_mode = MPP_ENC_SEI_MODE_ONE_FRAME;
    ret = mppApi->control(mppCtx, MPP_ENC_SET_SEI_CFG, &sei_mode);
    if (ret)
    {
        LOGE("mpi control enc set cfg failed ret %d", ret);
        return RET_ERROR;
    }

    MppEncHeaderMode header_mode;
    header_mode = MPP_ENC_HEADER_MODE_EACH_IDR;
    ret = mppApi->control(mppCtx, MPP_ENC_SET_HEADER_MODE, &header_mode);
    if (ret) {
        LOGE("mpi control enc set header mode failed ret %d", ret);
        return RET_ERROR;
    }

我感觉后面的MppEncSeiMode /MppEncHeaderMode 设置都没有生效。

HermanChen, 2022-10-09:
mpp_enc_proc_cfg 里面在 MPP_ENC_SEI_MODE_ONE_FRAME 和 MPP_ENC_HEADER_MODE_EACH_IDR 处理的位置加点打印看看

GrizaJ, 2022-10-12:
按照你说的我在这mpp_enc_proc_cfg 方法中添加了如下的日志信息：
[image]

日志输出如下：
[image]

同时我也在同文件的 mpp_enc_normal()/try_proc_low_deley_task()/proc_async_task()方法中 添加如下日志：
[image]
我期望在每次输出关键帧的写头数据的时候能打印这行。但是一直都没输出这天日志。

你看一下还有什么情况会导致我这个问题。

以下是我过滤mpp_enc 所得到的所有日志。

xiaohuihuige, 2025-06-27:
sei_mode =  MPP_ENC_SEI_MODE_DISABLE;