rockchip-linux/mpp #328: all tests run, but don't print anything

Timvrakas, 2022-10-10:
./mpp_platform_test for example, just runs and returns with no information printed.

I built from latest source, on a ROCK 3A, RK3568, with [rock-3a-ubuntu-focal-server-arm64-20221001-0212-gpt.img.xz](https://github.com/radxa-build/rock-3a/releases/download/20221001-0110/rock-3a-ubuntu-focal-server-arm64-20221001-0212-gpt.img.xz)

I can see `/dev/mpp_service`

What am I doing wrong?

HermanChen, 2022-10-10:
export mpp_syslog_perror=1 then run

Timvrakas, 2022-10-10:
Ah, I didn't realize the log output would end up in /var/log/syslog, rather than printed directly.

That export does not appear to change anything, but I now have the encoder working.

Is there a way to have the test programs print to stdout, rather than to syslog?

HermanChen, 2022-10-10:
Good point. We will change it.

HermanChen, 2022-10-10:
Change the os_log_xxx function to printf in os_log.cpp at osal/linux

closed 2023-02-05 by Timvrakas