rockchip-linux/mpp #332: The latest version of MPP decodes 4K JPEG pictures and sets the output format to MPP_FMT_BGRA8888, and the picture can only display half of it.

beifang-123, 2022-10-21:


beifang-123, 2022-10-21:
RK312X_LINUX_SDK_BETA_V0.1.1_20190429  （This is the SDK version）

beifang-123, 2022-10-21:
Please see the attached output00.yuv data, help to look at this problem, thank you!
[output00.zip](https://github.com/rockchip-linux/mpp/files/9835289/output00.zip)

HermanChen, 2022-10-21:
是 develop 分支么？

beifang-123, 2022-10-21:
是 develop 分支

beifang-123, 2022-10-21:
> 是 develop 分支么？

是的

HermanChen, 2022-10-21:
mpi_dec_test 的命令和 log 提供下

beifang-123, 2022-10-21:
> mpi_dec_test 的命令和 log 提供下
./mpi_dec_test -i /mnt/sdcard/jpeg3840X2160.jpg  -w 3840 -h 2160 -o /mnt/sdcard/output.yuv
[test.zip](https://github.com/rockchip-linux/mpp/files/9835613/test.zip)

HermanChen, 2022-10-21:
jpeg 文件也提供下
如果有旧版本是好的话，可以回退版本，用二分法找一下是哪个补丁引入的问题

beifang-123, 2022-10-21:
> jpeg 文件也提供下 如果有旧版本是好的话，可以回退版本，用二分法找一下是哪个补丁引入的问题

麻烦您也帮忙试试，感谢！
[jpeg3840X2160.zip](https://github.com/rockchip-linux/mpp/files/9836217/jpeg3840X2160.zip)

HermanChen, 2022-10-21:
这个源是 YUV444 的，解码出来要用 444 的方式打开，我这边看解码是正常的

HermanChen, 2022-10-21:
在 3288 上测试了 vdpu1 的路径，也是正常的，没有问题

beifang-123, 2022-10-21:
> 在 3288 上测试了 vdpu1 的路径，也是正常的，没有问题

好的，谢谢

beifang-123, 2022-10-21:
> 这个源是 YUV444 的，解码出来要用 444 的方式打开，我这边看解码是正常的

我有个疑问，我这里别的元也有yuv444格式的，MPP_JPEG_OUTPUT_FORMAT设置为MPP_FMT_BGRA8888，输出是正常的，源文件见附件
[04.zip](https://github.com/rockchip-linux/mpp/files/9837192/04.zip)

HermanChen, 2022-10-21:
先不要配置 OUTPUT_FORMAT，看看解码图像是不是正常，再配置上去对比看看

beifang-123, 2022-10-21:
> 先不要配置 OUTPUT_FORMAT，看看解码图像是不是正常，再配置上去对比看看

好的，谢谢！

Yg-Luo, 2022-10-22:
> 先不要配置 OUTPUT_FORMAT，看看解码图像是不是正常，再配置上去对比看看

您好，我用MppTask和poll/dequeue/enqueue 接口配合写了一个demo，可以正常解码，但解码出来的文件表层有一层绿幕，请查看附件现象和log,谢谢！
[image]

[norml.zip](https://github.com/rockchip-linux/mpp/files/9843876/norml.zip)

closed 2022-10-24 by Yg-Luo