rockchip-linux/mpp #333: VA-API implementation

Zipdox, 2022-10-22:
Would it be possible to implement a VA-API driver? This could be done in two possible ways.

- A shim on top of MPP, similar to https://github.com/elFarto/nvidia-vaapi-driver
- A native VA-API driver in place of MPP

HermanChen, 2022-10-31:
The VA-API is still support by intel? I thought intel had turn to media sdk?

Zipdox, 2022-10-31:
> The VA-API is still support by intel? I thought intel had turn to media sdk?

VA-API is the primary video acceleration API on Linux. Intel created it. There's no reason why they'd deprecate it.

HermanChen, 2022-10-31:
community.intel.com/t5/Media-Intel-oneAPI-Video/Media-SDK-API-vs-direct-call-to-VA-API/m-p/1050658#M7771
emm, Media SDK seems a wrapper of vaapi.
We do not have enough guys to do this job now.
But I think it is possible to make a vaapi middleware to support vaapi.

nyanmisaka, 2022-10-31:
It would be great if there is a shim between MPP and VAAPI since most Linux browsers and player adopt VAAPI as their default hardware acceleration method such as Firefox, Chrome and MPV player.

call VAAPI -> shim fills params and data to MPP -> return VAAPI surface

If this can be achieved, Rockchip will be the best ARM SoC with HW decoding performance under Linux desktop.