rockchip-linux/mpp #334: I use mpp to decode the video file hard and find that the decoding delay is relatively large. Please see the attachment for the video and log.

Yg-Luo, 2022-10-27:
I use mpp to decode the video file hard and find that the decoding delay is relatively large. Please see the attachment for the video and log. Most videos are normal with little delay.
[MPP硬解延时100ms.zip](https://github.com/rockchip-linux/mpp/files/9875487/MPP.100ms.zip)

HermanChen, 2022-10-27:
加上 IMEDIATE_OUT 的 control

HermanChen, 2022-10-27:
这个也要看片源，有一片源的显示序和解码序差别就是很大的

Yg-Luo, 2022-10-27:
> 加上 IMEDIATE_OUT 的 control

请教一下这个命令的control对应的第三个参数是什么

HermanChen, 2022-10-28:
整个 U32 变量为 1，把这个变量地址送进去

Yg-Luo, 2022-10-28:
> 整个 U32 变量为 1，把这个变量地址送进去

嗯，这样验证过了，输出延时还是一样的

HermanChen, 2022-10-28:
片源里有没有 B 帧？还是只有 I 帧和 P 帧？

Yg-Luo, 2022-10-28:
> 片源里有没有 B 帧？还是只有 I 帧和 P 帧？
是有B帧的
[image]

HermanChen, 2022-10-31:
有 B 帧就很难保证时延了，解码序和显示端不一致，时延就会不断波动

closed 2022-10-31 by HermanChen