rockchip-linux/mpp #335: ROC-RK3588 Debian OS 尝试rtsp+mpp拉流失败

Rao2000, 2022-10-27:
1. 使用gst-launch-1.0命令行，利用rtsp进行拉流并且展示在屏幕上，cpu的占用率符合我们的预期，4K视频拉流是30%占用率，1080P视频是7%的占用率，感觉上是符合瑞芯微官方的宣传的。
2. 使用OpenCV(4.5.1)版本进行拉流，加上Gstreamer（如下代码），效果和没有加一样。取流不是实时的，并且cpu占用率很高。
`capture.open(url_, cv::CAP_GSTREAMER);`
Pipeline如下所示：
`string url = "rtspsrc location=" + url_ + " latency=0 ! rtph264depay ! h264parse ! mppvideodec ! videoconvert ! video/x-raw,format=(string)BGR  ! appsink sync=false";`
3. 使用rtsp+ffmpeg+mpp，`ret= mpi->decode_get_frame(ctx, &frame);` 运行结果是ret=0。但是`err_info= mpp_frame_get_errinfo(frame)` 是err_info是1.有没有可能是mpp驱动有问题，或者是mpp版本有问题？
4. 使用rtsp(socket)拉流+mpp，`ret= mpi->decode_get_frame(ctx, &frame);` 运行结果是ret=0。但是frame的值一直为0x0。
请各位大佬指导一下哪有问题，或者有没有什么新的方法成功的。

Uhao-P, 2023-02-03:
我也出现了和你一样的问题3，请问你解决了吗

HermanChen, 2023-02-07:
3. get_frame 的返回为 0，说明流程正常，get_errInfo 报错，指的是解码出来的图像是错误的，可能是码流出错导致的，很可能是网络丢包

gitthb, 2023-12-16:
遇到同样的第3个问题。运行不到10秒钟mpp_frame_get_errinfo就会是1或者mpp_frame_get_discard是1，网络的话ping测试丢包率是0.不晓得怎么处理

HermanChen, 2023-12-18:
网络出错正常，异常的帧丢弃即可，然后继续送码流取图像

chenzhu005774, 2024-03-12:
> 遇到同样的第3个问题。运行不到10秒钟mpp_frame_get_errinfo就会是1或者mpp_frame_get_discard是1，网络的话ping测试丢包率是0.不晓得怎么处理

请问解决了吗