rockchip-linux/mpp #337: MPP硬解jpeg图片，设置MPP_DEC_SET_OUTPUT_FORMAT指定输出，输出格式还是原来的格式

Yg-Luo, 2022-10-28:
jpeg硬解图片问题是这样的，以YUV444格式为例：
1.不指定输出格式时，在mpp_frame_get_buffer后拿到的buffer dump出来去显示是YUV444格式。
2.指定输出格式，在mpp_frame_get_buffer后拿到的buffer dump出来去显示还是是YUV444格式，感觉MPP_DEC_SET_OUTPUT_FORMAT没使能，没有跑去转换格式。

HermanChen, 2022-10-28:
MPP_DEC_SET_OUTPUT_FORMAT 配置成什么格式了？还有这个也与芯片相关，有的芯片支持转格式，有的不支持

Yg-Luo, 2022-10-28:
芯片是rk3128, 配置的输出格式是MPP_FMT_RGBA8888

Yg-Luo, 2022-10-28:
> MPP_DEC_SET_OUTPUT_FORMAT 配置成什么格式了？还有这个也与芯片相关，有的芯片支持转格式，有的不支持

芯片是rk3128, 配置的输出格式是MPP_FMT_RGBA8888

Yg-Luo, 2022-10-28:
> MPP_DEC_SET_OUTPUT_FORMAT 配置成什么格式了？还有这个也与芯片相关，有的芯片支持转格式，有的不支持

芯片是rk3128,配置的输出格式是MPP_FMT_BGRA8888

Yg-Luo, 2022-10-28:
> MPP_DEC_SET_OUTPUT_FORMAT 配置成什么格式了？还有这个也与芯片相关，有的芯片支持转格式，有的不支持
目前测试验证，图片分辨率超过3840X1088, YUV444转YUV420SP就失效了，小于这个分辨率可以正常转换

Yg-Luo, 2022-10-31:
> MPP_DEC_SET_OUTPUT_FORMAT 配置成什么格式了？还有这个也与芯片相关，有的芯片支持转格式，有的不支持
一张YUV444格式3840X2160的图片，转换格式后，目前渲染出来只能显示3840X1080的部分图片，另外一半显示黑色。这种情况经过验证是MPP_DEC_SET_OUTPUT_FORMAT 大于3160X1080失效了，有没有什么解决办法，还是说只能我这边写代码重新转换格式了呢？请帮忙给个建议，非常感谢！！！

HermanChen, 2022-10-31:
用的是什么芯片？

Yg-Luo, 2022-10-31:
> 用的是什么芯片？

RK3128

HermanChen, 2022-10-31:
估计是 4.4 内核，解码图像的亮度大于 4M 之后，色度地址偏移可能出问题，这个是旧版本内核的问题，目前没有修改计划……

Yg-Luo, 2022-10-31:
> 估计是 4.4 内核，解码图像的亮度大于 4M 之后，色度地址偏移可能出问题，这个是旧版本内核的问题，目前没有修改计划……
好的，感谢!

Yg-Luo, 2022-10-31:
> 估计是 4.4 内核，解码图像的亮度大于 4M 之后，色度地址偏移可能出问题，这个是旧版本内核的问题，目前没有修改计划……
麻烦请教一下具体是vcodec_service问题还是哪一方面的问题呢？

Yg-Luo, 2022-11-05:
> > 估计是 4.4 内核，解码图像的亮度大于 4M 之后，色度地址偏移可能出问题，这个是旧版本内核的问题，目前没有修改计划……
> > 好的，感谢!

已解决此问题，4k可以正常显示了，非常感谢！

closed 2022-11-05 by Yg-Luo