rockchip-linux/mpp #339: 3588编译最新版本mpp报错“undefined reference to ‘__log_finite’”

FuHTong, 2022-11-22:
采用的是目前较新的源码。
环境为Ubuntu18.04，libm.so.6，arrch64-linux-gnu-gcc-7.5.0，arrch64-linux-gnu-g++-7.5.0
是我环境没有配置正确还是某些包没有对应版本呢？感谢百忙之中的解答。

HermanChen, 2022-11-22:
完整错误log？

FuHTong, 2022-11-22:
感谢您的答复,第一个错误发生在我运行加载视频时,所以我想更新mpp.
`type:test.h264
mpp[7266]: mpp_info: mpp version: 41c9c809 author: sayon.chen    2020-12-24 [mpp2str]: Add rc_gop_mode 2 str
mpp[7266]: mpp_rt: NOT found ion allocator
mpp[7266]: mpp_rt: found drm allocator
mpp[7266]: mpp_platform: can not found match soc name: rockchip,rk3588-evb4-lp4-v10 rockchip,rk3588
mpp[7266]: mpp_serivce: set client type 0 failed
mpp[7266]: hal_h264d_api: mpp_dev_init failed ret: -1
mpp[7266]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1
mpp[7266]: mpp_hal: mpp_hal_init could not found coding type 7
mpp[7266]: mpp_dec: mpp_dec_init could not init hal
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp_time: mpp_clock_put invalid clock (nil)
mpp[7266]: mpp: error found on mpp initialization

(clearance_bin:7266): GStreamer-CRITICAL **: 10:36:28.349: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed`

第二个错误发生在我更新mpp时,
`[ 67%] Built target hal_jpege
Scanning dependencies of target hal_h265e_vepu541
Scanning dependencies of target hal_jpege_rkv
[ 67%] Building C object mpp/hal/rkenc/h265e/CMakeFiles/hal_h265e_vepu541.dir/hal_h265e_vepu580.c.o
[ 67%] Building C object mpp/hal/rkenc/jpege/CMakeFiles/hal_jpege_rkv.dir/hal_jpege_vepu540c.c.o
[ 67%] Building C object mpp/hal/rkenc/h265e/CMakeFiles/hal_h265e_vepu541.dir/hal_h265e_vepu541.c.o
[ 68%] Building C object mpp/hal/rkenc/h265e/CMakeFiles/hal_h265e_vepu541.dir/hal_h265e_vepu540c.c.o
[ 68%] Linking CXX static library libhal_h264e.a
../libenc_rc.a(rc_model_v2_smt.c.o): In function `rc_model_v2_smt_start':
rc_model_v2_smt.c:(.text+0x1f8): undefined reference to `__log_finite'
../libenc_rc.a(rc_model_v2_smt.c.o): In function `bits_model_smt_init':
rc_model_v2_smt.c:(.text+0x1164): undefined reference to `__log_finite'
../libenc_rc.a(rc_model_v2.c.o): In function `reenc_calc_cbr_ratio':
rc_model_v2.c:(.text+0x1e50): undefined reference to `__log_finite'
collect2: error: ld returned 1 exit status
mpp/codec/rc/test/CMakeFiles/rc_api_test.dir/build.make:98: recipe for target 'mpp/codec/rc/test/rc_api_test' failed
make[2]: *** [mpp/codec/rc/test/rc_api_test] Error 1
CMakeFiles/Makefile2:3429: recipe for target 'mpp/codec/rc/test/CMakeFiles/rc_api_test.dir/all' failed
make[1]: *** [mpp/codec/rc/test/CMakeFiles/rc_api_test.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 68%] Built target hal_h264e
[ 69%] Linking CXX static library libhal_jpege_rkv.a
[ 69%] Built target hal_jpege_rkv
In file included from /home/rpdzkj/mpp/mpp/hal/rkenc/h265e/hal_h265e_vepu540c.c:28:0:
/home/rpdzkj/mpp/mpp/hal/rkenc/h265e/hal_h265e_vepu540c.c: In function ‘hal_h265e_v540c_gen_regs’:
/home/rpdzkj/mpp/osal/inc/mpp_common.h:203:13: warning: assuming signed overflow does not occur when assuming that (X + c) >= X is always true [-Wstrict-overflow]
     else if (a > amax) return amax;
             ^
[ 69%] Linking CXX static library libmpp_vproc.a
[ 69%] Built target mpp_vproc
[ 69%] Linking CXX static library libhal_h265e_vepu541.a
[ 69%] Built target hal_h265e_vepu541
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2`

再次感谢您

HermanChen, 2022-11-22:
> mpp[7266]: mpp_platform: can not found match soc name: rockchip,rk3588-evb4-lp4-v10 rockchip,rk3588
mpp[7266]: mpp_serivce: set client type 0 failed
mpp[7266]: hal_h264d_api: mpp_dev_init failed ret: -1
mpp[7266]: mpp_hal: mpp_hal_init hal h264d_rkdec init failed ret -1

这里说明内核驱动不对，内核的 mpp 设备没有正常加载
跑一下 mpp_platform_test 看看

FuHTong, 2022-11-22:
你好，运行了已经安装的mpp_platform_test,运行报错：
`mpp_log:can not found match soc name: rockchip,rk3588-evb4-lp4-v10 rockchip,rk3588
mpp_plat_test: chip name: rockchip,rk3588-evb4-lp4-v10 rockchip,rk3588
mpp_plat_test: 
mpp_plat_test: chip vcodec type 00000000
mpp_plat_test: 
mpp_plat_test: start probing decoder device name:
mpp_plat_test: H.264 decoder:(null)
mpp_plat_test: H.265 decoder:(null)
mpp_plat_test: MJPEG decoder:(null)
mpp_plat_test: VP9 decoder:(null)
mpp_plat_test: avs decoder:(null)
mpp_plat_test: 
mpp_plat_test: start probing encoder device name:
mpp_plat_test: H.264 encoder:(null)
mpp_plat_test: H.265 encoder:(null)
mpp_plat_test: MJPEG encoder:(null)
`

请问是需要我更新内核还是整体刷机呢？