rockchip-linux/mpp #341: 宏块级比特率控制

BlueWaterCrystal, 2022-11-29:
开发环境：萤火虫RK3588

是否有可能影响编码器的模式选择？
使用更多或强制跳过宏块。

有谁知道 h264_skip_sad_penalty 是做什么的？
除非有人知道寄存器的任何文档？

谢谢你。

HermanChen, 2022-11-29:
可以的影响
这个编码器的模式判决没有用 full rdo 就是 sad 和各种 penalty 权重叠加来选模式
skip_sad_penalty 就是 skip 模式的判决权重增加值

BlueWaterCrystal, 2022-11-29:
谢谢你的澄清

对不起，我有点糊涂了。

skip_sad_penalty是用来增加单个MB跳过标志中的跳过量的吗？

我假设任何青睐表，值越高，该模式的使用成本越高。
虽然我不知道惩罚表是否以同样的方式工作。

所以，如果我有3600个宏块，我想跳过80-95%的宏块
我应该调整哪些表格？

HermanChen, 2022-11-29:
可以实测下，0~255 好像是越大越倾向，所以我估计是这样：
skip cost = skip sad - skip penalty
intra cost = intra sad - intra penalty
inter cost = iner sad - inter penalty
然后三个模式选 cost 最小的一个？
可以实验看看 0 和 255 是哪个是更倾向的

BlueWaterCrystal, 2022-12-01:
谢谢你。

我很快就能测试它。

如果值为U32，它不能看到255以上吗？

如果有成本表，是否有预算？ 可能每帧的最大成本。

如果是这样，那么这些表在理论上应该可以工作，否则我认为由于编码器的逻辑它不会工作。

另一个糟糕的情况是预算是每 MB 而不是每帧。
                                                                                                                                                                                                 
我唯一不知道的是成本是用什么货币表示的，比特，只是整数，还是其他，以及是否有可能提取预算是多少。

否则是否有可能在处理成比特流之前提取宏块数据或数组？

HermanChen, 2022-12-05:
看下 hal 中配置寄存器的部分吧，数据位宽应该只有 8bit 这样，如果有更大的位宽，应该可以到 256 以上
这个机制应该是块级的，对于 cost 计算的部分，我们也不太清楚内部的细节……