rockchip-linux/mpp #342: RK3568 上运行 mpi_enc_test，编码h264出现白线，花屏

6358482, 2022-12-05:
你好，rock pi 3A上使用USB摄像头，运行./mpi_enc_test -i /dev/video1 -o /home/share/encodeout.h264  -t 7 -f 8 -w 640 -h 480
使用VLC播放encodeout.h264，画面有变化时出现白线，花屏。

quanjin9099, 2023-04-08:
这个问题解决了么?我在3588上压缩yuyv为h264的时候也碰上这个问题了

HermanChen, 2023-04-10:
这看起来应该是 buffer 的 cache 问题，没有把数据真正写到 ddr 里

quanjin9099, 2023-04-10:
同样的压缩,使用mipi摄像头做数据源就没有问题...

有问题的是使用usb3.0 的摄像头(yuyv)

kuangdazzidd, 2023-11-24:
这问题好几年了官方都没人修吗？

HermanChen, 2023-11-24:
cache 补丁还在测试中，还没正式合并

kuangdazzidd, 2023-11-24:
测试补丁可以分享下吗？目前项目使用RK3588S卡在这里了

HermanChen, 2023-11-24:
推了一个 cache_test 的分支上来，可以试下：
https://github.com/rockchip-linux/mpp/tree/cache_test

kuangdazzidd, 2023-11-24:
@HermanChen 好的，十分感谢

kuangdazzidd, 2023-11-24:
> 推了一个 cache_test 的分支上来，可以试下： https://github.com/rockchip-linux/mpp/tree/cache_test

[image]
hello，Herman，测试了下该分支，问题还是存在，是不是我需要配置什么？

Iyiren, 2024-01-04:
我们也遇到这个问题了，通过gst命令工具没有这个问题，通过mpp api就有这个问题，视频数据通过usb3.0进入到cpu

FumasterLin, 2024-01-05:
将输入的yuv保存在本地，用mpi_enc_test读取本地yuv进行编码是否有问题？

HermanChen, 2024-12-30:
新的版本还有问题么？

lxyevl, 2024-12-30:
rk3588s2用的内核是Linux rock-5c 6.1.43-20-rk2312，mpp是刚从clone下来编译的

lixsg, 2024-12-31:
我也遇到了同样的问题，在摄像头采集到运动物体时花屏，rk3588

Li-Chang-Yi, 2025-03-18:
还是有问题

Li-Chang-Yi, 2025-03-18:
还是有问题

Li-Chang-Yi, 2025-03-28:
> 新的版本还有问题么？

还是有问题

Li-Chang-Yi, 2025-03-28:
> 我也遇到了同样的问题，在摄像头采集到运动物体时花屏，rk3588

我也是

Li-Chang-Yi, 2025-03-28:
> 新的版本一直有问题，有个临时解决方案，把mpp buffer拷贝到用户内存一份，再给编码器使用，这里应该是底层驱动buffer一致性没处理好，一直也没解决，有遇到同样问题的可以参考临时解决下

好的，谢谢我试试

Li-Chang-Yi, 2025-03-29:
> 把mpp buffer拷贝到用户内存一份

[image]

[image]

请问下就是把第一张图的cam_buf拷贝到用户内存一份吗？后面是直接送的frame进编码器

kuangdazzidd, 2025-03-31:
[image] 把1这里的buffer拷贝一份，mpp_frame_set_buffer使用拷贝的buffer

Li-Chang-Yi, 2025-04-01:
好的，我试试> [image] 把1这里的buffer拷贝一份，mpp_frame_set_buffer使用拷贝的buffer

PIPIKAI, 2025-04-01:
> 新的版本一直有问题，有个临时解决方案，把mpp buffer拷贝到用户内存一份，再给编码器使用，这里应该是底层驱动buffer一致性没处理好，一直也没解决，有遇到同样问题的可以参考临时解决下

感谢，有用

Li-Chang-Yi, 2025-04-01:
> [image] 把1这里的buffer拷贝一份，mpp_frame_set_buffer使用拷贝的buffer

我拷贝 memcopy 到一份 new mppbuffer 中，直接送还是有问题，必须要对这个 new mppbuffer 调用 sync_end 才正常，这是正常操作吗？

Li-Chang-Yi, 2025-04-02:
> [image] 把1这里的buffer拷贝一份，mpp_frame_set_buffer使用拷贝的buffer
加了个 memcopy 和 ffmpeg h264 推流 试了下延时大了很多😟

HermanChen, 2025-04-02:
用 rga 拷贝看看

Li-Chang-Yi, 2025-04-02:
> 用 rga 拷贝看看

好的，我试试

Li-Chang-Yi, 2025-04-02:
> > 用 rga 拷贝看看
> 
> 好的，我试试

用 memcpy 和不用 memcpy， 编码 1000 帧 yuyv422 ，640x480
，都是大概 34900ms，一帧大概 34ms，正常吗？我怀疑是推流 和 那边直接用 opencv 软解码 rtsp 流的造成的延时

stz-source, 2025-06-28:
解决了吗兄弟？camera_frame_to_buf里是如何拷贝的？

stz-source, 2025-06-28:
> > > 用 rga 拷贝看看
> > 
> > 
> > 好的，我试试
> 
> 用 memcpy 和不用 memcpy， 编码 1000 帧 yuyv422 ，640x480 ，都是大概 34900ms，一帧大概 34ms，正常吗？我怀疑是推流 和 那边直接用 opencv 软解码 rtsp 流的造成的延时

解决了吗兄弟？camera_frame_to_buf里是如何拷贝的？我也遇到撕裂的问题了