rockchip-linux/mpp #344: mpp_frame_deinit()函数的释放问题。

kunknu, 2022-12-11:
你好，我模仿mpi_dec_test的用例写了个例子，把解码出来的MppFrame的frame用作一些图形处理，但是这个处理耗时比较长。
所以我把frame存到一个队列里面在别的线程做处理它，然后再释放。

现在出现了这样的问题：top命令查看到，程序的内存大小没有变化，但是系统的free空间一直在减少，used也没增加，cache就一直增加和free相反。free一直减少到一个值域，会导致我程序的另一个模块申请不到内存。
但是处理释放frame的操作如果放在解码线程里面就不会有问题

请问一下这个释放mpp_frame_deinit()函数是否对调用的时间点有要求。
能指点一下我编写错误的方向吗？

kunknu, 2022-12-11:
soc是rk3588
环境是ubuntu20.04

programcj, 2022-12-18:
我通过旧版本观察发现 在编码的时候encode_put_frame后不需要对frame进行mpp_frame_deinit，而新版本中需要mpp_frame_deinit， 这两者有什么区别吗？

zoutie126, 2023-01-05:
同样3588 ubuntu20.04 
长期解码测试,同样是进程内存不变 ,但是系统内存越来越少.

kunknu, 2023-01-16:
> 同样3588 ubuntu20.04 长期解码测试,同样是进程内存不变 ,但是系统内存越来越少.

后面我使用了ffmepg集成好的库去解码了，https://github.com/JeffyCN/FFmpeg，不过rockchip已经不再对ffmpeg进行更新了

closed 2023-01-16 by kunknu