rockchip-linux/mpp #346: mpi_enc_test编码mpi_dec_test生成的yuv，没有色彩

chuqingq, 2022-12-20:
你好！

我先用mpi_dec_test解码h264，生成yuv420sp的yuv；

然后用mpi_enc_test编码成h264，播放h264没有色彩。

请问需要怎样才能让编码后的h264有色彩？

谢谢！

HermanChen, 2022-12-20:
是不是编码器的格式配置有问题？参考下 mpi_rc2_test

chuqingq, 2022-12-20:
mpi_rc2_test的命令：
```sudo ./mpi_rc2_test -i mpi.yuv -f 0 -o mpi.h264 -t 7 -w 1920 -h 1080```

结果是：
```
Dec 20 12:47:47 x3588 mpp[13448]: mpi_enc_utils: cmd parse result:
Dec 20 12:47:47 x3588 mpp[13448]: mpi_enc_utils: input  file name: mpi.yuv
Dec 20 12:47:47 x3588 mpp[13448]: mpi_enc_utils: output file name: mpi.h264
Dec 20 12:47:47 x3588 mpp[13448]: mpi_enc_utils: width      : 1920
Dec 20 12:47:47 x3588 mpp[13448]: mpi_enc_utils: height     : 1080
Dec 20 12:47:47 x3588 mpp[13448]: mpi_enc_utils: format     : 0
Dec 20 12:47:47 x3588 mpp[13448]: mpi_enc_utils: type       : 7
Dec 20 12:47:47 x3588 mpp[13448]: mpi_rc_test: input file mpi.yuv size 1244160000
Dec 20 12:47:47 x3588 mpp[13448]: mpp_info: mpp version: 57f0767 author: Chandler Chen  2022-11-21 [hal_h264e] Vepu540c fix extra syntax error log for dpb
Dec 20 12:47:47 x3588 mpp[13448]: mpp_info: mpp version: 57f0767 author: Chandler Chen  2022-11-21 [hal_h264e] Vepu540c fix extra syntax error log for dpb
Dec 20 12:47:47 x3588 mpp[13448]: mpp: unable to create dec unused for soc rk3588 unsupported
Dec 20 12:47:47 x3588 mpp[13448]: mpi_rc_test: mpp_init dec failed
Dec 20 12:47:47 x3588 mpp[13448]: mpi_rc_test: mpi_rc_codec -1783464952 failed
```

之前用mpi_enc_test命令是：
```
sudo ./mpi_enc_test -i mpi.yuv -f 0 -o mpi.h264 -t 7 -w 1920 -h 1080
```
成功，但是播放没有色彩。

chuqingq, 2022-12-20:
我参考mpi_rc2_test进行配置，没有问题了，谢谢！

closed 2023-03-03 by chuqingq