rockchip-linux/mpp #348: RK3588: import decoded AFBC frames into GPU

kirgene, 2022-12-28:
I'm trying to decode H.265 video to AFBC compressed frames and import them into GPU using `EGL_EXT_image_dma_buf_import` extension.
On the decoder side I set output format:
```
    MppFrameFormat format = MPP_FMT_YUV420SP | MPP_FRAME_FBC_AFBC_V2;
    ret = mpi->control(ctx, MPP_DEC_SET_OUTPUT_FORMAT, &format);
    if (ret) {
            mpp_err("dec_mpi->control failed\n");
            goto MPP_TEST_OUT;
    }
```

and on the rendering side I import `dmaFd` (from decoded frame) and set modifiers:
```
uint64_t modifiers = DRM_FORMAT_MOD_ARM_AFBC(AFBC_FORMAT_MOD_BLOCK_SIZE_16x16 | AFBC_FORMAT_MOD_SPLIT | AFBC_FORMAT_MOD_SPARSE);

GLint img_attrs[] = {
                EGL_WIDTH, Width,
                EGL_HEIGHT, Height,
                EGL_LINUX_DRM_FOURCC_EXT, DRM_FORMAT_NV12,

                EGL_DMA_BUF_PLANE0_FD_EXT, dmaFd,
                EGL_DMA_BUF_PLANE0_OFFSET_EXT, 0,
                EGL_DMA_BUF_PLANE0_PITCH_EXT, 0,

                EGL_DMA_BUF_PLANE0_MODIFIER_HI_EXT, modifiers >> 32,
                EGL_DMA_BUF_PLANE0_MODIFIER_LO_EXT, modifiers & 0xFFFFFFFF,

                EGL_DMA_BUF_PLANE1_FD_EXT, dmaFd,
                EGL_DMA_BUF_PLANE1_OFFSET_EXT, Width * Height,
                EGL_DMA_BUF_PLANE1_PITCH_EXT, Width,

                EGL_DMA_BUF_PLANE1_MODIFIER_HI_EXT, modifiers >> 32,
                EGL_DMA_BUF_PLANE1_MODIFIER_LO_EXT, modifiers & 0xFFFFFFFF,

                EGL_NONE
        };
EGLImage image = eglCreateImageKHR( egl_display, EGL_NO_CONTEXT, EGL_LINUX_DMA_BUF_EXT, 0, &img_attrs[0] );
...
```

EGLImage is creating successfully, but when I try to render it, I see a noise (though I can distinguish some silhouettes of objects).

What I'm doing wrong here?
And is it even possible to import compressed frames generated by RK VPU into Mali GPU?

HermanChen, 2022-12-28:
Thank you for the feedback.

I ask my GPU colleague and get the following reference code:

`EGLint attr[] = {
    EGL_WIDTH, textureW,
    EGL_HEIGHT, textureH,
    EGL_LINUX_DRM_FOURCC_EXT, DRM_FORMAT_YUV420_8BIT,
    EGL_DMA_BUF_PLANE0_FD_EXT, fd,
    EGL_DMA_BUF_PLANE0_OFFSET_EXT, 0,
    EGL_DMA_BUF_PLANE0_PITCH_EXT, stride, 
    EGL_DMA_BUF_PLANE0_MODIFIER_LO_EXT, (AFBC_FORMAT_MOD_SPARSE | AFBC_FORMAT_MOD_BLOCK_SIZE_16x16)&0xffffffff,
    EGL_DMA_BUF_PLANE0_MODIFIER_HI_EXT, (0x08<<24),  // ARM platform flag
    EGL_NONE
};`

Please have a try : )

kirgene, 2022-12-28:
@HermanChen thanks, it works now!
Have tested a couple of videos and noticed a strange artifact: a grey line at the top of the image (4 pixels in height).
Here is a diff between the two images (without AFBC on the right and with AFBC on the left).
[image]

HermanChen, 2022-12-28:
The valid pixel data has a 4 line offset for H.264/H265/AVS2, and 0 line offset for VP9. AV1 fbc data is more complicated.
Get this offset_y by mpp_frame_get_offset_y with the output MppFrame.

kirgene, 2022-12-28:
Got it. Can we get rid of this offset in exported dma-buf?

HermanChen, 2022-12-28:
We can't. This is hardware decoder behavior. We have to setup the display offset to handle it.

kirgene, 2022-12-28:
Make sense. Anyway, that's not a problem.
Thanks again for the help.

closed 2022-12-28 by kirgene

xiangxud, 2023-02-24:
eglError EGL_BAD_ALLOC (0x3003) this error how to deal