rockchip-linux/mpp #355: [RK3568] Issues while decoding H265

alus96, 2023-02-10:
Hi,
when trying to decode **some** of h265 decoded videos I get these errors in logs:

```mpp[431]: H265D_PARSER: PPS id out of range: 0
mpp[431]: H265D_PARSER: hls_slice_header error ret = -1004
mpp[431]: H265D_PARSER: Error parsing NAL unit #0,error ret = 0xd.
mpp[431]: H265D_PARSER: current stream is no right skip it (nil)
mpp[431]: H265D_PARSER: PPS id out of range: 0
```

The video is successfully decoded by software decoders at the same time. 

If it could be of any use. here are parameters of normally decoded (by mpp) video:
```Stream #0:0: Video: hevc (Main) (HEVC / 0x43564548), yuvj420p(pc, bt709), 3840x2160,```

And the problematic video:
```Stream #0:0: Video: hevc (Main) (HEVC / 0x43564548), yuv420p(tv), 2560x1440```

alus96, 2023-02-10:
[DS.zip](https://github.com/rockchip-linux/mpp/files/10709219/DS.zip)
The video itself

chrisbeeze, 2025-10-30:
这个问题有最新进展吗？
在我的测试中，解码宇视（Uniview）摄像机的 H.265 时，也出现了类似的问题。
供参考，我的测试 SOC 是 RK3588。

```
rk3588 mpp[25245]: H265D_PARSER: mpp_hevc_decode_nal_sei error ret = -1004
rk3588 mpp[25187]: H265D_PARSER: mpp_hevc_decode_nal_sei error ret = -1004
rk3588 mpp[25245]: H265D_PARSER: mpp_hevc_decode_nal_sei error ret = -1004
rk3588 mpp[25187]: H265D_PARSER: mpp_hevc_decode_nal_sei error ret = -1004
```

VicLiDev, 2025-10-30:
@alus96 
Hi,
You can try updating to the latest version of MPP. After extracting the bitstream, I tested it on the latest MPP and didn’t encounter this issue.

chrisbeeze, 2025-10-30:
[log.zip](https://github.com/user-attachments/files/23225369/log.zip)

> [@chrisbeeze](https://github.com/chrisbeeze) 您好，您这边的码流可以发一下，可能与之前不是同一个问题

这是我从宇视IPC上抓取RTSP流时捕获的数据包文件。这有用吗？

VicLiDev, 2025-10-30:
这看起来不是有效的码流文件，可以试一下

# android 系统
setenforce 0
setprop mpp_debug 0x200
setprop vendor.mpp_debug 0x200
chmod -R 777 data

# linux 系统
export mpp_debug=0x200
chmod -R 777 data

执行命令之后，再启动app，进行解码操作，看能否dump得到相应的码流文件

VicLiDev, 2025-10-30:
@chrisbeeze 
您好，
过一下码流，可以看到sei信息应该是有问题的，可以使用码流工具，也可以使用ffmpeg过码流确认，
我这边使用ffmpeg可以看到：
[hevc @ 0x6156d3f1d9c0] Skipping invalid undecodable NALU: 39N/A speed=   2x
[hevc @ 0x6156d47f1a90] Skipping invalid undecodable NALU: 39N/A speed=2.19x
[hevc @ 0x6156d3ded1d0] Skipping invalid undecodable NALU: 39N/A speed=2.21x

怀疑可能是宇视加的特殊处理的sei，或者加的sei有问题，但这不影响对码流进行解码

VicLiDev, 2025-10-30:
我根据前边给我的码流分析，是无害的，您可以确认下画面是否也没有任何影响？

如果想屏蔽该问题，理解调整下编码侧或许更合理些，可以根据ffmpeg或者码流工具中报的错误调整