rockchip-linux/mpp #358: mpp解码之后就的yuv数据，将其拷贝出来效率明显降低

lawaarch, 2023-02-16:
最近用瑞芯微的3658芯片硬解码。1080p的视频每帧数据最快3ms,平均下来5/6ms,解码能力可见一斑。但是就在我将其memcpy的时候花费了将近40ms太慢了。我看有人说用rga,能将drm的内存转换成normal的。可是我试了一下，卡住了具体原因还没有深究。在此我虚心讨教，可否有大佬，答疑解惑

HermanChen, 2023-02-17:
同一个问题不要到处提啊。。。

硬解码之后，可以用硬件做拷贝，比如 RGA 这些硬件，这个是常规的做法

如果一定要软件访问的话，可以把 buffer 配置成 cacheable 的，这个补丁还在开发中，需要更新 5.10 的内核，使用 dma_heap 分出来的 buffer 是可以的，但是在访问前需要手工清 cache，速度也不是非常快。

lawaarch, 2023-02-17:
我还想请教一个问题，用rga进行数据拷贝。用纯外部分配模式导入还是按照示例程序就可以。我测了测用示例的方式，得到的数据，拷贝到用户空间的内存中依旧很慢，如果用external方式导入，在commit的时候总是说创建失败。我是不是有啥没注意到的，就按照最新开发手册整的。

HermanChen, 2023-02-20:
rga 拷贝的话，在 github 上找一下新的 librga 的库，mpp 里的做法是很老的了……

closed 2023-02-20 by lawaarch

Sherman86, 2023-02-21:
2楼使用 dma_heap 分配 cacheable buffer 需要如下补丁。

[dma_heap_opts.patch](https://github.com/rockchip-linux/mpp/files/10790210/dma_heap_opts.patch)
[test_case.patch](https://github.com/rockchip-linux/mpp/files/10790211/test_case.patch)

lawaarch, 2023-02-21:
驱动程序 我直接用不了吧

Sherman86, 2023-02-22:
你是哪个环境？我编译个mpp库和test case给你吧

lawaarch, 2023-02-22:
> 

先谢谢 大佬了  我用的是rk3568平台linux-aarch64环境  但是开发者说还要更新内核

Sherman86, 2023-02-22:
不客气，我也是小白。将mpi_dec_test和mpp库推到板子上，就可以把buffer配置成cacheable。

更新内核的话，你可以参考一下附件第二章自己编译看看。

[mpp.tar.gz](https://github.com/rockchip-linux/mpp/files/10801875/mpp.tar.gz)
[test.tar.gz](https://github.com/rockchip-linux/mpp/files/10801876/test.tar.gz)
[Rockchip_RV1126_RV1109_Instruction_Linux_Separate_Building_EN.pdf](https://github.com/rockchip-linux/mpp/files/10801949/Rockchip_RV1126_RV1109_Instruction_Linux_Separate_Building_EN.pdf)