rockchip-linux/mpp #359: Does h.264 encoder(rv1126) support Intra refresh?

Daniella1989, 2023-02-17:
板子上的mpp version: 57ff4c6b author: Herman Chen   2021-09-13 [cmake]: Enable HAVE_DRM by default，目前遇到在CRB模式下，编码 4K H264，I帧会特别大的问题，请问在这个版本下，是否支持intra refresh的功能？
如果支持，是不是不能设置Gop？
如果不支持，请问哪个版本能够支持呢？谢谢

HermanChen, 2023-02-17:
新版本已经支持了，搜下 refresh_en

Daniella1989, 2023-02-21:
Hi, 正在尝试把intra refresh的代码合到当前代码中，目前在合的commit是https://github.com/HermanChen/mpp/commit/0ee683acf33712257afbfb439df77fb0c2db588f，在板子上跑了一下，发现在
set_frm_refresh_flag中，由于p->cpb.seq_cnt = 1, frame的flag：is_i_refresh和is_i_recovery始终是0

> if (p->refresh_length) {
        frm->is_i_refresh = ((frm->seq_idx  % p->igop) < p->refresh_length) && p->cpb.seq_cnt > 1;
        frm->is_i_recovery = !(frm->seq_idx  % p->igop)  && p->cpb.seq_cnt > 1;
    } else
        frm->is_i_refresh = 0;

想问一下，cpb.seq_cnt的具体含义是什么呢？在什么情况下，seq_cnt会大于1？谢谢

Daniella1989, 2023-03-30:
把intra fresh的代码合进去，目前环境走的是HalH264eVepu541相关的代码，但是感觉效果出来还是挺模糊的，就跟没有I帧，只有P帧一样。请问一下，intra refresh的功能有测试过么？

HermanChen, 2023-03-30:
可以提供下码流，intra_refresh 就是部分 I 块刷新，为了不引起图像质量的剧烈变化，I 块的 qp 只比 P 帧 qp 小一些，避免明显的刷新感

Daniella1989, 2023-03-31:
[encode.zip](https://github.com/rockchip-linux/mpp/files/11120594/encode.zip)
参数如下：4K视频
p->bps = 10627760*0.5
p->rc_mode = MPP_ENC_RC_MODE_CBR;
p->gop_mode = 3;
p->gop_len = 300;

mpp_enc_cfg_set_s32(cfg, "rc:qp_init", 26);
mpp_enc_cfg_set_s32(cfg, "rc:qp_max", 51);
mpp_enc_cfg_set_s32(cfg, "rc:qp_min", 18);
mpp_enc_cfg_set_s32(cfg, "rc:qp_max_i", 51);
mpp_enc_cfg_set_s32(cfg, "rc:qp_min_i", 18);
mpp_enc_cfg_set_s32(cfg, "rc:qp_ip", 2);
mpp_enc_cfg_set_s32(cfg, "rc:qp_step", 16);

 mpp_enc_cfg_set_s32(cfg, "rc:refresh_en", 1);
 mpp_enc_cfg_set_s32(cfg, "rc:refresh_mode", 0);
 mpp_enc_cfg_set_s32(cfg, "rc:refresh_num", 10);

码流（encode.h264）已上传，里面树看起来就非常模糊，我也不知道intra refresh有没有刷部分的I帧，麻烦帮忙看一下，谢谢

rediceli, 2023-04-04:
另外，跟问一下，GDR的解码是不是支持了呢？

HermanChen, 2023-04-04:
能支持，用 disable_error 就可以一定程度的支持，但支持的不完善……

HermanChen, 2023-04-04:
intra refresh 的确是有做刷新的，编码 I 块是没有问题的，只是这部分的 I 块没有用很小的 qp 去编码，也是为了减小刷新感

Daniella1989, 2023-04-17:
还想问一下，intra refresh功能打开后，会影响单独申请关键帧的操作么？申请关键帧是通过mpi->control(mpi_->ctx, MPP_ENC_SET_IDR_FRAME, param);